home-assistant/operating-system #1081: RPi4 5.7 RC no SSD boot in contrary to 64bit version

tdn131, 2020-12-12:
**Hardware Environment**
- [X] Raspberry Pi 4 4GB
Original RPi power supply for Pi4. No power shortage.
Samsung T7 500GB usb SSD
No microsd installed.
Latest stable eeprom of 2020-09-14.

**Home Assistant OS release:**
- [X] Fresh installation of release 5.7 problems for the normal 32bits version, none for 64bits version.

**Logs:**
I can not get logs, since there is no activity.
 
**Description of problem:**
I installed the latest version of 5.7 on my Samsung T7 500GB external SSD. When I boot the 32 version I get only the default Raspberry Pi eeprom screen. After that the screen stays black. Whereas the 64bits version shows the boot lines and boots through. Because I have done no adjustments, it should be easy to reproduce I hope. 

I tried this with two separate Raspberry Pi 4 4GB with two separate T7 500GB disks.

simeneide, 2020-12-12:
Similar issue here: Ive tried same on rpi4 8gb 64bits on a kingstone 128usb stick. Does not boot.

agners, 2020-12-12:
@tdn131 unfortunately, I don't have a Samsung T7 500GB disk here to test, so I cannot reproduce your setup. On my 2 test SSDs 32-bit does seem to work :man_shrugging: 

@simeneide so on your side its seems to also not boot with 64-bit. Did you try booting Raspberry Pi OS, did that work? Not all combinations do work, unfortunately.

In both cases, I recommend booting from SD card as this is much more reliable. The `datactl move` command can then be used to attach the external storage after boot. With this we can rely on the well tested and reliable SD card boot for the OS (where almost no writes happen anyway), and use the external USB SSD for data storage.

simeneide, 2020-12-12:
Hi, thanks for quick answer!

Yes, rpi OS have always worked. I managed to boot Hassio now and start installing. however it finalized, went into emergency mode (when I logged in with root in terminal). Will try to use the usb as storage only as suggested, this has taken too much time already 👍

shirou93, 2020-12-13:
5.8 same issue :/

carras, 2020-12-14:
> In both cases, I recommend booting from SD card as this is much more reliable. The `datactl move` command can then be used to attach the external storage after boot. With this we can rely on the [...]

Where can I find more information about this? I asked Pascal on the conference chat and he told there will be a blog post with that information but I fail to see it. Were there a conference part about it? I saw all "main stage" and you just mentioned briefly.

NikolayBorisov, 2020-12-14:
Raspberry Pi 4 **8GB**
Original RPi power supply for Pi4. No power shortage
Samsung T7 500GB usb SSD
No microsd installed
beta eeprom of 2020-10-28
hassos_rpi4-64-5.8.img.xz

Won't boot either

agners, 2020-12-15:
> Where can I find more information about this?

There is some information what is new in the blog post of Release 5: https://www.home-assistant.io/blog/2020/12/13/home-assistant-os-release-5/#external-data-disk

Documentation can be found here: https://github.com/home-assistant/operating-system/blob/dev/Documentation/partition.md#data-disk

closed 2021-05-12 by stale[bot]