home-assistant/operating-system #1084: hassos_ova-5.8.ova image does not work.

Meph79, 2020-12-13:
The OVA image for latest release seems to not have been created correctly..
Import failes in:
ESXi 6.7 U2
ESXi 7.0.0 
VMware Workstation 16

In ESXi 6.7 U2 following error is displayed:
[image]

This file were used:
https://github.com/home-assistant/operating-system/releases/download/5.8/hassos_ova-5.8.ova

agners, 2020-12-13:
Hm, I recently increased the image size to 32GB in #1009 and it seems I forgot to update the description file: https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/board/intel/ova/home-assistant.ovf#L8.

As a work around, you can fix it manually by editing the `ovf` file in the `ova` archive. E.g.
```
tar xf hassos_ova-5.8.ova
nano home-assistant.ovf
tar cf hassos_ova-5.8.ova
```

Meph79, 2020-12-13:
Just wanted to notify you of the issue. A friend tried to download it. I'm running 4.17 on RPi 3 already. 
After it failed for friend I downloaded it and tried it in my environment too..

JGNi, 2020-12-13:
> Hm, I recently increased the image size to 32GB in #1009 and it seems I forgot to update the description file: [...]

Shall line 8 from your link into the ova file in line 8 also?

agners, 2020-12-13:
> Shall line 8 from your link into the ova file in line 8 also?

So the number in line 8 is wrong, that is the capacity it used to be (6GB). It should be 34359738368 (32GB).

agners, 2020-12-15:
Will be fixed in 5.9 onwards.

closed 2020-12-15 by agners