home-assistant/operating-system #1086: 5.8: Raspberry Pi 4 with USB stick, stuck on downloading Docker image

ner091, 2020-12-14:
**Hardware Environment**
- Raspberry Pi 4, USB boot with flash drive (128 GB)

**Home Assistant OS release:**
- Fresh installation of release 5.8 (64bit) / hassos_rpi4-64-5.8.img.xz 

**logs (http://homeassistant.local:8123/observer/logs):**
...
20-12-14 09:19:29 INFO (SyncWorker_0) [supervisor.docker.homeassistant] Starting Home Assistant homeassistant/raspberrypi4-64-homeassistant with version landingpage
20-12-14 09:19:29 INFO (MainThread) [supervisor.hassos] Detect HassOS 5.8 / BootSlot A
20-12-14 09:19:29 WARNING (MainThread) [supervisor.jobs] 'GitRepo.clone' blocked from execution, not enough free space (0.2GB) left on the device
20-12-14 09:19:29 WARNING (MainThread) [supervisor.store] Skip update to later for core
20-12-14 09:19:30 WARNING (MainThread) [supervisor.jobs] 'GitRepo.clone' blocked from execution, not enough free space (0.2GB) left on the device
20-12-14 09:19:30 WARNING (MainThread) [supervisor.store] Skip update to later for a0d7b954
20-12-14 09:19:30 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-243' coro=<StoreManager.update_repositories.<locals>._add_repository() done, defined at /usr/src/supervisor/supervisor/store/__init__.py:76> exception=TypeError("create_issue() got an unexpected keyword argument 'refrence'")>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/store/__init__.py", line 83, in _add_repository
    await repository.load()
  File "/usr/src/supervisor/supervisor/store/repository.py", line 99, in load
    await self.git.load()
  File "/usr/src/supervisor/supervisor/store/git.py", line 49, in load
    await self.clone()
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 57, in wrapper
    raise self.on_condition()
supervisor.exceptions.StoreJobError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/store/__init__.py", line 88, in _add_repository
    self.sys_resolution.create_issue(
TypeError: create_issue() got an unexpected keyword argument 'refrence'
20-12-14 09:19:30 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-244' coro=<StoreManager.update_repositories.<locals>._add_repository() done, defined at /usr/src/supervisor/supervisor/store/__init__.py:76> exception=TypeError("create_issue() got an unexpected keyword argument 'refrence'")>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/store/__init__.py", line 83, in _add_repository
    await repository.load()
  File "/usr/src/supervisor/supervisor/store/repository.py", line 99, in load
    await self.git.load()
  File "/usr/src/supervisor/supervisor/store/git.py", line 49, in load
    await self.clone()
  File "/usr/src/supervisor/supervisor/jobs/decorator.py", line 57, in wrapper
    raise self.on_condition()
supervisor.exceptions.StoreJobError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/store/__init__.py", line 88, in _add_repository
    self.sys_resolution.create_issue(
TypeError: create_issue() got an unexpected keyword argument 'refrence'
...
20-12-14 09:47:25 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
20-12-14 09:47:55 INFO (SyncWorker_2) [supervisor.docker.interface] Updating image homeassistant/raspberrypi4-64-homeassistant:landingpage to homeassistant/raspberrypi4-64-homeassistant:2020.12.0
20-12-14 09:47:55 INFO (SyncWorker_2) [supervisor.docker.interface] Downloading docker image homeassistant/raspberrypi4-64-homeassistant with tag 2020.12.0.
20-12-14 09:49:00 ERROR (SyncWorker_2) [supervisor.docker.interface] Can't install homeassistant/raspberrypi4-64-homeassistant:2020.12.0 -> 404 Client Error for http+docker://localhost/v1.40/images/homeassistant/raspberrypi4-64-homeassistant:2020.12.0/json: Not Found ("no such image: homeassistant/raspberrypi4-64-homeassistant:2020.12.0: No such image: homeassistant/raspberrypi4-64-homeassistant:2020.12.0").
20-12-14 09:49:00 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
...

**Description of problem:**
Everytime I try to install Home Assistant OS on my USB drive, the preparation is stuck on downloading the docker image of Home Assistant.
It is very odd, that it works fine with a SD card (32GB) but not with the flash drive (128 GB).

agners, 2020-12-14:
Is `ha resolution info` saying something?

What did happen to me is that resizing the data partition on first boot did not work when still having the SD card in. Make sure to remove the SD card when using USB boot, and remove the USB drive when using SD card.

apeeters, 2020-12-14:
I had a similar issue during/after upgrading to 5.8: HA did not come up (no ping either). Only after removing the USB stick and power cycling the upgrade completed successfully.

ner091, 2020-12-14:
Didn't had access to ssh (without ui, I'm not aware, that it could be enabled). I've switched to the split config (ha on sd card, data on the usb drive). This seems to work, it is just really slow (at least the first start took over 30 minutes).

ner091, 2020-12-15:
> I had a similar issue during/after upgrading to 5.8: HA did not come up (no ping either). Only after removing the USB stick and power cycling the upgrade completed successfully.

I've tried to boot without the USB stick and after inserting it again, it still runs into the same error loop.

NikolayBorisov, 2020-12-15:
CURRENT: Wed Oct 28 17:32:40 UTC 2020 (1603906360)
LATEST: Fri Dec 11 11:15:17 UTC 2020 (1607685317)

Just updated bootloader to latest beta version. And boot 5.8 from USB SSD without unplug Zigbee Stick works now! Seems that updating to the latest version is not such a bad idea, even if everything worked before 5.8.

agners, 2020-12-15:
@ner091 the problem might be that Home Assistant OS gets confused which partitions to use if you have the SD card *and* the USB SSD drive with HAOS connected. Try only SD card, or only USB SSD drive.

ner091, 2020-12-15:
> @ner091 the problem might be that Home Assistant OS gets confused which partitions to use if you have the SD card _and_ the USB SSD drive with HAOS connected. Try only SD card, or only USB SSD drive.

Hi @agners, the sd card was not connected when I connected the USB drive was connected. I thought the boot order was SD card and only if there is none, the raspberry will boot from a usb drive.

ner091, 2020-12-15:
> CURRENT: Wed Oct 28 17:32:40 UTC 2020 ([1603906360](tel:1603906360))
> LATEST: Fri Dec 11 11:15:17 UTC 2020 ([1607685317](tel:1607685317))
> 
> Just updated bootloader to latest beta version. And boot 5.8 from USB SSD without unplug Zigbee Stick works now! Seems that updating to the latest version is not such a bad idea, even if everything worked before 5.8.

I‘ve installed the latest stable bootloader version yesterday (don‘t know the version, but if I remember correctly it was built in September).

agners, 2020-12-16:
@ner091 the problem would not be the boot order itself, but Linux which subsequently mounts some partitions from the SD card and some from the USB SSD. We mount partitions by its ID, and since we are using disk images the ID is the same when flashing a SD card or USB SSD drive. So that can lead to problems on boot.

ner091, 2020-12-18:
@agners That makes sense to me. But no, I never booted the RPI 4 with both SD card and USB drive connected.

Filialen, 2020-12-29:
Hi @ner091 is this still a problem for you or did you solve it?
I've got a brand new Homassistant Blue (Odroid N2+) with a eMMC and I have the same problem!

20-12-29 00:13:12 INFO (SyncWorker_0) [supervisor.docker.interface] Updating image homeassistant/odroid-n2-homeassistant:landingpage to homeassistant/odroid-n2-homeassistant:2020.12.1
20-12-29 00:13:12 INFO (SyncWorker_0) [supervisor.docker.interface] Downloading docker image homeassistant/odroid-n2-homeassistant with tag 2020.12.1.
20-12-29 00:13:22 ERROR (SyncWorker_0) [supervisor.docker.interface] Can't install homeassistant/odroid-n2-homeassistant:2020.12.1 -> 404 Client Error for http+docker://localhost/v1.40/images/homeassistant/odroid-n2-homeassistant:2020.12.1/json: Not Found ("no such image: homeassistant/odroid-n2-homeassistant:2020.12.1: No such image: homeassistant/odroid-n2-homeassistant:2020.12.1").
20-12-29 00:13:22 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec

ner091, 2020-12-29:
Hi @Filialen 
Last time I tried, this was still happening. Even with version 5.9.

ner091, 2021-01-13:
Hi all,
I tried again with an external HDD and there I didn't had any issues. (Image: hassos_rpi4-64-5.10.img.xz) 
Maybe it is a problem with flash drives? 
Mine is a Kingston DataTraveler Kyson 128GB, USB 3.2

closed 2022-01-11 by github-actions[bot]

edwoody, 2022-01-29:
I'm sure some people have success with home assistant.  I was hoping to.  I've loaded SD cards and waited and waited and waited.  I get that you press the blue button and see where it fails.  I really don't want to see where it fails.  I bought a new rpi4b so I could see it succeed.  I'm getting a bad feeling about this.  BTW, issue still happening, not stale