home-assistant/operating-system #1087: DHT-22 sensor not working on PI 4 with Home Assistant OS 5.8

HedgehogCode, 2020-12-14:
**Hardware Environment**
- Raspberry Pi 4

**Home Assistant OS release:** 5.8
- Updated from version 4.16
- System health information at the end

**Description of problem:**
The DHT-22 sensor connected to the GPIO pins fails to update with the error:
```
2020-12-14 11:01:09 ERROR (MainThread) [homeassistant.components.sensor] dht: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 359, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 466, in async_device_update
    await task
  File "/usr/local/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/components/dht/sensor.py", line 137, in update
    self.dht_client.update()
  File "/usr/src/homeassistant/homeassistant/util/__init__.py", line 239, in wrapper
    result = method(*args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/dht/sensor.py", line 173, in update
    humidity, temperature = self.adafruit_dht.read_retry(self.sensor, self.pin)
  File "/usr/local/lib/python3.8/site-packages/Adafruit_DHT/common.py", line 94, in read_retry
    humidity, temperature = read(sensor, pin, platform)
  File "/usr/local/lib/python3.8/site-packages/Adafruit_DHT/common.py", line 80, in read
    platform = get_platform()
  File "/usr/local/lib/python3.8/site-packages/Adafruit_DHT/common.py", line 63, in get_platform
    raise RuntimeError('Unknown platform.')
RuntimeError: Unknown platform.
```
This has been working for me with Home Assistant OS 4.16 and 4.20. If I only downgrade Home Assistant OS to 4.20 the sensor starts working again (everything else unchanged). The problem is reproducible. Reboots do not help and updating again from 4.20 causes the same issue.

This is probably related to https://github.com/home-assistant/operating-system/issues/720 where the same issue occurred with 4.9 and was fixed with 4.10.

## System Health

version | 2020.12.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.6
os_name | Linux
os_version | 5.4.79-v7l
arch | armv7l
timezone | Europe/Berlin

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.8
-- | --
update_channel | stable
supervisor_version | 2020.12.6
docker_version | 19.03.13
disk_total | 28.6 GB
disk_used | 5.0 GB
healthy | true
supported | true
board | rpi4
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (8.10.0), Check Home Assistant configuration (3.6.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
mode | storage
views | 2
resources | 1

</details>

agners, 2020-12-14:
Hm, it seems the Adafruit_DHT library used in this component has problems with the new format of `/proc/cpuinfo`.

The problem is the new Kernel 

/proc/cpuinfo

HedgehogCode, 2020-12-14:
I see.

Changing the library has already been proposed here: https://github.com/home-assistant/core/issues/37382.
If this is the way to go feel free to close this issue and continue the discussion there.

closed 2020-12-15 by agners