home-assistant/operating-system #1090: PI4-B USB SSD boot fails after updating above OS 5.4

caro7372, 2020-12-14:
**Hardware Environment**

- [X] Raspberry Pi [4]

**Home Assistant OS release:**
- [X] Fresh installation of release above 5.4
- [X] Updated from version 5.4 to 5.8
- Additional information (if accessible):

System Health

version: 2020.12.0
installation_type: Home Assistant OS
dev: false
hassio: true
docker: true
virtualenv: false
python_version: 3.8.6
os_name: Linux
os_version: 4.19.127-v8
arch: aarch64
timezone: UTC

logged_in: false
can_reach_cert_server: ok
can_reach_cloud_auth: ok
can_reach_cloud: ok

host_os: HassOS 5.4
update_channel: stable
supervisor_version: 2020.12.6
docker_version: 19.03.12
disk_total: 109.3 GB
disk_used: 1.9 GB
healthy: true
supported: true
board: rpi4-64
supervisor_api: ok
version_api: ok
installed_addons: 

dashboards: 1
mode: auto-gen
resources: 0

**Description of problem:**

carras, 2020-12-14:
https://github.com/raspberrypi/rpi-eeprom/issues/262#issuecomment-743382789

There is a problem, probably, with your SSD enclosure and the EEPROM of the RasPi4. Check the issue on the link.

What's the controller of  your enclosure? Mine is Jmicron JMS580.

Also there is a similar issue in:
https://github.com/home-assistant/operating-system/issues/1079

carras, 2020-12-14:
Actually, this is one more important, also about the same: https://github.com/home-assistant/operating-system/issues/1045

Btw, if you turn on the raspi4 then you plug the usb ssd, it'll probably boot up.

caro7372, 2020-12-14:
Indeed it does boot up then.
Tried different versions above 5.4 with all different versions of boot loader as mentioned in the other threads.
Didn't see those threads so will look into it tomorrow.
Adding the boot order line maybe is a solution?

carras, 2020-12-14:
> Indeed it does boot up then.
> Tried different versions above 5.4 with all different versions of boot loader as mentioned in the other threads.
> Didn't see those threads so will look into it tomorrow.
> Adding the boot order line maybe is a solution?

Don't try other images, it's not HassOS is the Raspi4 new EEPROM firmware. If you try a different enclosure It will work.
https://github.com/raspberrypi/rpi-eeprom
We should open an issue there.

What's your enclosure?

caro7372, 2020-12-14:
Kingspec Externe Ssd Harde Schijf

[https://nl.aliexpress.com/item/4001278381663.html?spm=a2g0s.9042311.0.0.6e554c4dR2wUj4](url)

carras, 2020-12-14:
> Kingspec Externe Ssd Harde Schijf
> 
> [https://nl.aliexpress.com/item/4001278381663.html?spm=a2g0s.9042311.0.0.6e554c4dR2wUj4](url)

Whenever you have time, figure out what's the controller chip. I'd like to know if it's a JMS580 or another controller from Jmicron.

caro7372, 2020-12-14:
I asked the reseller if they know the chipset for the controller.
As soon as I have an answer I will let you know.

carras, 2020-12-14:
> I asked the reseller if they know the chipset for the controller.
> As soon as I have an answer I will let you know.

You can maybe check it reading the label if you have time to open de case.

caro7372, 2020-12-14:
If I know how to open without damage it.

carras, 2020-12-14:
I though It was easy to see. In your computer in hardware info you can see it too.

I've open a issue about those controller, please join me: raspberrypi/rpi-eeprom#266

caro7372, 2020-12-14:
I will check tomorrow. But I use Mac and am not sure if I can see what you ask for.
Otherwise I have to find it on raspbian.

carras, 2020-12-14:
> I will check tomorrow. But I use Mac and am not sure if I can see what you ask for.
> Otherwise I have to find it on raspbian.

If you connect it, go to about this Mac, system report and in there look for the USB and the attacted enclosure and you will find something like this:

Product ID:	0xab24
  Vendor ID:	0x0bc2  (Seagate LLC)
  Version:	1.00
  Serial Number:	NA9NJGGS
  Speed:	Up to 5 Gb/s
  Manufacturer:	Seagate
  Location ID:	0x14900000 / 2
  Current Available (mA):	900
  Current Required (mA):	144
  Extra Operating Current (mA):	0
  Media:

Copy it here and we will investigate.

MickPBduece, 2020-12-15:
I just rolled mine back to a backup after spending 3 afternoons thinking I did something wrong.  I sure know more about Docker and Linux now though.  My situation was the Pi would boot and the HA would launch at the end of the install (mmmm most times) but not always.  What would ALWAYS happen is a reboot from Supervisor or from cmd line would result in HA not starting.  I saw some auth errors on pi user having access to Docker but even with that resolved HA did not start correctly.

caro7372, 2020-12-15:
USB 3.1-bus:

  Besturingsbestand hostcontroller:	AppleUSBXHCIAR
  PCI-apparaatcode:	0x15d4 
  PCI-revisiecode:	0x0002 
  PCI-fabrikantcode:	0x8086 
  Busnummer:	0x01 

KingSpec Z3 Plus-120:

  Productcode:	0x0578
  Fabrikantcode:	0x0578
  Versie:	12.14
  Serienummer:	DD564198838738C19C
  Snelheid:	Maximaal 5 Gb/s
  Fabrikant:	JMicron
  Locatiecode:	0x01200000 / 1
  Beschikbare stroom (mA):	900
  Vereiste stroom (mA):	896
  Extra stroom in werking (mA):	0
  Media:

carras, 2020-12-15:
> USB 3.1-bus:
>
>
>
>   Besturingsbestand hostcontroller:	AppleUSBXHCIAR
>
>   PCI-apparaatcode:	0x15d4
>
>   PCI-revisiecode:	0x0002
>
>   PCI-fabrikantcode:	0x8086
>
>   Busnummer:	0x01
>
>
>
> KingSpec Z3 [...]

As I was expecting, you have Jmicron controller too. Your is JMS578, we can add that to the issue with EEPROM too, it's all about those chips.

carras, 2020-12-15:
> I just rolled mine back to a backup after spending 3 afternoons thinking I did something wrong.  I sure know more about Docker and Linux now though.  My situation was the Pi would boot and the HA [...]

What did you rollback?

agners, 2020-12-15:
This indeed seems like the same issue as #1045. [This comment](https://github.com/home-assistant/operating-system/issues/1045#issuecomment-738461340) seems to also mention the same drive. Closing this issue as it is essentially a duplicate of #1045.

MickPBduece, 2020-12-15:
Each month I make a copy of the SSD to a SD card.  I restarted the Pi on
the backup card w/o the SSD and then copied the SD card over the SSD --
Rolled everything back. The SD card copier in the Pi menu works well.  I do
stop HA when I make the copy

I also make a snapshot file and a copy of the yaml files each time I do a
HA version upgrade so I can reapply that to get even closer.  Eventually I
will enable the backup to GIT for the files but -- another day.

Do you know how to dump the info about the SSD card - I'd like to share
mine but not a Unix guru at all

agners, 2020-12-15:
@MickPBduece in a bootet environment, on the OS shell (root -> login) use `dmesg | grep -e sda -e scsi` when the USB SSD is inserted. This should get some high level information.

closed 2020-12-15 by agners