home-assistant/operating-system #1091: RPi4 - not upgrading to 5.8 from 4.17

nickidw, 2020-12-15:
**Hardware Environment**

- [x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 4.17
- Additional information (if accessible):

~ $ ha info
arch: armv7
channel: stable
docker: 19.03.12
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "4.17"
homeassistant: 2020.12.0
hostname: homeassistant
logging: info
machine: raspberrypi4
operating_system: HassOS 4.17
supervisor: 2020.12.6
supported: true
supported_arch:
- armv7
- armhf
timezone: Africa/Johannesburg

**Supervisor logs:**

20-12-14 14:57:13 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/5.8/hassos_rpi4-5.8.raucb
20-12-14 14:59:46 INFO (MainThread) [supervisor.hassos] OTA update is downloaded on /data/tmp/hassos-5.8.raucb
20-12-14 14:59:46 INFO (MainThread) [supervisor.utils.gdbus] Starting dbus monitor on de.pengutronix.rauc
~ $ ha supervisor log
{"result": "error", "message": "System is not ready with state: shutdown"}
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing...
[15:05:39] INFO: Update udev information
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[15:05:39] INFO: Starting local supervisor watchdog...
20-12-14 15:05:42 INFO (MainThread) [__main__] Initializing Supervisor setup
20-12-14 15:05:42 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
20-12-14 15:05:42 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4
20-12-14 15:05:42 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/armv7-hassio-supervisor with version 2020.12.6
20-12-14 15:05:42 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
20-12-14 15:05:43 INFO (SyncWorker_0) [supervisor.docker.interface] Cleanup images: ['homeassistant/armv7-hassio-supervisor:2020.11.0']
20-12-14 15:05:44 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
20-12-14 15:05:44 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
20-12-14 15:05:44 INFO (MainThread) [__main__] Setting up Supervisor
20-12-14 15:05:44 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
20-12-14 15:05:45 INFO (MainThread) [supervisor.host.info] Updating local host information
20-12-14 15:05:45 INFO (MainThread) [supervisor.host.services] Updating service information
20-12-14 15:05:46 INFO (MainThread) [supervisor.host.network] Updating local network information
20-12-14 15:05:47 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
[... 834 lines cut ...]
[ 1622.722659] systemd[1]: Started Journal Service.
[ 1841.113040] INFO: task kworker/0:2:77 blocked for more than 120 seconds.
[ 1841.113076]       Tainted: G         C        4.19.127-v7l #1
[ 1841.113101] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1841.113131] kworker/0:2     D    0    77      2 0x00000000
[ 1841.113176] Workqueue: events_freezable mmc_rescan
[ 1841.113222] [<c0af7a34>] (__schedule) from [<c0af81bc>] (schedule+0x50/0xa8)
[ 1841.113257] [<c0af81bc>] (schedule) from [<c08baa74>] (__mmc_claim_host+0xe4/0x238)
[ 1841.113293] [<c08baa74>] (__mmc_claim_host) from [<c08bac00>] (mmc_get_card+0x38/0x3c)
[ 1841.113331] [<c08bac00>] (mmc_get_card) from [<c08c3c5c>] (mmc_sd_detect+0x24/0x7c)
[ 1841.113370] [<c08c3c5c>] (mmc_sd_detect) from [<c08bd0b0>] (mmc_rescan+0xe8/0x3f8)
[ 1841.113409] [<c08bd0b0>] (mmc_rescan) from [<c023e170>] (process_one_work+0x200/0x4e0)
[ 1841.113448] [<c023e170>] (process_one_work) from [<c023f23c>] (worker_thread+0x4c/0x568)
[ 1841.113482] [<c023f23c>] (worker_thread) from [<c0244d30>] (kthread+0x140/0x170)
[ 1841.113516] [<c0244d30>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[ 1841.113545] Exception stack(0xecbbffb0 to 0xecbbfff8)
[ 1841.113570] ffa0:                                     00000000 00000000 00000000 00000000
[ 1841.113601] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 1841.113630] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 2066.123279] udevd[190]: starting version 3.2.9
[ 2066.130707] udevd[191]: starting eudev-3.2.9

**Description of problem:**

I upgraded HA core to 2020.12.0 before. Upgrade of HA OS from 4.17 to 5.8 takes a long time, and afterwards it's still on 4.17 and HA core is reverted to 0.118.4.

agners, 2020-12-15:
You are booting from a SD card I assume? It seems that some tasks take a very long time to complete SD card operations. It sounds like the SD card is very slow or maybe even failing...?

nickidw, 2020-12-15:
I think the system crashed after the 1st reboot post upgrade, and that caused issues. I just rebooted again, here is that log. I don't see the same errors.

```~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.127-v7l (builder@3d4ba40677c6) (gcc version 9.3.0 (Buildroot -g076f8ad8)) #1 SMP Wed Nov 18 05:07:31 UTC 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 64 MiB at 0x0000000037400000
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4f4 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s36940 r8192 d24500 u69632
[    0.000000] pcpu-alloc: s36940 r8192 d24500 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=3e76c1cb597147ec94ffa5821e123017 cgroup_enable=memory coherent_pool=1M 8250.nr_uarts=0 cma=64M bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:40:02:05 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 3930556K/4050944K available (10240K kernel code, 736K rwdata, 2716K rodata, 2048K init, 877K bss, 54852K reserved, 65536K cma-reserved, 3198976K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (12256 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 737 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 878 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 32798 entries in 97 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000021] Switching to timer-based delay loop, resolution 18ns
[    0.000257] Console: colour dummy device 80x30
[    0.000732] console [tty1] enabled
[    0.000788] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000824] pid_max: default: 32768 minimum: 301
[    0.000966] Security Framework initialized
[    0.001054] AppArmor: AppArmor initialized
[    0.001248] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001280] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002074] CPU: Testing write buffer coherency: ok
[    0.002499] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003157] Setting up static identity map for 0x200000 - 0x20003c
[    0.003335] rcu: Hierarchical SRCU implementation.
[    0.004218] smp: Bringing up secondary CPUs ...
[    0.005096] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006071] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007007] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007127] smp: Brought up 1 node, 4 CPUs
[    0.007195] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.007218] CPU: All CPU(s) started in HYP mode.
[    0.007238] CPU: Virtualization extensions available.
[    0.008041] devtmpfs: initialized
[    0.019063] VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
[    0.019297] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.019339] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.021276] pinctrl core: initialized pinctrl subsystem
[... 557 lines cut ...]
[  249.050369] hassio: port 8(veth2e110a9) entered blocking state
[  249.050381] hassio: port 8(veth2e110a9) entered forwarding state
[  255.446612] udevd[193]: starting version 3.2.9
[  255.481779] udevd[194]: starting eudev-3.2.9
[  308.033297] hassio: port 9(veth4b1515c) entered blocking state
[... cut ...] ned key=(null)
[  308.577930] audit: type=1327 audit(1608019844.437:239): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D4C002D6E
[  308.591473] audit: type=1325 audit(1608019844.457:240): table=nat family=2 entries=5
[  308.732716] eth0: renamed from veth3013536
[  308.792853] IPv6: ADDRCONF(NETDEV_CHANGE): veth4b1515c: link becomes ready
[  308.793030] hassio: port 9(veth4b1515c) entered blocking state
[  308.793060] hassio: port 9(veth4b1515c) entered forwarding state
~ $
```

stefsmeets, 2020-12-15:
> **Description of problem:**
> 
> I upgraded HA core to 2020.12.0 before. Upgrade of HA OS from 4.17 to 5.8 takes a long time, and afterwards it's still on 4.17 and HA core is reverted to 0.118.4.

I'm seeing a similar issue on my pi4, upgrading from 4.16 to 5.8. The Update does _something_, but afterwards I'm back on 4.16.

The most interesting thing is that it seems to 'reset' to a previous previous state somehow. Snapshots and unused addons that I deleted to make space re-appear as well.

iondarie, 2020-12-15:
I tried to update from 4.17 to 5.8 now it seems i need to reflash, HA is no longer booting, not reachable by smb...

agners, 2020-12-15:
> I think the system crashed after the 1st reboot post upgrade, and that caused issues. I just rebooted again, here is that log. I don't see the same errors.

@nickidw if it crashed, then the rollback is expected: The system has recovery capabilities in case of a faulty upgrade. The question of course is why it crashed....

What happens if you retry the upgrade? Make sure to create and download a snapshot first.

agners, 2020-12-15:
> The most interesting thing is that it seems to 'reset' to a previous previous state somehow. Snapshots and unused addons that I deleted to make space re-appear as well.

@stefsmeets the OS has recovery capabilities, but this does not affect addons etc... Are you using SD card to boot?

iondarie, 2020-12-15:
ok so in my case i found the problem, it will not boot if i have anything connected to the USB ports on my RPI4

nickidw, 2020-12-15:
> @nickidw if it crashed, then the rollback is expected: The system has recovery capabilities in case of a faulty upgrade. The question of course is why it crashed....
>
> What happens if you retry the [...]

I have done it three or for times, no change and I could not see anything in the logs. Only other option could be to add a monitor to it, does it log to console while upgrading?

agners, 2020-12-15:
@iondarie what do you have connected normally?

@nickidw you should see the reboot happening along with logs during shutdown and bootup, so probably gives some information

iondarie, 2020-12-15:
@angers i only have my APC ES550 UPS, witch i was monitoring with apcupsd. I even wiped the sd card and did a fresh 5.8 install. As soon as i connect the usb the device no longer boots.

iondarie, 2020-12-16:
update: it seems the problem lies with the rpi firmware, it messed up the USB and in my experience is doing something that also affects GPIO's.

My gates are controlled via GPIO and they work fine, but as soon as i plug in a usb all hell break's loose. I got 50 notifications from the gate open/closed and it keeps going until i pull the USB out.

stefsmeets, 2020-12-16:
> > The most interesting thing is that it seems to 'reset' to a previous previous state somehow. Snapshots and unused addons that I deleted to make space re-appear as well.
> 
> @stefsmeets the OS has recovery capabilities, but this does not affect addons etc... Are you using SD card to boot?

Yeah, I suspect some file corruption somewhere, but HA continues to function as it should, it is the core and os updates, and installing addons that gives problems. I am planning to do a fresh install in the weekend.

It is difficult for me to debug what is going on, because the log resets after the OS seemingly restarts. I'm seeing multiple failures, from git `fatal: index file corrupt` to `[supervisor.docker.interface] Can't install homeassistant/raspberrypi4-64-homeassistant:2020.12.0 -> 404 Client Error for http+docker://localhost/v1.40/images/homeassistant/raspberrypi4-64-homeassistant:2020.12.0/json: Not Found ("no such image: homeassistant/raspberrypi4-64-homeassistant:2020.12.0: No such image: homeassistant/raspberrypi4-64-homeassistant:2020.12.0").` (similar to #1086).

The addons/snapshots re-appearing is completely reproducible here, I have already seen it a couple of times on different attempts.

nickidw, 2020-12-16:
> update: it seems the problem lies with the rpi firmware, it messed up the USB and in my experience is doing something that also affects GPIO's.
>
> My gates are controlled via GPIO and they work fine, [...]

Does this new release install updated RPI firmware?

agners, 2020-12-17:
> Does this new release install updated RPI firmware?

Yes, with 5.8 we ship a newer RPi 4 EEPROM firmware. When booting from a USB flash drive this is being installed automatically (not so when booting from SD card). See https://github.com/home-assistant/operating-system/pull/939.

But it seems to cause too many issues, so we will remove this again, at least for now: https://github.com/home-assistant/operating-system/pull/1103

stefsmeets, 2020-12-18:
I just wanted to get back and say that my issue is probably because of the SD card being corrupted. I was not even able to reflash HA anymore!

closed 2021-01-07 by agners