home-assistant/operating-system #1093: Laptop running NUC image goes into suspend with lid closed.

Pecillo, 2020-12-15:
**Hardware Environment**

- [*] Intel NUC (or compatible) in this case a Lenovo Laptop

**Home Assistant OS release:**
- [* ] Fresh installation of release x.y
- [*] Updated from version x.y

**Supervisor logs:**

n/a

**Journal logs:**

n/a

**Kernel logs:**

n/a

**Description of problem:**
- Is the problem reproducible? - yes

Laptop goes into suspend with lid closed. Apparently more users face this issue when running HAOS on a laptop based on a discussion on Discord.

Fix:
Change these lines in "/etc/systemd/logind.conf":

#HandleLidSwitch=suspend
#HandleLidSwitchExternalPower=suspend
#HandleLidSwitchDocked=ignore

to this:

HandleLidSwitch=ignore
HandleLidSwitchExternalPower=ignore
HandleLidSwitchDocked=ignore

MiG-41, 2020-12-16:
Hi , can you give me a hint , how to edit this file ? With File Editor o'm not able to go out other place then /config directory ( tried to find something in config but no succes).
Tried to run "docker exec -it homeassistant /bin/bash" and then edit this file , but in /etc there is no directory /systemd ....

I'm using laptop and just update it , and such problem apear...

Pecillo, 2020-12-16:
> Hi , can you give me a hint , how to edit this file ? With File Editor o'm not able to go out other place then /config directory ( tried to find something in config but no succes).
> Tried to run [...]

The file is located on a read-only filesystem. Editing it is not something you do just like that.

It's in a squashfs filesystem, so you would have to unpack it, edit the file and then repack it. Based on the discussion on Discord yesterday this fix will likely be included in the next release.

I added the fix to this ticket for the maintainers and I will wait for their next release since I don't want to put in the effort of editing it myself.

/Adam

MiG-41, 2020-12-16:
Ok , so hopfully in next realise this will be back to normal. 
Till that time i have to live with LCD permanently turned on...

MiG-41, 2020-12-19:
Updated OS to 5.9 , and back to normal :)
Lid can be closed again

ChipMIK1, 2020-12-19:
Hmm...Any hint on how to upgrade to 5.9 on NUC-image running straight on computer?

gerritjandebruin, 2020-12-19:
I was surprised that it did not show up as an update. I ssh'ed into the machine and went for `ha os update --version 5.9`.

ChipMIK1, 2020-12-19:
TY very much, 5.9 now running....Case AND lid closed 🥇 👍

agners, 2020-12-21:
@gerritjandebruin 5.9 was still on the beta channel that is why it did not show up yet. We will push it to stable shortly.

Thanks for the feedback!

closed 2020-12-21 by agners