home-assistant/operating-system #1096: Hass OS 5.8 unstable crashes immediately or after a few minutes

liudger, 2020-12-16:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

```
$ ha info
arch: aarch64
channel: stable
docker: 19.03.13
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.8"
homeassistant: 2020.12.0
hostname: homeassistant
logging: info
machine: odroid-n2
operating_system: Home Assistant OS 5.8
supervisor: 2020.12.6
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: Europe/Amsterdam
```

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

```
$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.9.14 (builder@171c99eb2197) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g6d5c41c) 9.3.0, GNU ld (GNU Binutils) 2.34) #1 SMP PREEMPT Sun Dec 13 12:51:03 UTC 2020
[    0.000000] Machine model: Hardkernel ODROID-N2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x00000000e0c00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000000f4e5afff]
[    0.000000] NUMA: NODE_DATA [mem 0xf465f100-0xf4660fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000f4e5afff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000004ffffff]
[    0.000000]   node   0: [mem 0x0000000005300000-0x00000000f4e5afff]
[    0.000000] Zeroed struct page in unavailable ranges: 421 pages
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000f4e5afff]
[    0.000000] On node 0 totalpages: 1002331
[    0.000000]   DMA zone: 4096 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[... 611 lines cut ...]
[   69.200112] hassio: port 13(veth3ba4e88) entered blocking state
[   69.205758] hassio: port 13(veth3ba4e88) entered forwarding state
[   74.347693] kauditd_printk_skb: 26 callbacks suppressed
[   74.347697] audit: type=1334 audit(1608103164.598:244): prog-id=11 op=LOAD
[   74.358166] audit: type=1334 audit(1608103164.598:245): prog-id=12 op=LOAD
[   83.474813] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[   83.478677] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[   83.628666] audit: type=1325 audit(1608103173.878:246): table=filter family=2 entries=4 op=xt_replace pid=11379 subj==docker-default (enforce) comm="iptables"
[   83.640658] audit: type=1325 audit(1608103173.878:247): table=filter family=2 entries=5 op=xt_replace pid=11380 subj==docker-default (enforce) comm="iptables"
[   83.655686] audit: type=1325 audit(1608103173.878:248): table=nat family=2 entries=15 op=xt_replace pid=11381 subj==docker-default (enforce) comm="iptables"
[  104.677815] audit: type=1334 audit(1608103194.930:249): prog-id=12 op=UNLOAD
[  104.681030] audit: type=1334 audit(1608103194.930:250): prog-id=11 op=UNLOAD
```

**Description of problem:**

I got a message that the supervisor died
```
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
s6-svwait: fatal: supervisor died
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
```

liudger, 2020-12-16:
I can't get any info where the culprit lies when the system gets unresponsive. Any feedback is welcome for places to look. The only thing I see is that the supervisor ram usage it getting quite high

agners, 2020-12-16:
Is the Supervisor log showing anything unusual when its running?

pvizeli, 2020-12-16:
`s6-svwait: fatal: supervisor died` that is the supervisor of s6

liudger, 2020-12-16:
The supervisor is only showing info logs no error or other logs.

pvizeli, 2020-12-16:
journald -u hassos-supervisor

liudger, 2020-12-16:
`journald -u hassos-supervisor` where do I run this command? Not in the terminal add-on I guess

liudger, 2020-12-17:
It was running for quite long but next day the system was down again. (unresponsive)

supervisor log:

```20-12-17 07:40:08 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
20-12-17 07:40:08 INFO (MainThread) [supervisor.host] Host information reload completed
20-12-17 07:40:08 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
20-12-17 07:40:08 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
20-12-17 07:40:09 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-dns with version 2020.11.0
20-12-17 07:40:09 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
20-12-17 07:40:09 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_dns application
20-12-17 07:40:09 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS homeassistant/aarch64-hassio-dns with version 2020.11.0 - 172.30.32.3
20-12-17 07:40:09 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
20-12-17 07:40:09 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-audio with version 17
20-12-17 07:40:10 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
20-12-17 07:40:10 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_audio application
20-12-17 07:40:10 INFO (SyncWorker_0) [supervisor.docker.audio] Starting Audio homeassistant/aarch64-hassio-audio with version 17 - 172.30.32.4
20-12-17 07:40:10 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-cli with version 2020.11.1
20-12-17 07:40:10 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
20-12-17 07:40:10 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_cli application
20-12-17 07:40:11 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI homeassistant/aarch64-hassio-cli with version 2020.11.1 - 172.30.32.5
20-12-17 07:40:11 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-observer with version 2020.10.1
20-12-17 07:40:11 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-multicast with version 3
20-12-17 07:40:11 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
20-12-17 07:40:11 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_multicast application
20-12-17 07:40:12 INFO (SyncWorker_0) [supervisor.docker.multicast] Starting Multicast homeassistant/aarch64-hassio-multicast with version 3 - Host
20-12-17 07:40:12 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
20-12-17 07:40:12 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/odroid-n2-homeassistant with version 2020.12.0
20-12-17 07:40:12 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 7 Home Assistant secrets
20-12-17 07:40:12 INFO (MainThread) [supervisor.hassos] Detect HassOS 5.8 / BootSlot B
20-12-17 07:40:13 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
20-12-17 07:40:13 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
20-12-17 07:40:13 INFO (MainThread) [supervisor.store] Loading add-ons from store: 66 all - 66 new - 0 remove
20-12-17 07:40:13 INFO (MainThread) [supervisor.addons] Found 14 installed add-ons
20-12-17 07:40:13 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-mariadb with version 2.2.1
20-12-17 07:40:13 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to hassioaddons/adguard-aarch64 with version 2.6.1
20-12-17 07:40:13 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-configurator with version 5.2.0
20-12-17 07:40:13 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-samba with version 9.3.0
20-12-17 07:40:13 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-ssh with version 8.10.0
20-12-17 07:40:13 INFO (SyncWorker_9) [supervisor.docker.interface] Attaching to hassioaddons/bitwarden-aarch64 with version v0.6.2
20-12-17 07:40:13 INFO (SyncWorker_7) [supervisor.docker.interface] Attaching to hassioaddons/wireguard-aarch64 with version 0.4.0
20-12-17 07:40:13 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-deconz with version 6.6.1
20-12-17 07:40:13 INFO (SyncWorker_8) [supervisor.docker.interface] Attaching to hassioaddons/vscode-aarch64 with version 2.9.1
20-12-17 07:40:13 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to hassioaddons/glances-aarch64 with version 0.9.1
20-12-17 07:40:13 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to hassioaddons/phpmyadmin-aarch64 with version 0.1.4
20-12-17 07:40:13 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-check_config with version 3.6.0
20-12-17 07:40:13 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to hassioaddons/grafana-aarch64 with version 5.3.6
20-12-17 07:40:13 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to hassioaddons/nginxproxymanager-aarch64 with version 0.8.2
20-12-17 07:40:14 INFO (MainThread) [supervisor.snapshots] Found 2 snapshot files
[... cut ...]
20-12-17 07:48:11 INFO (MainThread) [supervisor.core] Supervisor is up and running
20-12-17 07:48:11 INFO (MainThread) [supervisor.host.info] Updating local host information
20-12-17 07:48:12 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
20-12-17 07:48:12 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
20-12-17 07:48:12 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete```

liudger, 2020-12-17:
What was the command to install another version of hassos?

agners, 2020-12-17:
`ha os update --version=x.y`

liudger, 2020-12-18:
from an issue https://github.com/home-assistant/core/issues/44324
A screenshot
[image]

agners, 2020-12-19:
@liudger is that booting from a SD card or eMMC?

liudger, 2020-12-19:
This was not mine screen. I don't have a hdmi screen around for my odroid. So this could be a different problem. But the tread is was posted has about the same issues I  have. 

I am now downgraded to ha os 4.20 and my system is stable again. I tested also with 5.9 but still unstable.

The screen was taken from an pi3:
> Yes, Pi 3, and I'm only usind an SD Card. Some people on Reddit suggested that my SD card might be worn out.

TuomasPakkanen, 2020-12-19:
I have the unstability issues on a pi4, posted some info here:
https://github.com/home-assistant/core/issues/44324

It definitely stays running for more than a few minutes, sometimes hours, sometimes few days but then it completely freezes. How do you downgrade to 4.20 and does that version support USB boot on pi4? I used to run 5.2 beta and it was all fine.

liudger, 2020-12-19:
`ha os update --version=5.2`

go back to the beta I would suggest for now

HumanSkunk, 2020-12-21:
Same issue. I will do the same and hope for the best. I was trying to identify what the issue is as i thought it was something in the config. At least i am not alone with this. Fingers crossed it gets resolved.

liudger, 2020-12-21:
It looks like it's in the firmware shipped with these new versions

HumanSkunk, 2020-12-21:
> It looks like it's in the firmware shipped with these new versions

You had any issues recently? I realised i had a full backup i made about a week ago prior to me updating so i have just rolled back to that for now.

agners, 2020-12-30:
> It looks like it's in the firmware shipped with these new versions

We did update the firmware for DDR in 5.5. However, I haven't seen problems on N2 here. Can you try if the problem you are seeing start between 5.4 and 5.5?

@HumanSkunk do you have an ODROID N2 as well? Is your kernel log (Host log in Supervisor frontend) also showing "Unable to handle kernel execute from non-executable memory at virtual address" as posted in this issue?

HumanSkunk, 2020-12-30:
I have a Pi 4 running off an attached SSD. I have slowly worked my way backwards through the releases and settled back in the version I used to build the ssysten when I migrated to the SSD. Running 5.3.

agners, 2020-12-31:
@HumanSkunk this issue is ODROID N2 related. What you are experience is more likely related to #1090

HumanSkunk, 2020-12-31:
My issue wasn’t/isn’t boot related. My issue occurred after a random time period, could be a few hours or a few days where the whole system would just hang, the heatsink would get super hot and it would kill my network connectivity. I see the link to the issue has been closed shy yourself as a duplicate of another issue also closed but no resolution provided. Neither of them seem to be tied to what I had been experiencing. My instance was fine on boot and worked perfectly, it just all shat itself over the next few days. The one closest to what I am seeing is #1119

bschatzow, 2021-03-11:
Not sure why this is closed as it was never fixed.

wodka, 2021-05-12:
and here we go again... still open

bschatzow, 2021-05-12:
Still open for me also

liudger, 2021-05-15:
Did anyone test the new version 6 release candidate?
https://github.com/home-assistant/operating-system/releases/tag/6.0.rc1

goncalvespedro, 2021-05-18:
Still open for me too

agners, 2021-07-28:
@liudger do you still experience this issue? Which version are you using currently?

liudger, 2021-07-28:
Hi @agners I am still using version 4.20 which is stable. Lately I didn't try to test new version of the OS anymore. It was till 5.8 that I did try to test. But others had issues with 6.1 so I didn't bother. Do you think it's a try worth? 
I'll make a full snapshot and have etcher ready to downgrade back to 4.20
It's quite some work to get the  eMMC chip out of de device and in the reader. 

Let me know what you think the change of succes is and if there is a way it could work.

agners, 2021-07-29:
I wonder if #1471 is experience the same issue.

HAOS 6.2 comes with the latest LTS mainline kernel 5.10. That would probably be a good release to test.

liudger, 2021-07-30:
so 6.2 is still a no go I see, but we now know it's really in the kernel and only affecting Odroid N2
https://github.com/home-assistant/operating-system/issues/1471

agners, 2021-10-03:
@liudger can you connect a HDMI cable to see if the screen shows anything interesting when the crash happens?

From the kernel (host) logs, this sounds rather scary:

```
[   44.351523] Unable to handle kernel execute from non-executable memory at virtual address ffff00008dda6580
[   44.355811] Mem abort info:
[   44.358313]   ESR = 0x8600000f
[   44.361309]   EC = 0x21: IABT (current EL), IL = 32 bits
[   44.366596]   SET = 0, FnV = 0
[   44.369598]   EA = 0, S1PTW = 0
```

I haven't seen that on other devices. It seems as if that device experience bit flips or similar during when reading memory.

> so 6.2 is still a no go I see, but we now know it's really in the kernel and only affecting Odroid N2 #1471

The issue @marcosgmeli has seems to be memory related too, so quite likely that your issues relate.

liudger, 2021-10-04:
> @liudger can you connect a HDMI cable to see if the screen shows anything interesting when the crash happens?
>
> From the kernel (host) logs, this sounds rather scary:
>
> ```
> [   44.351523] Unable to [...]

I'll try a new test. I make a backup raspberry so my house is still running. 

It's hackoctober so we should be hacking!

liudger, 2021-12-15:
I tried 6.6 but still a crash after about 12 hours. I'll fire my old raspberry up and install everything on this device so I can start testing more on the n2 with a monitor.

agners, 2021-12-15:
What would be interesting is to find the first version which introduced the instability (it seems you did test 5.2 at one point according to https://github.com/home-assistant/operating-system/issues/1096#issuecomment-748473622?)

liudger, 2021-12-16:
My stable version was 4.20 That was the version I stayed on for ages. But I could start testing. I start with os 7.0 today and see when it crashes. I'll have a clean install with nothing added on my N2. 

ps. I am using now a raspberry 3 to run my home. (it's so slow)

liudger, 2022-02-05:
I am now at 7.2 I'll switch back to older version as this version also crashes. It looks indeed something with memory related. 
```
 0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.91 (builder@dd7119443ce4) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g3cedcad6) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Mon Jan 17 16:05:14 UTC 2022
[    0.000000] Machine model: Hardkernel ODROID-N2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x00000000e4c00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000000f4e5afff]
[    0.000000] NUMA: NODE_DATA [mem 0xe4651100-0xe4652fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000f4e5afff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000004ffffff]
[    0.000000]   node   0: [mem 0x0000000005000000-0x00000000052fffff]
[    0.000000]   node   0: [mem 0x0000000005300000-0x00000000f4e5afff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000f4e5afff]
[    0.000000] On node 0 totalpages: 1003099
[    0.000000]   DMA zone: 15674 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 1003099 pages, LIFO batch:63
[    0.000000] On node 0, zone DMA: 421 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.1
[    0.000000] percpu: Embedded 23 pages/cpu s56664 r8192 d29352 u94208
[    0.000000] pcpu-alloc: s56664 r8192 d29352 u94208 alloc=23*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 987425
[    0.000000] Policy zone: DMA
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=48b5e39712084f789b1af1944e49cbdd cgroup_enable=memory fsck.repair=yes root=PARTUUID=48617373-06 rootfstype=squashfs ro rootwait rauc.slot=A console=tty0 console=ttyAML0,115200n8
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3644428K/4012396K available (13440K kernel code, 1424K rwdata, 4120K rodata, 2624K init, 455K bss, 105824K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_meson_gpio: 100 to 8 gpio interrupt mux initialized
[    0.000000] random: get_random_bytes called from start_kernel+0x31c/0x4e0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000004] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000198] Console: colour dummy device 80x25
[    0.000579] printk: console [tty0] enabled
[    0.000664] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.000687] pid_max: default: 32768 minimum: 301
[    0.000771] LSM: Security Framework initializing
[    0.000864] AppArmor: AppArmor initialized
[    0.000921] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.000950] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.002758] rcu: Hierarchical SRCU implementation.
[    0.004806] EFI services will not be available.
[    0.005202] smp: Bringing up secondary CPUs ...
[    0.005828] Detected VIPT I-cache on CPU1
[    0.005887] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[... 475 lines cut ...]
[   45.688631] hassio: port 3(veth5765117) entered blocking state
[   45.694363] hassio: port 3(veth5765117) entered forwarding state
[   45.981031] hassio: port 4(vethb6626ad) entered blocking state
[   45.981281] hassio: port 4(vethb6626ad) entered disabled state
[   45.987314] device vethb6626ad entered promiscuous mode
[   45.994607] hassio: port 4(vethb6626ad) entered blocking state
[   45.998086] hassio: port 4(vethb6626ad) entered forwarding state
[   46.302849] hassio: port 4(vethb6626ad) entered disabled state
[... cut ...]
[   48.751379] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   48.761040] axg-sound-card sound: ASoC: PRE_PMU: be.dai-link-8-playback event failed: -22
[   48.782881] hdmi-audio-codec hdmi-audio-codec.4.auto: Not able to map channels to speakers (-22)
[   48.786094] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   48.795801] axg-sound-card sound: ASoC: PRE_PMU: be.dai-link-8-playback event failed: -22

agners, 2022-02-07:
@liudger what model/revision is that exactly and what eMMC are you using?

liudger, 2022-02-07:
It is the original odroid n2. It's the first revision. Ordered when it was new.

closed 2022-05-09 by pvizeli