home-assistant/operating-system #1106: Expand data partition on US

rp3tya, 2020-12-19:
**Hardware Environment**

- [x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [x] Fresh installation of release 2020.12.7
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

Completely reinstalled HA on my RPi4 today and restored addons/configuration successfully. My goal was to move data to a USB drive (for the first time). This was also successful and everything works but the data partition on USB does not use the full capacity of the drive:

```
# df -h /mnt/data
Filesystem                Size      Used Available Use% Mounted on
/dev/sda1                57.8G     14.1G     40.8G  26% /mnt/data

# fdisk -l /dev/sda1
Disk /dev/sda1: 114.61 GiB, 123059813888 bytes, 240351199 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
```

What is to do?

lucasHSA, 2020-12-20:
I have the same problem on my Pi after moving the data partition to the SSD. Everything is running fine, but the partition is not expanding.

**Hardware Environment**

- [x ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ x] Updated from version 0.118.5
- Additional information (if accessible):
## System Health

version | 2020.12.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.6
os_name | Linux
os_version | 5.4.79-v7l
arch | armv7l
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4906
Installed Version | 1.9.0
Stage | running
Available Repositories | 761
Installed Repositories | 5

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | January 3, 2021, 1:00 AM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.8
-- | --
update_channel | stable
supervisor_version | 2020.12.7
docker_version | 19.03.13
disk_total | 13.9 GB
disk_used | 4.7 GB
healthy | true
supported | true
board | rpi4
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.3.0), AppDaemon 4 (0.3.2), Mosquitto broker (5.1), Zigbee2mqtt (1.16.2.2), File editor (5.2.0), Grocy (0.9.2), AdGuard Home (2.6.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
mode | storage
views | 2
resources | 1

</details>

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
```
# df -h /mnt/data
Filesystem                Size      Used Available Use% Mounted on
/dev/sda1                13.9G      4.7G      8.5G  36% /mnt/data
```
```
fdisk -l /dev/sda1
Disk /dev/sda1: 223.54 GiB, 240023838720 bytes, 468796560 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 33553920 bytes
```

agners, 2020-12-23:
Can you login onto the operating system (see https://developers.home-assistant.io/docs/operating-system/debugging) and post the output of `journalctl -u hassos-expand`?

lucasHSA, 2020-12-23:
@agners Sadly, there is nothing to see. I moved the data partition on 19.12.2020 to the USB driver.

```
# journalctl -u hassos-expand
-- Logs begin at Wed 2020-11-11 17:53:49 UTC, end at Wed 2020-12-23 14:05:08 UTC. --
-- No entries --
```

rp3tya, 2020-12-23:
I get some old timestamps in journalctl:
```
...
Sep 12 12:16:38 homeassistant systemd[1]: hassos-expand.service: Succeeded.
Sep 12 12:16:38 homeassistant systemd[1]: Finished HassOS data resizing.
-- Reboot --
Sep 12 12:25:18 homeassistant systemd[1]: Starting HassOS data resizing...
Sep 12 12:25:18 homeassistant hassos-expand[299]: [INFO] Detected GPT partition label
Sep 12 12:25:18 homeassistant hassos-expand[299]: [INFO] No resize of data partition needed
Sep 12 12:25:18 homeassistant systemd[1]: hassos-expand.service: Succeeded.
Sep 12 12:25:18 homeassistant systemd[1]: Finished HassOS data resizing.
```

Manually:
```
# /usr/libexec/hassos-expand
[INFO] Detected GPT partition label
[INFO] No resize of data partition needed
```

karlg, 2020-12-26:
Same on an Odroid N2+, booting from 16 GB eMMC, with 120 GB SSD.
After datactl move:
- lsblk says sda and sda1 size is 111.8G (= capacity of the SSD)
- df -h says sda1 size is 13.6G, which is the size of /mnt/data before the move

pepsonEL, 2020-12-26:
But is any chance to boot Odroid N2 from SSD on USB?

karlg, 2020-12-27:
> But is any chance to boot Odroid N2 from SSD on USB?

That would be very nice, meaning an easy path to getting a good hardware foundation to HA without having to do command line magic - good for newbies and non-unixers like me. Especially so since it looks like the N2 is getting some airplay at the moment.

You can boot Ubuntu from an SSD on an N2, but I think there's a lot of mucking around to make that happen, and that's put it in the too-hard basket for the heroic HA developers!

agners, 2020-12-28:
@lucasHSA  @rp3tya thanks for the logs, indeed it seems that the expand service does not get started here. The reason is that we use partition label in the expand service, but I use a new partition label in the refactored `datactl`. So this is really a bug in the current implementation.

I could do the expand just after the move as a one time operation, but this would not help if the move already has been completed (like yours). I am looking in a generic fix for this using systemd-growfs which also takes care of data partitions already moved.

agners, 2020-12-28:
@pepsonEL the ODROID-N2+ firmware doesn't have support to boot from USB, so no. 

@karlg I think the USB boot flow for Ubuntu makes use of the boot loader present in the SPI ROM: There is a SPI ROM which has a Linux based boot loader on it. It allows to boot the Linux kernel directly, but that would break our update system (which relies on U-Boot). In theory we could use this bootloader to boot U-Boot, but a couple of weeks ago when I tried it it did not work just like that. Maybe there is a hack, but it seems not entirly trivial.

So for now, external data disk is the way to go. It is better since it splits I/O's: The OS has full I/O bandwidth of the eMMC, and Home Assistant Supervisor/Core etc. can use the full I/O bandwidth of the USB SSD.

karlg, 2020-12-28:
Thanks @agners!

I agree about bandwidth. You can think about HA as home automation plus data logging.

For the easiest possible install, I think the best would be to boot off a microSD card, but record data, and cache the card onto an SSD the system might automatically find.. :-)

Ran:
/Iib/systemd/systemd-growfs /mnt/data
Worked!

Thank you!

agners, 2020-12-30:
Fix will be in 5.10 onwards.

closed 2020-12-30 by agners