home-assistant/operating-system #1108: Hassos 5.8

YvesDufresne, 2020-12-19:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[12:16:51] INFO: Update udev information
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[12:16:51] INFO: Starting local supervisor watchdog...
20-09-12 12:16:55 INFO (MainThread) [__main__] Initializing Supervisor setup
20-09-12 12:16:55 INFO (MainThread) [supervisor.docker.network] Can't find Supervisor network, creating a new network
20-09-12 12:16:56 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
20-09-12 12:16:56 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4-64
20-09-12 12:16:56 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/aarch64-hassio-supervisor with version 2020.12.6
20-09-12 12:16:56 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
20-09-12 12:16:56 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
20-09-12 12:16:56 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
20-09-12 12:16:56 INFO (MainThread) [__main__] Setting up Supervisor
20-09-12 12:16:56 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
20-09-12 12:16:57 INFO (MainThread) [supervisor.host.info] Updating local host information
20-09-12 12:16:57 INFO (MainThread) [supervisor.host.services] Updating service information
20-09-12 12:16:57 INFO (MainThread) [supervisor.host.network] Updating local network information
20-09-12 12:16:58 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
20-09-12 12:16:58 INFO (MainThread) [supervisor.host] Host information reload completed
20-09-12 12:16:58 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
20-09-12 12:16:58 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
20-09-12 12:16:58 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-dns versions: ['2020.11.0']
20-09-12 12:16:58 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-dns with version 2020.11.0
20-09-12 12:16:58 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
20-09-12 12:17:00 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS homeassistant/aarch64-hassio-dns with version 2020.11.0 - 172.30.32.3
20-09-12 12:17:00 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
20-09-12 12:17:00 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-audio versions: ['17']
20-09-12 12:17:00 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-audio with version 17
20-09-12 12:17:00 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
20-09-12 12:17:01 INFO (SyncWorker_1) [supervisor.docker.audio] Starting Audio homeassistant/aarch64-hassio-audio with version 17 - 172.30.32.4
20-09-12 12:17:01 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-cli versions: ['2020.11.1']
20-09-12 12:17:01 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-cli with version 2020.11.1
20-09-12 12:17:01 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
20-09-12 12:17:03 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI homeassistant/aarch64-hassio-cli with version 2020.11.1 - 172.30.32.5
20-09-12 12:17:03 INFO (SyncWorker_1) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-observer versions: ['2020.10.1']
20-09-12 12:17:03 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-observer with version 2020.10.1
20-09-12 12:17:03 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
20-09-12 12:17:05 INFO (SyncWorker_0) [supervisor.docker.observer] Starting Observer homeassistant/aarch64-hassio-observer with version 2020.10.1 - 172.30.32.6
20-09-12 12:17:05 INFO (SyncWorker_1) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-multicast versions: ['3']
20-09-12 12:17:05 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-multicast with version 3
20-09-12 12:17:05 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
20-09-12 12:17:06 INFO (SyncWorker_0) [supervisor.docker.multicast] Starting Multicast homeassistant/aarch64-hassio-multicast with version 3 - Host
20-09-12 12:17:06 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
20-09-12 12:17:06 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 0 Home Assistant secrets
20-09-12 12:17:06 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/raspberrypi4-64-homeassistant versions: ['landingpage']
20-09-12 12:17:06 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/raspberrypi4-64-homeassistant with version landingpage
20-09-12 12:17:06 INFO (MainThread) [supervisor.homeassistant.core] Starting HomeAssistant landingpage
20-09-12 12:17:06 INFO (MainThread) [supervisor.homeassistant] Update pulse/client.config: /data/tmp/homeassistant_pulse
20-09-12 12:17:07 INFO (SyncWorker_1) [supervisor.docker.homeassistant] Starting Home Assistant homeassistant/raspberrypi4-64-homeassistant with version landingpage
20-09-12 12:17:07 INFO (MainThread) [supervisor.hassos] Detect HassOS 5.8 / BootSlot A
20-09-12 12:17:07 INFO (MainThread) [supervisor.store.git] Cloning add-on https://github.com/hassio-addons/repository repository
20-09-12 12:17:07 INFO (MainThread) [supervisor.store.git] Cloning add-on https://github.com/home-assistant/addons repository
20-12-19 21:28:50 INFO (MainThread) [supervisor.store] Loading add-ons from store: 66 all - 66 new - 0 remove
20-12-19 21:28:50 INFO (MainThread) [supervisor.addons] Found 0 installed add-ons
20-12-19 21:28:50 INFO (MainThread) [supervisor.snapshots] Found 0 snapshot files
20-12-19 21:28:50 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
20-12-19 21:28:50 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
20-12-19 21:28:50 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
20-12-19 21:28:50 INFO (MainThread) [supervisor.resolution.check] System checks complete
20-12-19 21:28:50 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
20-12-19 21:28:50 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
20-12-19 21:28:50 INFO (MainThread) [__main__] Running Supervisor
[... 24 lines cut ...]
20-12-19 21:29:07 INFO (MainThread) [supervisor.host] Host information reload completed
20-12-19 21:29:20 ERROR (SyncWorker_0) [supervisor.docker.interface] Can't install homeassistant/raspberrypi4-64-homeassistant:2020.12.1 -> 500 Server Error for http+docker://localhost/v1.40/images/create?tag=2020.12.1&fromImage=homeassistant%2Fraspberrypi4-64-homeassistant: Internal Server Error ("Get "https://registry-1.docker.io/v2/": context deadline exceeded").
20-12-19 21:29:20 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
20-12-19 21:29:50 INFO (SyncWorker_1) [supervisor.docker.interface] Updating image homeassistant/raspberrypi4-64-homeassistant:landingpage to homeassistant/raspberrypi4-64-homeassistant:2020.12.1
20-12-19 21:29:50 INFO (SyncWorker_1) [supervisor.docker.interface] Downloading docker image homeassistant/raspberrypi4-64-homeassistant with tag 2020.12.1.
20-12-19 21:30:05 ERROR (SyncWorker_1) [supervisor.docker.interface] Can't install homeassistant/raspberrypi4-64-homeassistant:2020.12.1 -> 500 Server Error for http+docker://localhost/v1.40/images/create?tag=2020.12.1&fromImage=homeassistant%2Fraspberrypi4-64-homeassistant: Internal Server Error ("Get "https://registry-1.docker.io/v2/": context deadline exceeded (Client.Timeout exceeded while awaiting headers)").
20-12-19 21:30:05 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
- [ x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ x] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

agners, 2020-12-20:
Hm, this is not the only report I saw with this error:

```
20-12-19 21:29:20 ERROR (SyncWorker_0) [supervisor.docker.interface] Can't install homeassistant/raspberrypi4-64-homeassistant:2020.12.1 -> 500 Server Error for http+docker://localhost/v1.40/images/create?tag=2020.12.1&fromImage=homeassistant%2Fraspberrypi4-64-homeassistant: Internal Server Error ("Get "https://registry-1.docker.io/v2/": context deadline exceeded").
```

I cannot reproduce this here, so it depends on the network environment. But since it used to work before 5.6, I assume it is realted to the upgrade to Buildroot 2020.11 or/and systemd-reolved.

I just updated to the latest stable version of systemd-resolved and created a build, can you try with this build?
https://os-builds.home-assistant.io/6.0.dev20201220/

YvesDufresne, 2020-12-20:
This is the way for me to reproduce
------------------
I have a strange problem to share
My network: Cable ISP to WiFi router -> Client ( Iogear GWU637 ) + 50 Feet
cat6 to router Netgear R6020
Pi 4 connecter wire to Netgear . With OS Version 5.5 and before everything
work fine
With 5.6 and up: Not able to clean install ( stuck at Preparing Home
Assistant ) and not able to update or install new addons. Have this message:
500 Server Error .......bla bla bla :.... context deadline exceeded
(Client.Timeout exceeded while awaiting headers)")
If I connect Pi 4 direct to IoGear + 50 Feet cat6 (bypass Netgear) trouble
disapper. Its look like version 5.6 and up is more sensitive to timing. (
Timing have been change ? )
----------------------

I will try this build .... But I need instruction to do it

ha core update --version 2020.12.20
May be ?????

Thanks for quick answer .

agners, 2020-12-20:
You can install it using balenaEtcher (if you do a fresh install, just download the `tar.xz` file). If you want to use the built-in update mechanism you need to switch to the `dev` channel:

```
ha supervisor options --channel dev
ha supervisor reload
ha os update --version 6.0.dev20201220
```

Don't forget to switch back to the stable channel afterwards.
```
ha supervisor options --channel stable
```

YvesDufresne, 2020-12-21:
I did clean from sd install   .... FAIL with same result  You have the
complete logs  ...The logs say retiring in 30 sec...never did

If you need more clean install from sd .. you welcome

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[17:25:33] INFO: Update udev information
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[17:25:34] INFO: Starting local supervisor watchdog...
20-12-16 17:25:37 INFO (MainThread) [__main__] Initializing Supervisor setup
20-12-16 17:25:37 INFO (MainThread) [supervisor.docker.network] Can't find Supervisor network, creating a new network
20-12-16 17:25:38 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
20-12-16 17:25:38 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4-64
20-12-16 17:25:38 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/aarch64-hassio-supervisor with version 2020.12.7
20-12-16 17:25:38 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
20-12-16 17:25:38 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
20-12-16 17:25:38 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
20-12-16 17:25:38 INFO (MainThread) [__main__] Setting up Supervisor
20-12-16 17:25:38 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
20-12-16 17:25:39 INFO (MainThread) [supervisor.host.info] Updating local host information
20-12-16 17:25:39 INFO (MainThread) [supervisor.host.services] Updating service information
20-12-16 17:25:39 INFO (MainThread) [supervisor.host.network] Updating local network information
20-12-16 17:25:40 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
20-12-16 17:25:40 INFO (MainThread) [supervisor.host] Host information reload completed
20-12-16 17:25:40 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
20-12-16 17:25:40 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
20-12-16 17:25:40 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-dns versions: ['2020.11.0']
20-12-16 17:25:40 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-dns with version 2020.11.0
20-12-16 17:25:41 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
20-12-16 17:25:42 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS homeassistant/aarch64-hassio-dns with version 2020.11.0 - 172.30.32.3
20-12-16 17:25:42 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
20-12-16 17:25:42 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-audio versions: ['17']
20-12-16 17:25:42 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-audio with version 17
20-12-16 17:25:42 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
20-12-16 17:25:43 INFO (SyncWorker_0) [supervisor.docker.audio] Starting Audio homeassistant/aarch64-hassio-audio with version 17 - 172.30.32.4
20-12-16 17:25:44 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-cli versions: ['2020.11.1']
20-12-16 17:25:44 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-cli with version 2020.11.1
20-12-16 17:25:44 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
20-12-16 17:25:45 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI homeassistant/aarch64-hassio-cli with version 2020.11.1 - 172.30.32.5
20-12-16 17:25:45 INFO (SyncWorker_1) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-observer versions: ['2020.10.1']
20-12-16 17:25:45 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-observer with version 2020.10.1
20-12-16 17:25:45 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
20-12-16 17:25:47 INFO (SyncWorker_0) [supervisor.docker.observer] Starting Observer homeassistant/aarch64-hassio-observer with version 2020.10.1 - 172.30.32.6
20-12-16 17:25:47 INFO (SyncWorker_1) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-multicast versions: ['3']
20-12-16 17:25:47 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-multicast with version 3
20-12-16 17:25:47 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
20-12-16 17:25:48 INFO (SyncWorker_0) [supervisor.docker.multicast] Starting Multicast homeassistant/aarch64-hassio-multicast with version 3 - Host
20-12-16 17:25:48 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
20-12-16 17:25:48 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 0 Home Assistant secrets
20-12-16 17:25:48 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/raspberrypi4-64-homeassistant versions: ['landingpage']
[... cut ...]
20-12-21 00:05:33 ERROR (SyncWorker_1) [supervisor.docker.interface] Can't install homeassistant/raspberrypi4-64-homeassistant:2020.12.1 -> 500 Server Error for http+docker://localhost/v1.40/images/create?tag=2020.12.1&fromImage=homeassistant%2Fraspberrypi4-64-homeassistant: Internal Server Error ("Get "https://registry-1.docker.io/v2/": context deadline exceeded (Client.Timeout exceeded while awaiting headers)").
20-12-21 00:05:33 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec

YvesDufresne, 2020-12-21:
I have a idea ( may be not so good ? ) just instatt 2 or more router WiFi
in and WiFi out till you have this condition with 5.6 and up ...after try
with 5.5 .. If 5.5 work and 5.6 fail you have my condition.

Just a idea ...Or I will make few test for you..

agners, 2020-12-21:
Just to be clear, your network looks like this right?

```
|--------|            |----------------------|               |---------------|              |-----------------|
| ISP    |  - WiFi -  | WiFi to Eth (GWU637) |  - Ethernet - | Netgear R6020 | - Ethernet - | RPi 4 with HAOS |
|--------|            |----------------------|               |---------------|              |-----------------|
```
And when you take the Netgear R6020 out of the equation things start to work?

It could be an MTU issue, can you try login onto the OS using ssh (port 22222), see https://developers.home-assistant.io/docs/operating-system/debugging/

```
sysctl net.ipv4.tcp_mtu_probing=1
```

YvesDufresne, 2020-12-21:
Yes My configuration is exactly that

When I was in at   ha > Prompt I have to login and # .....Issue the command
and it fix the problem..
Does this fix will stay on reboot. and be implement in next issue
P.S. this test was done OS 5.9

Thanks

YvesDufresne, 2020-12-21:
Does MTU setting is persistent ...Because after few hours trouble came
back. Reissue the command and work again ???? Strange ?

agners, 2020-12-22:
The command is not persistent! Currently there is no way to make it persistent. We might consider setting it by default in HAOS.

So the problem went away as soon as that option was set?

YvesDufresne, 2020-12-22:
I am not so sure anymore because if I reboot and wait 1 hour I am not able
to install new addons , if I try 2 ,3 times it will go even without option.
For me the option seemed to help but not a fix. After The  first addons all
next one work fine.
This issue NEVER happened with OS 5.5 and before.

agners, 2020-12-22:
Hm, it could be related to #1113 then. I have a fix for that which will eventually go into 5.10. Would be good if you can give it a try with 5.10 again.

YvesDufresne, 2020-12-22:
Yes as soon 5.10 ready or if I have access I will.

agners, 2020-12-22:
@YvesDufresne I made a pre-release, you can find it here: https://os-builds.home-assistant.io/6.0.dev20201222/

If you want to use the update system, you need to switch to the dev channel:
```
ha supervisor options --channel dev
ha supervisor reload
ha os update --version 6.0.dev20201222
```

Don't forget to switch back to the stable channel afterwards:
```
ha supervisor options --channel stable
```

YvesDufresne, 2020-12-22:
Same Issue and message

YvesDufresne, 2020-12-23:
I try a few more tests. I was able for the first time since (5.5 ) to make
a clean install  .On the observer logs 3 time retry in 30 sec but get on
finishing install . After trying to install addons it fail twice and get ok
after... When first addons get ok all others get ok. after a few hours add
new addons fail on first try but get ok on multiple retry.

LarsVenema95, 2020-12-23:
I've also got this error when I try to update to the latest Home Assistant Core.
I am running on a virtual machine for test purposes. But since a while (i am running 0.118.4) I can't update anymore.

I get to following error: 
20-12-23 08:21:36 ERROR (SyncWorker_4) [supervisor.docker.interface] Can't install homeassistant/qemux86-64-homeassistant:2021.1.0.dev20201223 -> 500 Server Error for http+docker://localhost/v1.40/images/create?tag=2021.1.0.dev20201223&fromImage=homeassistant%2Fqemux86-64-homeassistant: Internal Server Error ("Get "https://registry-1.docker.io/v2/": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)").

I tried to update to the latest dev build by executing the following commands but that didn't solve the problem for me.
```
ha supervisor options --channel dev
ha supervisor reload
ha os update --version 6.0.dev20201222
```

Some others have the same problem: 
https://www.reddit.com/r/homeassistant/comments/kejpyg/2020121_broke_my_system/

LarsVenema95, 2020-12-23:
I tried to create a new virtual machine with the latest dev build (6.0.dev20201222).
However, when install it, it gets stucked at the "Preparing Home Assistant" screen with the same error as mentioned above. Thus, a fresh install with the latest HASS OS doesn't work work me.

Changing the DNS server is the solution for me. On the forum there is a thread about this problem: https://community.home-assistant.io/t/cant-install-homeassistant-2020-12-0/256131

YvesDufresne, 2020-12-23:
Might help.....
If I try to add new addons will fail most of the time. After few retry will
add OK.
But ( very strange ) If I log in before ssh root@192.168.1.3 -p 22222 will
add Ok in the first attempt.
I make at least 20 test ..
Hope will help but it's not logical for me that when your login it will
make a difference.

YvesDufresne, 2020-12-26:
I did 20+ more tests... on both on my Pi 4 4Gb Running HassoS 5.9  HA
2020.12.1
If I try to add any new addons it's ALWAYS fait on first attempt ..
If I just  ssh root@192.168.1.x -p 22222 to Pi 4  doing nothing prompt ( ha
>) ...The add any addons will ALWAYS install on first try.

Strange but true.

agners, 2020-12-29:
> I try a few more tests. I was able for the first time since (5.5 ) to make
a clean install 

@YvesDufresne was that with the development build? You can also download the latest development build for a fresh installation from https://os-builds.home-assistant.io/6.0.dev20201228/.

YvesDufresne, 2020-12-29:
Yes it was with dev build but it's retry 3 times ..This build has the same
issue .. I change my netgear  ( $25 ) for tp-link ($100) and the problem is
gone.
That does not say 5.6 and up work ok but with a better router it's fine.

P.S. I tried the dev 20201228 on my old setup of course.
What do you think of my 2 last messages ( strange ? but real  )

scyto, 2021-01-10:
was helping someone troubleshoot this, we verified he had full working DNS and internet connectivity and could manually pull the image from, docker.io, reverting to an older build fixed the issue.

Did anything change in the code wrt to timeouts (which might explain why some hit it and some don't), is this really just timing out as the error indicates? or is the some other issue with the create function being called?

bschatzow, 2021-01-10:
> Hm, this is not the only report I saw with this error:
>
> ```
> 20-12-19 21:29:20 ERROR (SyncWorker_0) [supervisor.docker.interface] Can't install homeassistant/raspberrypi4-64-homeassistant:2020.12.1 [...]

is there a developers version of the OS?  Above 5.2 my system freezes.  I tried 5.10 again and it froze after a little over 3 hours.  I did post the logs in the 1119 issue.  Not sure what to try next on my PI4 64 bit with SSD.  Thanks

YvesDufresne, 2021-01-10:
I try 20201228 2 week ago. Are you sure you want to try 20201220 and not
20210105 ?

bschatzow, 2021-01-10:
I am talking about the OS version. Lots of different versions (Supervisor, OS, HA)  so easy to confuse.

YvesDufresne, 2021-01-10:
Me too os version 20201228

If I try will be la test is version

YvesDufresne, 2021-01-10:
I try this one 2 week ago
os-builds.home-assistant.io/6.0.dev20201228

bschatzow, 2021-01-11:
Were you having the same RPI4 freeze issues?  If so, did this fix it?

YvesDufresne, 2021-01-11:
No freeze issue RPI4 boot from ssd

JW6677, 2021-02-09:
It's not just the RPI.
I have a virtual machine and I can't get it to work either!
On my last instance I couldn't update and now after a reinstall I can't get to the CLI at all with it unable to download the docker image. Please fix!

umrath, 2021-05-18:
I also see Post "http://supervisor/core/check": context deadline exceeded (Client.Timeout exceeded while awaiting headers) when issuing a ha core check since 2 weeks or so.

Home Assistant is seemingly unphased by this and the UI check is working. Reboots are working.

I have no idea what to do.

BartoszPierzchlewiczMacrix, 2021-05-18:
I have the same timeout, while command:
`docker exec homeassistant python -m homeassistant --script check_config --config /config`
is working properly, without timeout

wojski, 2023-09-20:
I believe that after two days of grappling with my setup, which consists of a Wyse 5060 and a Netgear R6800, I have finally managed to get it working.

I tried multiple configurations:
1. Ubuntu with Home Assistant (HA)
2. HAOS 10.5
3. HAOS 10.4
4. Debian with HA
5. HAOS 10.5 (This one worked!)

The solution involved two main steps:
1. Disabling IPv6 on HAOS (using the `network update` command, among others).
2. Checking the status of the network connection, especially the nameserver. In my case, it was pointing to my router, which couldn't resolve any addresses. This was similar to an issue I faced with Debian, where the DNS stopped working after installing the network manager. To fix this, I manually set the `nameserver` to `8.8.8.8` and restarted the entire machine. After that, all images began to load correctly, and all add-ons became available and installed without any issues.

Final configuration:

[image]