home-assistant/operating-system #1109: OS 5.9 Boot SSD on USB3 fail but OK on USB2 (RPI4)

scoubidoo, 2020-12-20:
**Hardware Environment**

- [X ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [X ] Updated from version 5.8
- Additional information (if accessible):

`## System Health

version | 2020.12.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.6
os_name | Linux
os_version | 5.4.79-v8
arch | aarch64
timezone | Europe/Paris

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4526
Installed Version | 1.9.0
Stage | running
Available Repositories | 692
Installed Repositories | 30

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 7 janvier 2021 à 1:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.9
-- | --
update_channel | beta
supervisor_version | 2020.12.7
docker_version | 19.03.13
disk_total | 116.7 GB
disk_used | 4.9 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Let's Encrypt (4.11.0), Samba share (9.3.0), UniFi Controller (0.19.3), deCONZ (6.6.1), Terminal & SSH (8.10.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
mode | storage
views | 7
resources | 37

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>`

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

agners, 2020-12-20:
This is likely due to firmware/USB SSD issue. Which firmware are you using (date on the initial screen)? You probably have to downgrade your firmware (see #1045).

scoubidoo, 2020-12-21:
hello, i use the last EEPROM on stable channel (11/12/2020)
i need to use the eeprom on september ?

i format my ssd and i install  RASPIAN on it .. i plug it on the USB3 and it boot directly, something on HASSIO OS don't work correctly no ?

scoubidoo, 2020-12-21:
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
HDMI_DELAY=0
BOOT_ORDER=0xf41
USB_MSD_PWR_OFF_TIME=0

i already test all this possiblities. i will try with 5.9 from scratch

scoubidoo, 2020-12-21:
same with HASSIO 5.9 from SCRATCH with this EEPROM.

agners, 2020-12-21:
Does this combination boot with Raspberry Pi OS?

scoubidoo, 2020-12-21:
yes the rasp boot on USB 3 with RaspOS 32bit
i update the firmware of the chip of my NVME case but nothing better.

scoubidoo, 2020-12-21:
i don't know if that can help you to solve this issue 

[image]
info of my SSD on raspOS
in RaspOS i have few directory i think in Hassio i have less 
[image]

agners, 2020-12-22:
What kind of USB 3.0 <-> NVMe is your enclosure using?

Do you have a USB to TTL cable? If so, enabling UART (by editing `config.txt` and setting `enable_uart=1`) and connecting the serial console would show you the output of U-Boot, which might give some hints why it doesn't boot.

scoubidoo, 2020-12-27:
I format my ssd and install the x64 OS, and it boot directly 
to resume 
* update EEPROM on last version
* fresh install on x64 HASSIO OS

scoubidoo, 2020-12-27:
> What kind of USB 3.0 <-> NVMe is your enclosure using?
> 
> Do you have a USB to TTL cable? If so, enabling UART (by editing `config.txt` and setting `enable_uart=1`) and connecting the serial console would show you the output of U-Boot, which might give some hints why it doesn't boot.

i'm on NVME on USB 3.0

Dezorian, 2021-01-15:
> 
> 
> I format my ssd and install the x64 OS, and it boot directly
> to resume
> 
>     * update EEPROM on last version
> 
>     * fresh install on x64 HASSIO OS

I have the exact same problems.

Raspberry Pi 4B 4GB Ram
Raspberry Pi OS Lite 2021-01-11 -> boots great from SSD
EEPROM 2021-01-14
HassOS 5.10, 5.9, 4.20 (tried them all) -> black screen

Enclosure: Argon ONE with M2 SSD Kingston 120GB (confirmed working says [this site](https://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/)) uses M.2 SATA (B+M Key) Pi 4 Case and the ASMedia ASM1051E sata bridge.

First attempt:
Updated EEPROM with Raspberry Pi v1.5 Image EEPROM boot recovery (using this [guide](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md) from Raspberry Foundation)
Tried a dd copy from the SD card to the SSD the i got a 4 times blinking green led (start*.elf not found). Guessing it was a long shot for an easy copy paste session, I tried using snapshots and a clean install of HassOS...

Second and other attempts:
Updated raspbian with sudo apt full-upgrade
change boot order using sudo raspi-config to USB boot (not defaults)
Burn SSD with balenaEtcher and Rufus with the 5.10/5.9/4.20 HassOS image
Does not startup. Black screen.
Tried other USB ports (USB2 and USB3 using USB2 extension cable)

Edited config.txt:
Uncommented #hdmi_safe=1 in config.txt trying to get some info -> black screen

Edited cmdline.txt
Tried adding root=/dev/sda1 (and /dev/sda2) to see if that helps -> black screen

I'll try the "boot first, plug in SSD later" option tonight. 
But it seems HassOS is missing something that Rasperry Pi OS Lite has to boot properly.

closed 2022-01-11 by github-actions[bot]