home-assistant/operating-system #1112: "NUC" missing MMC kernel modules

superm1, 2020-12-21:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [x] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

Dell IOT edge gateway 3003 (w/ 8G of MMC based storage).
  
**Home Assistant OS release:**
- [x] Fresh installation of release 5.9
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

Unable to SSH in since I can't finish a boot, so here is the best I can offer.
[image]

**Description of problem:**

I've found that dd'ing the NUC image to `/dev/mmcblk0` from a live media, I'm unable to boot the machine.  Looking through the kernel log I see that it's been waiting for the root partition at the matching `PARTUUID`  to show up.  I double checked from my live media that the partition wasn't corrupted and that the `PARTUUID` matched what was expected.  I double checked `./buildroot/board/pc/linux.config` and it looks to me that you're missing the various `CONFIG_MMC` related kernel options for the `intel_nuc` build.

I know you guys mentioned in the conference recently the intention to have a "generic" x86_64 image eventually, so please let me know if you feel this might be the type of thing that falls into that bucket rather than adding MMC into the nuc build.

superm1, 2020-12-28:
@agners I saw the commits to fix this thanks.  I'm curious - does development image https://os-builds.home-assistant.io/6.0.dev20201222/ have this commit or was it done after that?  It isn't clear to me with the ambiguity of 6 days ago happening to fall on that day.

superm1, 2020-12-28:
@agners I saw in the github actions workflows output it should include the commit.  However it's still not working.  Comparing to a Ubuntu live key, I notice that `sdhci_acpi` is still missing (`MMC_SDHCI_ACPI`).  Can you add that in and trigger a new action build?

agners, 2020-12-28:
@superm1 ah yeah that make sense. Will add this config and trigger another build, was anyway planning on running one.

superm1, 2020-12-29:
This makes it further but sadly [image] it appears something else still missing. I'll try to sleuth some more comparing the Ubuntu build.

agners, 2020-12-29:
Hm, maybe `CONFIG_GPIOLIB` and `CONFIG_GPIO_ACPI` is required for card detect, but that is a rather wild guess.

If you have a toolchain and kernel sources on the Ubuntu installation, you can use `make localmodconfig` to generate a config which includes all loaded modules etc.

This config might be useful as well: https://gist.github.com/pdp7/9c726e18bd91a5f30d44ee885fcd2aba

Created a manual build with the GPIOs enabled locally and uploaded, can you give this a try? https://os-builds.home-assistant.io/6.0.dev20201229/

superm1, 2020-12-29:
Unfortunately that one didn't do the trick.   I don't have anything installed on the box, so I grabbed the kernel and toolchain and  generated a config from loaded modulesin the live media as you suggested here: https://paste.ubuntu.com/p/7JNTbhZZ3k/

> Hm, maybe CONFIG_GPIOLIB and CONFIG_GPIO_ACPI is required for card detect, but that is a rather wild guess.

It seems that both of those were set for Ubuntu.

The kernel default for `CONFIG_MMC_BLOCK_MINORS` appears to be 8 (which is what Ubuntu hardcodes in the config), so that can't be it.

I noticed this in the dmesg on Ubuntu:
```
[    2.683469] dw_dmac INTL9C60:00: DesignWare DMA Controller, 8 channels
[    2.697847] mmc0: SDHCI controller on ACPI [80860F14:01] using ADMA
[    2.712313] mmc1: SDHCI controller on ACPI [INT33BB:00] using ADMA
[    2.723601] dw_dmac INTL9C60:01: DesignWare DMA Controller, 8 channels
[    2.724251] mmc2: SDHCI controller on ACPI [80860F16:00] using ADMA
[    2.760838] mmc1: new high speed SDIO card at address fffd
[    2.786645] mmc0: new HS200 MMC card at address 0001
```

I don't see any DMA controller set in the NUC config, so my guesses are these:
```
CONFIG_DMADEVICES=y

CONFIG_DMA_ENGINE=y
CONFIG_DMA_ACPI=y
CONFIG_DW_DMAC_CORE=m
CONFIG_DW_DMAC=m
```

agners, 2020-12-30:
@superm1 created another build with those configs set. Let's hope this did the trick :crossed_fingers: 
https://os-builds.home-assistant.io/6.0.dev20201230/

superm1, 2020-12-30:
@agners unfortunately still no dice.  I cloned the tree locally and experimented with adding in some of the missing options and just copying over the new `bzImage`.  I've come up with a combination that works.   I think it's likely the LPSS devices, but I'll try to narrow it down and submit a pull request when I've got something more minimal.

[./scripts/diffconfig ~/hassos_broken.txt ~/hassos_working.txt](https://github.com/home-assistant/operating-system/files/5752905/diff.txt)

superm1, 2020-12-30:
Yep, it seems adding just that one config option seems to be enough to do it for me.

agners, 2020-12-30:
Thanks! Will backport the fix to release-5, so it will be in 5.10 onwards.

closed 2020-12-30 by agners

superm1, 2020-12-30:
> Thanks! Will backport the fix to release-5, so it will be in 5.10 onwards.

Great, thanks!

PR #1145 (2020-12-30): Add LPSS support for intel_nuc
This is needed for MMC to enumerate properly (Fixes: #1112)
commit cc53ad4f (2020-12-30): Add LPSS support for intel-nuc (#1145)

This is needed for MMC to enumerate properly (Fixes: #1112)
commit 01923727 (2020-12-30): Add LPSS support for intel-nuc (#1145)

This is needed for MMC to enumerate properly (Fixes: #1112)