home-assistant/operating-system #1115: Migration Raspberry PI snapshot to ORDROID N2 hangs (high Core RAM Usage)

rck777, 2020-12-21:
**Hardware**
- [ X] Raspberry Pi [1/2/3/4] (Source Snapshot)
- [ X ] ODROID [C2/C4/N2(+)/XU4] (Destination Snapshot)
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- [ ] Migrate version 2020.12.1 and Home Assistant OS 5.9 (problem also in 5.8)

Migration to ODROID-N2 4Gig. I make a snapshot on Raspberry PI4 and restore snapshot on the new ODROID-N2. I restored two instances. Both have the same problem

- Home Assistant Frontend -> Configuration -> Info (use the copy button in the System Health block)
- Or use this command: `ha info`

~ $ ha info
arch: aarch64
channel: stable
docker: 19.03.13
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.9"
homeassistant: 2020.12.1
hostname: homeassistant
logging: info
machine: odroid-n2
operating_system: Home Assistant OS 5.9
supervisor: 2020.12.7
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: Europe/Amsterdam

**Supervisor logs:**
20-12-21 21:16:07 INFO (SyncWorker_0) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-ssh with version 8.9.1
20-12-21 21:16:12 INFO (MainThread) [supervisor.core] Start Home Assistant Core
20-12-21 21:16:12 INFO (SyncWorker_4) [supervisor.docker.interface] Starting homeassistant
20-12-21 21:16:12 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
20-12-21 21:20:37 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
20-12-21 21:21:08 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
20-12-21 21:21:08 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.APPLICATION' starting 1 add-ons
20-12-21 21:21:08 INFO (SyncWorker_4) [supervisor.docker.interface] Cleaning addon_core_configurator application
20-12-21 21:21:09 INFO (SyncWorker_4) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-configurator with version 5.2.0
20-12-21 21:21:14 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled
20-12-21 21:21:14 INFO (MainThread) [supervisor.misc.hwmon] Started Supervisor hardware monitor
20-12-21 21:21:14 INFO (MainThread) [supervisor.core] Supervisor is up and running
20-12-21 21:21:14 INFO (MainThread) [supervisor.host.info] Updating local host information
20-12-21 21:21:14 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
20-12-21 21:21:14 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
20-12-21 21:21:14 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
20-12-21 21:21:14 INFO (MainThread) [supervisor.host.services] Updating service information
20-12-21 21:21:14 INFO (MainThread) [supervisor.host.network] Updating local network information
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[21:23:57] INFO: Update udev information
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[... 622 lines cut ...]
[  136.079804] audit: type=1325 audit(1608586297.355:166): table=nat family=2 entries=15 op=xt_replace pid=5659 subj==unconfined comm="iptables"
[  136.084291] audit: type=1325 audit(1608586297.359:167): table=filter family=2 entries=30 op=xt_replace pid=5661 subj==unconfined comm="iptables"
[  136.088978] audit: type=1325 audit(1608586297.363:168): table=nat family=2 entries=16 op=xt_replace pid=5663 subj==unconfined comm="iptables"
[  136.364862] audit: type=1325 audit(1608586297.639:169): table=nat family=2 entries=0 op=xt_register pid=5719 subj==unconfined comm="iptables"
[  136.370578] audit: type=1325 audit(1608586297.647:170): table=filter family=2 entries=0 op=xt_register pid=5720 subj==unconfined comm="iptables"
[  136.376576] audit: type=1325 audit(1608586297.651:171): table=nat family=2 entries=5 op=xt_replace pid=5723 subj==unconfined comm="iptables"
[  136.419086] eth0: renamed from vethd7252e0
[  136.438802] IPv6: ADDRCONF(NETDEV_CHANGE): vethf8179a6: link becomes ready
[  157.458053] kauditd_printk_skb: 7 callbacks suppressed
[  157.458057] audit: type=1334 audit(1608586318.735:179): prog-id=12 op=UNLOAD
[  157.458066] audit: type=1334 audit(1608586318.735:180): prog-id=11 op=UNLOAD
[  224.946918] rc rc0: two consecutive events of type space
~ $ 

**Description of problem:**
After migration HA from Raspberry PI4 to ODROID-N2 system hangs after few minutes. 

- installation with emmc image hassos_odroid-n2-5.8.img.xz (system = OK)
- Make first snapshot (system = OK)
- Upload snapshot file from Raspberry PI 4 with the same versions
- Restore snapshot 
- After restore Core RAM Usage memory usage increases quickly and system hangs after a few minutes
- Restore only works when i not restore the folder Home assistant configuration folder (i deselect this folder in restore snapshot properties. The most important folder ;-(

All USB hardware are disconnected at this time.

agners, 2020-12-22:
Hm, this is indeed interesting. It would be interesting which process' memory increases. Can you login onto the OS (using SSH, see https://developers.home-assistant.io/docs/operating-system/debugging/) and run the following command?

````
ps aux --sort -rss | head
```

rck777, 2020-12-22:
# ps aux --sort -rss | head
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root        5020  105 85.6 3636704 3291144 ?     Ssl  19:29   5:35 python3 -m homeassistant --config /config
root        1518  1.6  1.2 126140 47828 ?        Ssl  19:29   0:05 python3 -m supervisor
root         391 11.1  1.2 2155612 46456 ?       Ssl  19:29   0:37 /usr/bin/dockerd -H fd:// --storage-driver=overlay2 --log-driver=journald --data-root /mnt/data/docker
root         160  8.3  0.9 304968 37448 ?        Rs   19:28   0:28 /usr/lib/systemd/systemd-journald
root         399  0.5  0.5 2432832 21564 ?       Ssl  19:29   0:01 containerd --config /var/run/docker/containerd/containerd.toml --log-level info
root        2058  0.2  0.3 743840 12912 ?        Ssl  19:29   0:00 coredns -conf /etc/corefile
root        5331  0.4  0.3  27320 11988 ?        Ss   19:30   0:01 /usr/bin/python3 /usr/bin/hass-configurator /etc/configurator.conf
root        1150  0.0  0.1 1481364 6192 ?        Ssl  19:29   0:00 docker container start --attach hassio_supervisor
root        3192  0.5  0.1  94180  4592 ?        S<sl 19:29   0:01 pulseaudio --system -vvv --disallow-exit --exit-idle-time=-1 --disable-shm

rck777, 2020-12-22:
Problem found i think ;-) After i used ps i also applied jounalctl to see system messages. Since I was in debug mode anyway. There are problems with the integration dsmr (Dutch Smart Meter Requirements). See log below. There are hundreds / thousands of error lines present about this error. 
So, i removed the integration and the system now seems stable.

Dec 21 20:22:04 homeassistant fb413cc9aaba[393]:   File "/usr/src/homeassistant/homeassistant/components/dsmr/sensor.py", line 186, in connect_and_reconnect
Dec 21 20:22:04 homeassistant fb413cc9aaba[393]:     transport, protocol = await hass.loop.create_task(reader_factory())
Dec 21 20:22:04 homeassistant fb413cc9aaba[393]:   File "/usr/local/lib/python3.8/asyncio/coroutines.py", line 124, in coro
Dec 21 20:22:04 homeassistant fb413cc9aaba[393]:     res = func(*args, **kw)
Dec 21 20:22:04 homeassistant fb413cc9aaba[393]:   File "/usr/local/lib/python3.8/site-packages/serial_asyncio/__init__.py", line 410, in create_serial_connection
Dec 21 20:22:04 homeassistant fb413cc9aaba[393]:     ser = serial.serial_for_url(*args, **kwargs)
Dec 21 20:22:04 homeassistant fb413cc9aaba[393]:   File "/usr/local/lib/python3.8/site-packages/serial/__init__.py", line 88, in serial_for_url
Dec 21 20:22:04 homeassistant fb413cc9aaba[393]:     instance.open()
Dec 21 20:22:04 homeassistant fb413cc9aaba[393]:   File "/usr/local/lib/python3.8/site-packages/serial/serialposix.py", line 268, in open
Dec 21 20:22:04 homeassistant fb413cc9aaba[393]:     raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
Dec 21 20:22:04 homeassistant fb413cc9aaba[393]: serial.serialutil.SerialException: [Errno 2] could not open port /dev/serial/by-id/usb-FTDI_P1_Converter_Cable_P17UI81-if00-port0: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-FTDI_P1_Converter_Cable_P17UI81-if00-port0'

agners, 2020-12-22:
Uh yeah that make sense. So did the serial device name `/dev/serial/by-id/usb-FTDI_P1_Converter_Cable_P17UI81-if00-port0` change?

rck777, 2020-12-23:
I did not connect any USB devices. So, there was actually no device. Only in the config. It seems a solution that a non-existent device ignored in de configuration? 

i removed the integration, insert P1 USB cable en enable config
sensor:
  - platform: dsmr
    port: /dev/serial/by-id/usb-FTDI_P1_Converter_Cable_P17UI81-if00-port0

The serial device name has remained the same.
Right now it works

agners, 2020-12-28:
> Right now it works

So I guess it was the integration then which caused the high CPU/RAM usage?

rck777, 2020-12-29:
The migrated instance form PI4 to ODROID N2 is very unstable at this moment. After remove and installation DSMR it goes a bit better but still crashes after about 12 hours. With debugging options are available to debugging automations?

pznamenskii, 2021-01-11:
I believe I am experiencing the same, though I originally though it was caused by the HA update rather than migration from RPi3 to N2+: https://github.com/home-assistant/core/issues/44691

few hours after startup, HA is getting real slow: automations fire up with huge delay, my zigbee buttons react very poorly; and in around 12-15 hours goes totally unresponsive: nothing works anymore, even web UI, so I have to physically reboot N2.

pznamenskii, 2021-01-11:
I might have found the cause:

ERROR (Recorder) [homeassistant.components.recorder] Error executing query: (sqlite3.DatabaseError) database disk image is malformed

removed config/home-assistant_v2.db and CPU+memory usage is way better now. I will monitor if the freeze issue is gone completely.

rck777, 2021-01-11:
I also regularly have a corrupt database file. The problem remains even after deletion and restart. I am interested in your findings

pznamenskii, 2021-01-15:
3.5 days so far, everything looking good. all I did was removing the corrupted database.

closed 2022-01-11 by github-actions[bot]