home-assistant/operating-system #1116: Fail to boot NUC after upgrade to 5.9

haniffm, 2020-12-22:
**Hardware Environment**
- [X] Intel NUC 10i3

**Home Assistant OS release:**
- [X] Updated from version 5.8

**Description of problem:**

NUC not starting after upgrade from OS 5.8 to 5.9. I get the following error message on boot:
```
bootchooser: No valid targets found: ...
```
I've also tried to boot from a USB flash with fresh HA OS 5.9, with the same error.
[image]

agners, 2020-12-22:
Hm, it looks like our bootloader Barebox can't find a bootable image.

Can you check what is on the first boot partition?

haniffm, 2020-12-22:
This is on the boot image.
[image]

aqualx, 2020-12-22:
Same issue with NUC D5425 :( "Reboot and Select proper Boot device". IMHO 5.9 version for NUC must be blocked for update!
Is there any possible quick fix without reinstalling?

agners, 2020-12-22:
@haniffm are the file sizes ok (e.g. not 0 byte?) in particular /EFI/barebox/state.dtb?

@aqualx I did update Intel NUC just fine here, so its really not clear to me why it fails in some situation. Did you have to power it off or something?

Are the files corrupted in your case?

If the boot partition is corrupted, restoring it from the image should be possible. Probably the easiest method is using an Ubuntu Live image on the Intel NUC. Download the Home Assistant OS image and extract the downloaded image using `unxz hassos_intel-nuc-5.9.img.xz`. Then use "Disks" to mount the boot partition of the image:
[image]

You then can mount it using the "Play" symbol. You then 
[image]

Now you should have access to the FAT boot partition of a fresh installed image. Copy the files to the boot partition of your Intel NUC.

aqualx, 2020-12-23:
Thanks!
hassos-boot partitions only contained some log files... Copied EFI folder and hassos booted.

PS: I rebooted PC after 0.5-1 hour after hassos update.

haniffm, 2020-12-23:
@agners Thx for the help, but that did not work for me. I can also add that the size of the the `/EFI/barebox/state.dtb` was about 1.2 KB.

Any other ideas?

agners, 2020-12-26:
@haniffm so you restored the whole boot partition and still get the "no valid targets" message?

haniffm, 2020-12-26:
@agners that is correct. I mounted the partition as you discribed above and copied the entire hassos-boot partition sith `cp -r ...` command.

atze09, 2020-12-30:
Same issue here need to do a fresh installation of 5.9.

orohde, 2021-01-05:
I have the same issue. I restarted my NUC today and did not come up again. Upgraded to 5.9 a few days ago. I get the same error as above but I am only able to read it if I hold the Enter button, otherwise the screen is turning black right away.

haniffm, 2021-01-07:
I'm guessing that this happened to me because I was a little bit impatient and forcefully rebooted the NUC. I have solved my problem by moving from HA OS install to a self maintained OS and HA in docker, because I don't want to experience this kind of downtime ever again.

However, I can still assist if there is a need for debugging this issue further (have the old install on a different drive).

superm1, 2021-01-08:
Those affected, can you try `bootchooser -s` to reset the failed boot counter?

closed 2021-05-12 by stale[bot]