home-assistant/operating-system #1117: Odroid N2+ CPU fan trip temperature much higher than stock linux image

kevdliu, 2020-12-22:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [x] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [x] Fresh installation of release 2020.12.0
- [ ] Updated from version x.y
- Additional information (if accessible):

## System Health

version | 2020.12.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.6
os_name | Linux
os_version | 5.9.15
arch | aarch64
timezone | America/New_York

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | January 18, 2021, 7:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.9
-- | --
update_channel | stable
supervisor_version | 2020.12.7
docker_version | 19.03.13
disk_total | 13.6 GB
disk_used | 3.0 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.2.0), Terminal & SSH (8.10.0), Signal (10.10.0), Aux Speaker (1.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
mode | storage
views | 1
resources | 0

</details>

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

The 2 temperature thresholds which controls the CPU fan connected to the 2pin fan connector on the Odroid N2+ are set to 85C (/sys/devices/virtual/thermal/thermal_zone0/trip_point_0_temp) and 95C (/sys/devices/virtual/thermal/thermal_zone0/trip_point_1_temp) on the Home Assistant OS image, which seems very high to me. From reading [this thread](https://forum.odroid.com/viewtopic.php?p=300594#p300594), it looks like the defaults on the stock linux image are 65C and 75C. If the Home Assistant OS values were not changed intentionally, it would be great to lower them to match the stock linux image.

agners, 2020-12-22:
Thanks for the report, yeah I think lowering the tripping points does make sense. However, I wonder did you actually reach 65°C in practice? With that huge heat sink I haven't seen it going above ~45°C so far.

agners, 2020-12-22:
It seems that current upstream Linux doesn't support the fan at all. So first we would need to have to add support for a fan. It seems that Armbian added support, I would have to look into that patch (https://github.com/armbian/build/blob/master/patch/kernel/meson64-current/hardkernel-igor-0091-Add-fan-support-to-Odroid-N2-plus.patch).

kevdliu, 2020-12-23:
Thanks for the update! It didn't occur to me that the super high values could be caused by lack of kernel support. I haven't personally seen it go above 65C yet since I just set this up a few days ago, but my place gets absurdly hot during the summer so I can see it going past that under sustained load.

rliessum, 2021-03-02:
> It seems that current upstream Linux doesn't support the fan at all. So first we would need to have to add support for a fan. It seems that Armbian added support, I would have to look into that [...]

Talked about this on discord and this was the exact conclusion. For now, my fan is and stay idle then ;-)

hkrob, 2021-04-15:
Has any progress been made on this? 
This is probably my main reason for sticking to Armbian for now... Wouldn't mind switching to a 'supported' installation...

mdewyer, 2021-06-06:
Would like to see this supported going into summer here.

agners, 2021-06-07:
Fan support with a tripping point of 65°C is in the current dev branch. So with the current 6.0.rc3 and the upcoming final release a fan should work.

closed 2021-06-07 by agners

bakerkj, 2021-11-29:
Would it be possible to make the tripping point configurable?

petebowden, 2022-03-03:
@agners Has this made it's way into the release? I'm running 7.4 and the above defaults are still present. And agreed with @bakerkj it'd be nice to make the tripping point configurable.

agners, 2022-03-03:
Basic fan support should be in 7.4, the fan should start at 65°C.