home-assistant/operating-system #1119: HASS unstable

sidwin9, 2020-12-22:
## The problem
Since the 2020.12.0 update and the 2020.12.7 Supervisor update, my Hass has been drinking 1 or 2 a day and the interface is no longer accessible either locally or remotely. I have to make an electric power stop to restore. The 2020.12.1 version does not solve my problem

## Environment

- Home Assistant Core release with the issue: 2020.12.0
- Last working Home Assistant Core release (if known): 0.118.5
- Operating environment (OS/Container/Supervised/Core): HASS os 5.8/2020.12.7 Supervisor/HASS 2020.12.1/PI 3B+
- Integration causing this issue: 
- Link to integration documentation on our website: 

## Problem-relevant `configuration.yaml`

```yaml

```

## Traceback/Error logs

```txt

```

## Additional information

Gawronnek, 2020-12-23:
I uploaded the 5.9 version in the evening, unfortunately in the morning I already had the first system overhang, it was necessary to disconnect the power supply and reconnect, unfortunately I return to version 4.2 again

bschatzow, 2020-12-23:
Mind froze after 10 hours. In one of the forums another user reported that 5.2beta works for him and all after this fails.

SBRK, 2020-12-23:
Weirdly, I haven't had any crash since I updated to OS 5.8 and Core 2020.12.1

HumanSkunk, 2020-12-23:
I currently moved back to 5.6 HadsOS (moved from 5.8) and core 0.118.5 from 2020.12.1, and all working stably at the moment. Fingers crossed it all stays up!

plmilord, 2020-12-23:
I experienced a lot of instabilities in past days (after updated to 5.8 and 5.9). I deleted "home-assistant_v2.db" and everything goes fine since 24h!

danielsjf, 2020-12-23:
Same issue here after upgrading to 5.8. Subsequently upgrading to 5.9 doesn't fix the issue.

To add something on the discussion, the upgrade seems lan related. For some reason after upgrading I also had a couple of other devices complaining about dhcp errors. When I unplug the pi, the problems disappear.

Also, while hassos is nowhere to be found on the network (pinging doesn't work), zwave and zigbee continue to work. Sensors that only use those still respond ad they should.

Power cycling the pi solves all issues for a day or so.

Minius90, 2020-12-24:
Same issue here: Have to unplug the pi every day. 5.9 did not fix the issue

HumanSkunk, 2020-12-24:
> Same issue here after upgrading to 5.8. Subsequently upgrading to 5.9 doesn't fix the issue.
>
> To add something on the discussion, the upgrade seems lan related. For some reason after upgrading I [...]

Yes when the issue happened for me all my devices were unable to connect to the internet. Sonos/Alexa would all think they had connectivity but were unable to resolve connectivity. When I logged into my router it would also tell me that I had no WAN connectivity. As soon as I unplugged or hard reset the Pi, everything came back instantly.

I am currently remote from home and I am now unable to remotely VPN into my router or home assistant. My hope it is just an internet issue, but my suspect is that it is home assistant related. I hoped that rolling back would have fixed it, but looks like it hasn’t.

HumanSkunk, 2020-12-24:
> @HumanSkunk `journald -b 1 -u hassos-supervisor` for supervisor logs of the previous boot
> 
> `journald -b 1` for the host logs of the previous boot
> 
> These need to be run on the host, and not from an add-on.
> 
> https://wiki.archlinux.org/index.php/Systemd/Journal#Filtering_output

So another update from me, as I have recently installed some DIY blind motors I was concerned that something had gone wrong so I went home to investigate, low and behold Home Assistant has crashed exactly as before. Taken my internet down with it. I could still connect to my router and it was reporting no internet connectivity as before, the second I unplugged it... internet restored. As pointed out before OS 5.6 and Home Assistant version 0.118.5 with supervisor 2020.12.7 is still causing crashes.

I am unable to review the previous boot logs as the journald command doesnt work on my Pi, via ssh directly on the host. I get the error:

```
-bash: journald: command not found
```

I am going to try rolling back the OS to 5.2 (which I think it what it was for ages and was stable on my Pi4 running via an SSD. Fingers crossed.

Minius90, 2020-12-26:
Stepped back to 5.7, hoping daily need for unplugging my pi is "solved" this way

danielsjf, 2020-12-26:
Roled back to 4.20 since three days and all my lan devices are working properly and HA hasn't gone offline since.

matejzero, 2020-12-28:
Another case here. After upgrading to 5.8 or 5.9, don’t remember 100%, I had a rpi freeze every night. Downgraded to 4.10 and we have a smooth sailing ever since.

bschatzow, 2020-12-28:
I went to 5.2 and up 8 days. 5.8 or 5.9 froze daily

kpcz, 2020-12-28:
Same problem. After HassOS upgrade from 4.20 to 5.9, my HA has already froze twice in 3 hours. Even ping is not going through. In logs no traces of any issues. Before that everything was working like Swiss watch. :) RPi4 + SDcard.  Rolling back to 4.20 hoping that it will resolve the issue for now. Thanks.

agners, 2020-12-28:
This is all likely connected with #1129.

BarBaar44, 2020-12-29:
Same here on an Odroid C2 with both 5.8 and 5.9

The machine randomly crashes within 24h after boot. Filesystem goes to read-only. 

Now back on 4.2 and everything seems OK again..

HumanSkunk, 2020-12-30:
Upon moving back to 5.3 almost a week ago, my instance has remained up. This was the version I installed fresh when I updated to a Pi 4 with an SSD. Unsure why all later versions cause the Pi to lock up and get super hot, but i'll take this as a solution for now.

cogneato, 2020-12-30:
For most, I believe it is this issue: https://github.com/home-assistant/operating-system/pull/1121

bcutter, 2020-12-31:
Seeing this as many other users (exactly: https://community.home-assistant.io/t/hass-os-becomes-unresponsive-then-almost-unusable-and-is-finally-dead-starting-every-10-hours-after-last-ha-start/261800) as well.
- one side: not sure if it´s related to HASS OS 5.8, system ran fine for a few days before (and not sure when I updated the HASS OS)
- other side: Raspberry Pi 3 B+. When the lock storm starts (CPU load-1/-5/-15 goes up) in Glances I see "CPU_IOWAIT" warnings (value 50 or higher).

My system is heavily unstable and almost unusable. Started few days ago I think when I integraded InfluxDB and downsized recorder database (SQLite). HA only ran about 10 hours, today I saw it running for only 3 hours.

What a nightmare! Any advices?

Did you regain a stable system with running an earlier HASS OS version? Feedback please @HumanSkunk @BarBaar44 @kpcz @matejzero @Minius90 @bschatzow @danielsjf.

bschatzow, 2020-12-31:
Mine is very stable on 5.2.  5.8 and 5.9 froze after less than 10 hours. I downgraded to 5.2 from cli and all is well again.

HumanSkunk, 2020-12-31:
For me 5.3 works perfectly. It crashed on Christmas Eve running 5.6, but since the rollback to 5.3 it has remained operational for a week. I have my fingers crossed for longer.

Minius90, 2020-12-31:
Went back to 5.7 some days ago, seems pretty stable. Only had issues today after upgrading core to 2020-12-2. Propably needs some further testing

hughhallhh56, 2021-01-01:
Running OS version 4.16 and Core version 0.117.6.  I have been having this issue for more than a month.  Sometimes goes a few days, other times less than 12 hours until system become non responsive.  Was able to catch it a few times before it completely hung and found the load average (using top command) to be 18 to 25 instead of the normal .5.  Within another hour it was completely hung. 

I unloaded a few add ons that were large memory users such as the unifi controller since I was using a lot of the available memory.  I was going to upgrade OS and Core, but in searching for what was causing my issues I came across all these posts describing the same issue so have not updated anything further.  System has been great for 24 hours now so will continue to watch it.

bcutter, 2021-01-01:
18 to 25, hah, see one of the last snapshots before my system went to death:

[image]

So many affected users - where are the developers or debugging experts by the way? I thought that´s one big advantage of using a ***supported*** version, otherwise I could have gone with HA Supervised on a Raspberry Pi OS which would give me much more insights what´s going on on the system as I can check now on HASS OS. Looking forward... andy happy new year by the way to everybody! :-)

bcutter, 2021-01-01:
Still on HASS OS 5.8. I have no idea where *that massive peak* came from and WHY the system survived it:
[image]
[image]
[image]

Actually the whole system ran for almost 24 hours - first time for days. Only planned HA restart after ~ 16 hours after
[image]

Only thing I changed yesterday: stopped all HA supervisor addons not urgently needed to free some RAM. That way system RAM usage stayed below 80 % most of the time.

Everything only assumptions, don´t have more to offer unfortunately.

fernplant, 2021-01-01:
> Same issue here after upgrading to 5.8. Subsequently upgrading to 5.9 doesn't fix the issue.
>
> To add something on the discussion, the upgrade seems lan related. For some reason after upgrading I [...]

Am also having this issue on RPi 4 w/ SSD and on mine, when it locks up, the network activity light goes bananas blinking like crazy. So I would say yes, LAN related seems likely

liudger, 2021-01-02:
New ha os version is realesed.
Anyone want to try it out and sees if it's stable?
 `ha os update --version 5.10`

bschatzow, 2021-01-02:
I'll try it now and see what happens. It is not showing up under supervisor on my system yet.  Updating from cli at 7:45 AM EST.  Been up over 10 days on 5.2.  Hope it works.

Gawronnek, 2021-01-02:
I installed it for a trial of 5.10, I will watch if the situation repeats
[image]

bcutter, 2021-01-02:
Updating currently to 5.10 too. But I doubt it improves things, only bugfix included was for a bug introduced with 5.9 (see https://github.com/home-assistant/operating-system/issues/1129). As I´m still on 5.8 chances are almost zero it will run more stable/longer.

mloebl, 2021-01-02:
I think maybe add me to the list too. For the first time I had HA go offline and was trying to figure out what happened. I noticed a few days ago HA had gone totally non-responsive, kept turning some devices on from a command I did an hour earlier. I looked at the status, and many entities hadn't updated in over an hour.  It finally died, and was sitting on a similar message as posted above when I plugged the monitor on to the Dell 7040 Micro I use.  I installed Glances and watched as the homeassistant container steadily grew by 10mb every few seconds until it used all available ram.  I disabled quite a few addons and some of things I thought may be causing it, but still can't find it. I added 16gb additional RAM to the system and seems to have extended that period before it dies, and HA will happy consume all the RAM in this cycle over and over until it either does some sort of cleanup or hang.  I have been running stably for months, and started sometime around 2020.12.x when I noticed.   *Most* of the time for me it seems to recover after all ram is consumed, sometimes it crashes.
[image]
[image]

liudger, 2021-01-02:
> Updating currently to 5.10 too. But I doubt it improves things, only bugfix included was for a bug introduced with 5.9 (see https://github.com/home-assistant/operating-system/issues/1129). As I´m still on 5.8 chances are almost zero it will run more stable/longer.

I was thinking the same. But there was also a firmware update for odroid. So maybe this resolve the issue for odroid n2?!?

bcutter, 2021-01-02:
> I was thinking the same. But there was also a firmware update for odroid. So maybe this resolve the issue for odroid n2?!?

In case it does: congratulations :-) Me and others (most of users I found on community forums) run HASS OS on a Raspberry Pi.

I mean we don´t even know what causes it, right? So it might be platform dependent, it might not.

HASS OS 5.10 running for 5 hours now, next 10 to 12 hours will give first indications if 5.10 changed anything.

bschatzow, 2021-01-02:
Went to 5.10 at 7:45, just notice it at 1:45 that it froze again.  Going back to 5.2.

bcutter, 2021-01-02:
That being said I started InfluxDB to have a look at some long-term remaining memory numbers.

RAM used and load went straight up and system became unresponsive for a few minutes
[image]

Supervisor log gave 
```
21-01-02 18:51:44 WARNING (MainThread) [supervisor.misc.tasks] Watchdog/Application found a problem with observer plugin!
21-01-02 18:53:28 INFO (SyncWorker_3) [supervisor.docker.interface] Stopping hassio_observer application
21-01-02 18:54:16 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning hassio_observer application
21-01-02 18:54:33 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
```

- No idea what hassio_observer docker container is for and why it failed.
- Supervisor log did not refresh for minutes, Portainer could not receive information of docker environment anymore and so on. Felt like a "okay whole system will fail completely shortly".
- Kind of recovered a bit but only time tells if it was just a short hickup (due to memory limitations of the Pi 3 B+) or the beginning of well-known system freeze.
- Anyway it tastes like some kind of RAM issue to me.

mloebl, 2021-01-02:
@bcutter 
> * Anyway it tastes like some kind of RAM issue to me.

This is what I am thinking too.  Memory is used up at a rate of 10mb every few seconds on my HA system until it fills up and basically dies or recovers.

Observer, came out a couple of months ago; https://www.home-assistant.io/blog/2020/09/16/supervisor-joins-the-party/ , spefically https://github.com/home-assistant/plugin-observer
`The latest plugin added is the Observer plugin. The observer keeps an eye on the Supervisor. `

sidwin9, 2021-01-02:
Hey,
As the update to version 5.10 is not available via the console, how can we update ?

mloebl, 2021-01-02:
Now that I've been running for awhile today since resetting last night, this very much looks like a memory leak, growing exponentially as it goes:
[image]
When I only had 8gb, it was never this pronounced. The homeassistant container started at under 500mb, and now over 11gb 
and growing.

bschatzow, 2021-01-02:
It seems that all that had problems with 5.8 / 5.9 are still having issues with 5.10.  Is there any information that I can post that can help establish the root cause?  Also something else to that may help,  I had no issues with OS version 5.8 with the SD memory.  My issue only happens with the SSD.  Not sure if this is related?

bcutter, 2021-01-02:
One basic question facing the fact that running HASS OS with an officially provided image on an officially supported hardware is a *supported setup*:

Where is that "support"? I'm quite new to the HA game but I'm missing the experts, the devs, the integrators, the ones who build HASS OS. The ones really knowing how things work and what changes might have caused issue A or B. The ones asking users to test or change or debug things. The ones who finally fix issues like this massive one.

I'm sorry but I pretty much feel like even GitHub (dev corner, right?) is just a user forums where lots of people report their problems (which is okay), but no guru is mentoring and working on them. After spending (wasting) four (!) full holidays - which were intended to build up my smart home - only for all that HASS OS trouble I feel pretty lost. How can I have fun and build up my smart home whit that really great platform when the system base, the underlying OS is screwed up?

Maybe someone can prove me wrong or just tell me how things work in this repo. All I see so far is people giving lots of information, making assumptions (me explicitly included) and doing downgrades. This can't be it... rant over.

bcutter, 2021-01-02:
> It seems that all that had problems with 5.8 / 5.9 are still having issues with 5.10.  Is there any information that I can post that can help establish the root cause?  Also something else to that [...]

Have a look at https://github.com/home-assistant/operating-system/issues/1095 this seems to be focused on USB boot/SSD issue.

alex88, 2021-01-02:
> One basic question facing the fact that running HASS OS with an officially provided image on an officially supported hardware is a _supported setup_:
> 
> Where is that "support"?

It only means that if you don't have the supported image/hardware you won't get any support at all, otherwise you get support whenever maintainers or other users have time.

In my case downgrading to 5.7 worked, HA has been working fine for 3 days (5.9 was freezing every day). You won't miss any features since it's just the OS to be downgraded

denilsonsa, 2021-01-03:
**TL;DR:** Same complaints/symptoms as other people. May I suggest disabling the coredump feature?

---

I'm running HA on RPi 3B+. A few days ago (2020-12-23) I started having issues after I updated from 5.8 to 5.9: https://discord.com/channels/330944238910963714/330990055533576204/791392528405757972

Essentially, the system couldn't boot properly. After I connected the device to a display (and disconnected from the network, because I didn't have both Ethernet and HDMI at the same place), I started seeing the boot messages. Here's a summary:

* Decompression errors on zram [1](https://discord.com/channels/330944238910963714/330990055533576204/791401731417440266) [2](https://discord.com/channels/330944238910963714/330990055533576204/791431929361858580) [3](https://discord.com/channels/330944238910963714/330990055533576204/791437745662722079)
* Many `FSCK0xxx.REC` files on the first partition [1](https://discord.com/channels/330944238910963714/330990055533576204/791405581399359498)
* Boot gets stuck for several minutes (even more than 10 minutes) [1](https://discord.com/channels/330944238910963714/330990055533576204/791418207306907658)
* Out-of-memory killing some processes [1](https://discord.com/channels/330944238910963714/330990055533576204/791418385959223346) [2](https://discord.com/channels/330944238910963714/330990055533576204/791420613192712202) [3](https://discord.com/channels/330944238910963714/330990055533576204/791429492089683968)
    * systemd-resolve
    * rauc
    * networkmanager
    * systemd-timesync
    * wpa_supplicant
    * coredump [1](https://discord.com/channels/330944238910963714/330990055533576204/791432425737289730)
    * dockerd
* Randomly, by sheer luck, it managed to boot and show a login prompt. But trying to login never yields a login shell, likely because the main shell is based on `ha` and might require HA core to be running fine, or because the system is under very high load. [1](https://discord.com/channels/330944238910963714/330990055533576204/791426417799462942) [2](https://discord.com/channels/330944238910963714/330990055533576204/791433365232418837) [3](https://discord.com/channels/330944238910963714/330990055533576204/791436939541741618)

So, pretty much the same complaints already mentioned by other people. And it's near-impossible to investigate if I can't get a shell, and also because many issues happen even before the login prompt.

---

However, I noticed something that other people might have missed. The OOM killed the `coredump` process (twice!). So I have a theory:

1. System is unstable, some processes crash.
2. A process has crashed? Oh, well, let's save a coredump.
3. The coredump process creates even higher CPU and I/O load, making the system more unresponsive and causing more crashes and OOM.

So… Can we disable coredumps in HASS OS? (That won't prevent the initial crashes, but will prevent making the system worse.)

---

I just noticed [HASS OS already disables it `/etc/systemd/coredump.conf`](https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/rootfs-overlay/etc/systemd/coredump.conf).

However, why the heck was `coredump` process killed during OOM?

HumanSkunk, 2021-01-03:
Is everyone here running off an SSD? I am just tying to identify something we all have in common. I see that not everyone is using a Pi4 so it is spanning multiple host types. I am happy to provide some more details on some logs or settings, but I am not technically savvy enough to understand how to debug this.

Running OS 5.3 has ‘solved’ my crashing issue as it’s been well over a week no. If it is a OS issue and not a Core issue then I may update as I currently went back to 0.118.5 when I downgraded. Is anyone running 2020.12.2 and not experiencing issues?

Gawronnek, 2021-01-03:
I use HassOS 5.10 with Home Assistant 2020.12.2 on a Raspberry Pi 3B + with a USB SSD drive, on HassOS 4.20 it ran steadily for several days, now after uploading the latest system at night he reported to me

Warning or critical alerts (last 4 entries)
2021-01-03 4:25:44 (00:00:26) - CRITICAL on CPU_IOWAIT (32.6)
2021-01-03 4:25:14 (00:31:10) - WARNING on MEMSWAP (89.4)
2021-01-03 4:24:46 (00:00:42) - CRITICAL on CPU_IOWAIT (43.7)
2021-01-03 4:23:35 (00:00:28) - CRITICAL on CPU_IOWAIT (28.6)

but it works without crashing, I will see what will happen next, when it crashes, I will put info here

gcb2018, 2021-01-03:
Copy/pasted my comment from: [Home Assistant Community](https://community.home-assistant.io/t/hass-os-becomes-unresponsive-then-almost-unusable-and-is-finally-dead-starting-every-10-hours-after-last-ha-start/261800/20?u=gcb2018)

**I am probably not adding anything useful but I have the same issue:**

* behavior: after full manual reboot it's a 5-7hrs before it crashes (I have uptime robot to monitor if it's up/down). Yesterday it crashed at 9pm and when I woke up this morning it was still down...
* checked all intergrations and removed/cleaned up all errors I saw (still crashes)
* turned off all automations (still crashes)
* turned off all auto-adds & discoveries (still crashes)
* moved from SD card to SSD (still crashes)
* SWAP full (99.4%) / RAM high (81,8%)
* I don't have many integrations / add-ons / devices actually
* Raspberry Pi 3b+
* logs shows this one sometimes:
```
2021-01-03 08:43:31 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 314, in data_received
    messages, upgraded, tail = self._request_parser.feed_data(data)
  File "aiohttp/_http_parser.pyx", line 546, in aiohttp._http_parser.HttpParser.feed_data
aiohttp.http_exceptions.BadStatusLine: 400, message="Bad status line 'invalid HTTP method'"
```

Read somewhere the crashing can be related to: 
* snapshot creation (and maybe the google drive back up add-on) and I have disabled that one for now. Will report back.
* it being the Raspberry Pi 3b+ and upgrading to a Pi4 would do the trick as RAM would max out at 1220 which would be too much for the 3b+

Btw: I stopped all add-ons that are not crucial for me at the moment so only have the following running:
* Node-RED
* Mosquitto Broker
* zigbee2mqtt

michaelarnauts, 2021-01-03:
I don't think it's related to the RPI. I'm running on a VM in Proxmox and I'm seeing related issues. The interface is still available, but lost of my integrations don't work anymore. I think everything that needs network access. Restarting core itself doesn't work. I always need to reboot the full VM.

bcutter, 2021-01-03:
5.10 changed nothing.

Last screenshots before whole system freezed and needed to unplug power to recover:

[image]

[image]

It started when my iOS device left flight mode after night and connected to HA.

Enough of this crap for me, will downgrade too.

gcb2018, 2021-01-03:
To what version will you downgrade @bcutter?

bcutter, 2021-01-03:
I'll go for 5.2. Last safe harbour probably 4.20.

Minius90, 2021-01-03:
> I'll go for 5.3. Last safe harbour probably 4.20.

Running 5.7 since the problems started. Works nice by now

bschatzow, 2021-01-03:
Maybe a list can be compiled on what works and doesn't?  When many have tried a version and no failures, this can be passed on to the developers to say these work and these don't and hopefully a comparison can be done to see what was changed.
Example I have a 
Startech USB3S2SAT3CB SATA to USB Cable
Kingston 120GB A400 SATA 3 2.5" Internal SSD SA400S37
OS 5.2 - Has never froze.
Not sure this helps, but I would think it world.

noide35, 2021-01-03:
Hello everybody,

After having a boot issue with Hass 5.x on my Pi3B+ with USB/SSD boot (#1095), I think I now have the same than you... It's the second time my Pi is crashing in two weeks since I am in HassOS 5.9 (with a hack using boot files of 4.20). Unfortunately, I don't have a screen connected to it to see exactly what happens when HASS is not responding.
I'm using the last version of HA Core 2012.20.1.

The only thing I can say when it happens :
- No more activity on ACT LED of the Pi.
- HASS UI is not responding anymore.
- The workaround is a hard reboot (unplug/plug power supply).

I'll go back in 4.20 and my last snapshot to see if it's more stable.

gcb2018, 2021-01-03:
> Copy/pasted my comment from: [Home Assistant [...]

Just wanted to update you guys in between: I am 12hrs since last reboot. all looking good still... 🤞

Besides the details above I am on 2020.12.7 and 5.9 with most add-ons turned off besides:
* Node-RED
* Mosquitto Broker
* zigbee2mqtt

Lectere, 2021-01-03:
I think I have the same; 

https://community.home-assistant.io/t/update-2020-12-7-5-9-broke-my-system/263648/8

But only worse; my system is up for about 5 minutes before it dies.

How can you downgrade to a previous version of Home Assistant OS?

alex88, 2021-01-03:
> How can you downgrade to a previous version of Home Assistant OS?

`ha os update --version 5.7` worked for me

HumanSkunk, 2021-01-03:
Install the terminal add on or via ssh enter

`ha os update —version <version>`

so replace <version> with what you want to downgrade to. Most people have had success with 5.2, but 5.3 works for me. So to move to 5.2 enter

`ha os update —version 5.2`

Lectere, 2021-01-04:
Thanks Alex/Skunk

I've tried both 5.2 and 5.7, both crashed with excessive memory use within minutes. 

So is there a way to downgrade HA to a version before 2020.12?

HumanSkunk, 2021-01-04:
> So is there a way to downgrade HA to a version before 2020.12?

`ha core update --version <version>`

will downgrade the home assistant version (again replace <version> with the release you want. Supervisor in think is just the same but replace core with supervisor. 

The fact you are crashing even on 5.2 May indicate something else is at fault here, was your install ever stable before updating? Sorry I won’t be able to provide any real debugging help as I am not an expert, but I know enough to get around install and configuration of Home Assistant.

Lectere, 2021-01-04:
Thanks, I'll try that. 

It was stable before the upgrade, it runned stable for over a year. I didn't update in a while, like 1 or 2 months or so.

[530 more replies]

closed 2022-01-03 by agners