home-assistant/operating-system #1120: oDroid N2+: # CONFIG_HIDRAW is not set

mvieleers, 2020-12-22:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [X] ODROID [C2/C4/N2(+)/XU4] (Home Assistant Blue device)
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [X] Updated from version 5.8 
- Additional information (if accessible): Home Assistant Blue came with 5.8, updated to 5.9

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

agners, 2020-12-22:
Just wondering, what is your use case?

mvieleers, 2020-12-22:
@agners I have an Alarm System (Jablotron 100). The system is connected to home assistant using a usb cable. Using the HACS package (https://github.com/kukulich/home-assistant-jablotron100) to read the status of devices and to interact with it.

chadlyy, 2020-12-30:
Wyze Sense also uses hidraw:  https://github.com/kevinvincent/ha-wyzesense

agners, 2020-12-30:
Fix will be in 5.10 onwards.

closed 2020-12-30 by agners

chadlyy, 2020-12-30:
What is the targeted release date for 5.10?  Is that also the first Wednesday of every month? Thank you! :)