home-assistant/operating-system #1124: missing ti_3410.fw

djcelvin, 2020-12-22:
unable to recognize a Honeywell HGI80 due to missing firmware.
Here is an extract of dmesg:

[474533.101225] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
[474533.255347] usb 1-1.3: New USB device found, idVendor=10ac, idProduct=0102, bcdDevice= 1.01
[474533.255367] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[474533.255385] usb 1-1.3: Product: TUSB3410 Boot Device
[474533.255402] usb 1-1.3: Manufacturer: Texas Instruments
[474533.255418] usb 1-1.3: SerialNumber: TUSB3410
[474533.307281] usbcore: registered new interface driver ti_usb_3410_5052
[474533.307368] usbserial: USB Serial support registered for TI USB 3410 1 port adapter
[474533.307449] usbserial: USB Serial support registered for TI USB 5052 2 port adapter
[474533.307537] ti_usb_3410_5052 1-1.3:1.0: TI USB 3410 1 port adapter converter detected
[474533.307682] usb 1-1.3: Direct firmware load for ti_usb-v10ac-p0102.fw failed with error -2
[474533.307725] usb 1-1.3: Direct firmware load for ti_3410.fw failed with error -2
[474533.307737] usb 1-1.3: ti_download_firmware - firmware not found
[474533.307798] ti_usb_3410_5052: probe of 1-1.3:1.0 failed with error -2

agners, 2020-12-22:
The firmware seems to be required for the rather generic TI USB to UART converted. I think it make sense adding the firmware to the OS to support them.

agners, 2020-12-30:
Fix will be in 5.10 onwards.

closed 2020-12-30 by agners

djcelvin, 2021-01-03:
Thanks a lot and happy new year ;)

djcelvin, 2021-01-08:
Hello, i confirm the hardware is recognized correctly now.
Thx!!

silabs-LasseM, 2021-01-10:
Hi Guys,
I have a USB to RS485 converter which was working yesterday _(I may have been on some 5.9 beta upgrade image to 5.10 i'm not totally sure)_ but after reflashing the latest 5.10 image from hass.io this morning it doesn't work any longer.

(I have tried both RPI3 32 and 64bit images) I can confirm that this issue still persists (no driver found for TI 3410)
The console output is:
usb 1-1.1.2: ti_download_firmware - firmware not found

Yours truly
Lasse