home-assistant/operating-system #1127: Error: Failed to fetch" when restoring snapshot on fresh install

fyrmangeoff, 2020-12-24:
HassOS release with the issue:

QCOW2
2020.12.1

Supervisor logs:

Journal logs:

Kernel logs:

Description of problem:

Reinstalled HassOS in the clean version of Proxmox using hassos_ova-5.9.qcow2 and attempted to upload a snapshot and received a "Fail to fetch" notice. I have re installed Proxmox several times as well as Hassio. It will allow a new set up.

mj-sakellaropoulos, 2020-12-24:
Did not get the same issue, but maybe its an issue with browser ? Try to upload the snapshot (.tar file) directly into the "backups" folder using Samba (hassio addon), SSH or similar. (make sure to press the top right three dots menu and reload)

fyrmangeoff, 2020-12-24:
MJ did it even on a Pi. The backup is coming off of a Ubuntu and one version old could that cause it? It begins to take it and then Fail's.

mj-sakellaropoulos, 2020-12-24:
That sound like a real bug with the snapshot process, I wouldn't know much more than that 👍