home-assistant/operating-system #1128: Image hacked. Why is there dutch bank information in it

daemenseth, 2020-12-24:
I download the vmdk files for hassio 5.8 and 5.9
When I only start it and did no configuring. I find some strange dutch bank files in it
see screenshot

https://www.screenpresso.com/cloud/bK2Od/

rbray89, 2020-12-24:
Lookup MT940, and look at the directory those files are in. They are test
files for a python MT940 parser library.

Home assistant is more more hacked than your github account. Please do more
research before making accusations or sowing misinformation.

daemenseth, 2020-12-24:
Ok sorry. Maybe I was too fast
Buy they look strange to me. On a another test I dont der dutch banks but german Banks.
And last week my firewall was reporting some p2p connections from my hassio box.
In a old home assistant installation. Is dont see these files.

Where are mt940 for used in hassio?

rbray89, 2020-12-24:
If I had to guess, MT940 is probably included as a dependency from another
package.

agners, 2020-12-28:
Yeah those files are part of the Home Assistant Core container. You can use `pipdeptree` in a Core devcontainer development environment to find which package pulled it in as a dependency:
```
$ pipdeptree
...
fints==1.0.1
  - mt-940 [required: Any, installed: 4.23.0]
```

Closing, as those files are part of Home Assistant Core to fulfill this dependency.

closed 2020-12-28 by agners