home-assistant/operating-system #1129: OS 5.9: Raspberry pi 3 freezes, zram decompression failed

cuveland, 2020-12-26:
**Hardware Environment**

- [X] Raspberry Pi 3

**Home Assistant OS release:**
- [X] Updated from version 5.8
- Additional information:

```
arch: armv7
channel: stable
docker: 19.03.13
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.9"
homeassistant: 2020.12.1
hostname: hassio
logging: info
machine: raspberrypi3
operating_system: Home Assistant OS 5.9
supervisor: 2020.12.7
supported: true
supported_arch:
- armv7
- armhf
timezone: Europe/Berlin
```

**Supervisor logs:**

[supervisor_log.txt](https://github.com/home-assistant/operating-system/files/5742522/supervisor_log.txt)

**Kernel logs:**

[kernel_log.txt](https://github.com/home-assistant/operating-system/files/5742524/kernel_log.txt)

**Description of problem:**

Occasionally, the Raspberry Pi freezes completely (even the numlock LED ceases to work on a local USB keyboard) and requires a power cycle. Also, with heavy use, processes occasionally terminate unexpectedly.

This problem has started either with Home Assistant OS Release-5 build 9 or build 8. The full freeze first occurred on build 9 for me, but the crash of a docker container's process might have also happened before. I had been operating this setup with stable HA OS releases for more than a year now without any of these problems.

Error messages related to the zram appear in the kernel log and on the local console:
```
[ 1472.597586] zram: Decompression failed! err=-22, page=32663
[ 1472.597883] zram: Decompression failed! err=-22, page=32663
[ 1472.598047] Read-error on swap-device (254:0:261304)
[ 1590.459156] zram: Decompression failed! err=-22, page=10855
[ 1590.459375] zram: Decompression failed! err=-22, page=10855
[ 1590.459474] Read-error on swap-device (254:0:86840)
[ 1735.075095] zram: Decompression failed! err=-22, page=5410
```

**Additional remarks:**
This problem might to be related to the current Linux kernel and could affect a number of arm devices. According to this email, disabling the CONFIG_ZSMALLOC_PGTABLE_MAPPING in the Linux kernel config may resolve the problem: https://www.spinics.net/lists/stable/msg428110.html

ArjanTel, 2020-12-26:
Following, having the exact same issue. Pi 3B+.
I also receive messages that it runs out of memory and is killing tasks
And this:
`systemd-coredump[313]: Failed to get COMM: No such process`

After this it freezes indefinitely

agners, 2020-12-28:
@cuveland thanks for the report and the link to the mailing list thread. It indeed looks very much like what you are seeing. It seems that in Linux 5.4 (what we are using for Raspberry Pi) the config symbol was called `CONFIG_PGTABLE_MAPPING`, and the config is indeed enabled in Home Assistant OS. However, it was already enabled in release 4 (Linux 4.19)... So if it is this, something else must have changed with 5.4 which made the problem start to appear.

I'll do a development build with the config disabled then we can test if this indeed fixes the problem.

agners, 2020-12-29:
@cuveland or @ArjanTel can you try the latest version on dev channel to see if it fixes the problem? Make (and download) a snapshot before  trying, just to be on the safe side.

```
ha supervisor options --channel dev
ha supervisor reload
ha os update --version 6.0.dev20201228
```

Don't forget to switch back to the stable channel afterwards:
```
ha supervisor options --channel stable
```

cuveland, 2020-12-29:
@agners Thank you very much for the kind feedback and the quick response!

I would like to test the dev build. Unfortunately, I'm currently on the go and don't have physical access to the setup. If anyone else has had the same problem and can test in the short term, I'd be happy - otherwise I'll get back to you in a little over a week.

Thanks again for the quick response and a great project.

PabloAriza, 2020-12-29:
Hello

I've just applied your patch @agners in a Raspi 3b+ + SD card up-to-date. Tomorrow i'll tell you how it goes.

My father is also having the same problem but his system is a Raspi4 2GB + SSD Can I apply the same patch?

Thanks

agners, 2020-12-30:
> My father is also having the same problem but his system is a Raspi4 2GB + SSD Can I apply the same patch?

It is likely that RPi 4 was equally affected. Just make sure to have a backup and all, after all its a dev release... :-)

PabloAriza, 2020-12-30:
Morning,

The raspi 3b has just freeze, so this patch seems to not be enough.

If you need any log just tell me.

Thanks.

agners, 2020-12-30:
@PabloAriza was there anything on the screen when it freezed? The kernel log of the last boot would be interesting, e.g. from the OS shell:

```
journalctl -t kernel -b -1
```

PabloAriza, 2020-12-30:
Im sorry but i dont know where i can put that command because via ssh or via terminal its not working, I'm running the regular supervised image on the raspi.  

Meanwhile the web says just "loading"and nothing happens. The thing is if I unplug and plug the ethernet cable it recovers but some systems like "duckdns" service are stopped. More or less 12 hours later it freezes again.

regards and thanks

cbarreholm, 2020-12-30:
I have experienced the same kind of issue, where the system is totally frozen within roughly 24 hours, since upgrading many of the components before Christmas. I have now moved it to my monitor so that I can see that Pi has become unresponsive in the console as well, which clearly indicates OS level issues. 
Now it has been up for 30 hours after disabling Hey Ada and AppDaemon 4. I'm not really using Ada and only running two simple scripts in AppDeamon to use Ikea remote controls, so nothing really intensive.

Will let things run for another day to see if things work and then enable Ada & AppDeamon one by one too see if things break.

Running 
Supervisor 2020.12.7
HA OS 5.9 
on a Pi 4 4GB.

vahaldor, 2020-12-31:
Hi,
it seems that I have the same problem. My HA freezed four consecutive times each time over night. I just downgraded to HA OS 5.8 on PI 4 4GB.

if I can be of help in providing you some logs, let me know

agners, 2020-12-31:
This issue is related to freezes with zram decompression failure, so make sure to check if you have this errors in the Host log to make sure its this actual issue.

A fix for zram decompression failure will be in 5.10 onwards. If you still experience freezes with 5.10 and decompression error messages, please reopen. If freezes occure for other reasons, please create a new issue.

closed 2020-12-31 by agners

bcutter, 2020-12-31:
So when will 5.10 be available (on stable channel)?

bcutter, 2021-01-01:
General question: how long does it usually take until new stable HASS OS releases show up in HA? Any way to trigger them?

liudger, 2021-01-02:
`ha os update --version 5.10` should work

elbueno222, 2021-01-02:
> `ha os update --version 5.10` should work

It does work, thanks!!

bcutter, 2021-01-02:
Will give it a try too. But as mine's issues are on HASS OS 5.8 I think 5.10 won´t have anything shipped to improve my situation.

vahaldor, 2021-01-03:
Ok so, I downgraded to 5.8 and rpi had been stable for two days. Now I updated to 5.10 and it freezed.