home-assistant/operating-system #1130: Home assistant blue crashes after internet outage (repeadadly)

frog32, 2020-12-26:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [x] ODROID N2+
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:** 5.9
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):
## System Health

version | 2020.12.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.6
os_name | Linux
os_version | 5.9.15
arch | aarch64
timezone | Europe/Zurich

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4912
Installed Version | 1.9.0
Stage | running
Available Repositories | 764
Installed Repositories | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 17. Januar 2021, 1:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.9
-- | --
update_channel | beta
supervisor_version | 2020.12.7
docker_version | 19.03.13
disk_total | 113.9 GB
disk_used | 11.8 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | OpenZWave (0.7.1), IDE (2.2.0), SQLite Web (2.3.2), SSH & Web Terminal (7.8.0), Let's Encrypt (4.11.0), MQTT Server & Web client (1.2.0), File editor (5.2.0), NGINX Home Assistant SSL proxy (3.0.1), Mosquitto broker (5.1), Home Assistant Google Drive Backup (0.102.0), Duck DNS (1.12.4), ESPHome (1.15.3), RTL_433 to MQTT Bridge (0.2), AppDaemon 4 (0.3.2), Traccar (0.9.2), JupyterLab Lite (0.3.1), MariaDB (2.2.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
mode | storage
views | 5
resources | 8

</details>

**Description of problem:**
I just replaced my router since it might have a dram issue. I did not expect this to affect my home assistant installation on a brand new home assistant blue but everytime the router crashed home assistant completely broke and even deactivated the etc0 interface (light out on the switch). Unfortunately I do not have a ton of logs for that but attaching it to a hdmi screen showed that it is not able to read write to rtc.
[image]

What would I check next? Just out of curiosity, does Home Assistant Blue get shipped with battery?

frog32, 2020-12-27:
Fixed my internet but my blue is still going offline. Figured out that it has the following lines logged by the kernel when it goes offline:

```
Dec 27 11:31:34 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: Reset adapter.
Dec 27 11:31:34 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: PHY [0.0:00] driver [RTL8211F Gigabit Ethernet] (irq=31)
Dec 27 11:31:34 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: No Safety Features support found
Dec 27 11:31:34 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: PTP not supported by HW
Dec 27 11:31:34 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: configuring for phy/rgmii link mode`
```

frog32, 2020-12-27:
I am not entirely sure when this started but it is likely that it ran the first 2 days without problems on the OS that it was shipped on. I think I upgraded it to 5.9 just before this all started but since there are not logs from the upgrade I can not align it 100%. I will pull more kernel and Network Manager logs today and downgrade to 5.8 again to verify if that runs stable.

frog32, 2020-12-27:
Dec 27 16:31:49 homeassistant kernel: ------------[ cut here ]------------
Dec 27 16:31:49 homeassistant kernel: NETDEV WATCHDOG: eth0 (meson8b-dwmac): transmit queue 0 timed out
Dec 27 16:31:49 homeassistant kernel: WARNING: CPU: 4 PID: 0 at net/sched/sch_generic.c:442 dev_watchdog+0x300/0x310
Dec 27 16:31:49 homeassistant kernel: Modules linked in: xfrm_user xfrm_algo cfg80211 dw_hdmi_i2s_audio dw_hdmi_cec btusb btrtl btintel btbcm bluetooth ecdh_generic ecc rfkill pl2303 realtek dwmac_generic dwmac_meson8b stmmac_platform stmmac mdio_xpcs meson_dw_hdmi dw_hdmi meson_drm cec meson_canvas drm_kms_helper rc_odroid panfrost meson_ir rc_core gpu_sched rtc_meson_vrtc crct10dif_ce snd_soc_simple_amplifier cdc_acm drm
Dec 27 16:31:49 homeassistant kernel: CPU: 4 PID: 0 Comm: swapper/4 Not tainted 5.9.15 #1
Dec 27 16:31:49 homeassistant kernel: Hardware name: Hardkernel ODROID-N2Plus (DT)
Dec 27 16:31:49 homeassistant kernel: pstate: 20000005 (nzCv daif -PAN -UAO BTYPE=--)
Dec 27 16:31:49 homeassistant kernel: pc : dev_watchdog+0x300/0x310
Dec 27 16:31:49 homeassistant kernel: lr : dev_watchdog+0x300/0x310
Dec 27 16:31:49 homeassistant kernel: sp : ffff80001157bd90
Dec 27 16:31:49 homeassistant kernel: x29: ffff80001157bd90 x28: ffff0000dbe12940
Dec 27 16:31:49 homeassistant kernel: x27: 0000000000000004 x26: 0000000000000140
Dec 27 16:31:49 homeassistant kernel: x25: 00000000ffffffff x24: 0000000000000004
Dec 27 16:31:49 homeassistant kernel: x23: ffff0000da80439c x22: ffff0000da804000
Dec 27 16:31:49 homeassistant kernel: x21: ffff0000da804438 x20: ffff8000113d7000
Dec 27 16:31:49 homeassistant kernel: x19: 0000000000000000 x18: 0000000000000000
Dec 27 16:31:49 homeassistant kernel: x17: 0000000000000000 x16: 0000000000000000
Dec 27 16:31:49 homeassistant kernel: x15: ffff8000113f5000 x14: ffff8000114d816a
Dec 27 16:31:49 homeassistant kernel: x13: 0000000000000000 x12: ffff8000114d7000
Dec 27 16:31:49 homeassistant kernel: x11: ffff8000113f5000 x10: ffff8000114d77b0
Dec 27 16:31:49 homeassistant kernel: x9 : 0000000000000000 x8 : 0000000000000000
Dec 27 16:31:49 homeassistant kernel: x7 : 0000000000000004 x6 : 0000000000000003
Dec 27 16:31:49 homeassistant kernel: x5 : 0000000000000000 x4 : 0000000000000000
Dec 27 16:31:49 homeassistant kernel: x3 : ffff8000113de368 x2 : 0000000000000006
Dec 27 16:31:49 homeassistant kernel: x1 : cc8a0c72f46eb000 x0 : 0000000000000000
Dec 27 16:31:49 homeassistant kernel: Call trace:
Dec 27 16:31:49 homeassistant kernel:  dev_watchdog+0x300/0x310
Dec 27 16:31:49 homeassistant kernel:  call_timer_fn.isra.0+0x24/0x80
Dec 27 16:31:49 homeassistant kernel:  run_timer_softirq+0x3f0/0x420
Dec 27 16:31:49 homeassistant kernel:  efi_header_end+0x128/0x264
Dec 27 16:31:49 homeassistant kernel:  irq_exit+0xc0/0xe0
Dec 27 16:31:49 homeassistant kernel:  __handle_domain_irq+0x68/0xc0
Dec 27 16:31:49 homeassistant kernel:  gic_handle_irq+0x54/0xb0
Dec 27 16:31:49 homeassistant kernel:  el1_irq+0xb8/0x180
Dec 27 16:31:49 homeassistant kernel:  arch_cpu_idle+0x14/0x20
Dec 27 16:31:49 homeassistant kernel:  do_idle+0x208/0x260
Dec 27 16:31:49 homeassistant kernel:  cpu_startup_entry+0x28/0x70
Dec 27 16:31:49 homeassistant kernel:  secondary_start_kernel+0x154/0x190
Dec 27 16:31:49 homeassistant kernel: ---[ end trace 56ddfaf1cc78e94f ]---
Dec 27 16:31:49 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: Reset adapter.
Dec 27 16:31:49 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: PHY [0.0:00] driver [RTL8211F Gigabit Ethernet] (irq=31)
Dec 27 16:31:49 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: No Safety Features support found
Dec 27 16:31:49 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: PTP not supported by HW
Dec 27 16:31:49 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: configuring for phy/rgmii link mode
Dec 27 16:31:52 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Dec 27 16:31:52 homeassistant kernel: IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Dec 27 16:31:58 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: Reset adapter.
Dec 27 16:31:58 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: PHY [0.0:00] driver [RTL8211F Gigabit Ethernet] (irq=31)
Dec 27 16:31:58 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: No Safety Features support found
Dec 27 16:31:58 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: PTP not supported by HW
Dec 27 16:31:58 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: configuring for phy/rgmii link mode
Dec 27 16:32:01 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Dec 27 16:32:01 homeassistant kernel: IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Dec 27 16:32:07 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: Reset adapter.
Dec 27 16:32:07 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: PHY [0.0:00] driver [RTL8211F Gigabit Ethernet] (irq=31)
Dec 27 16:32:07 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: No Safety Features support found
Dec 27 16:32:07 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: PTP not supported by HW
Dec 27 16:32:07 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: configuring for phy/rgmii link mode
Dec 27 16:32:10 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[... cut ...]
Dec 27 16:32:25 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: PHY [0.0:00] driver [RTL8211F Gigabit Ethernet] (irq=31)
Dec 27 16:32:25 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: No Safety Features support found
Dec 27 16:32:25 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: PTP not supported by HW
Dec 27 16:32:25 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: configuring for phy/rgmii link mode
Dec 27 16:32:28 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx

frog32, 2020-12-28:
After downgrading to 5.8 I still had a crash last night and then downgraded to 4.20 which is running since ~6h which did not happen the last few days. Will report back later today and it that did the trick will file a clean bug without all the noise.

agners, 2020-12-28:
@frog32 thanks for the report. The Blue should get shipped with RTC battery. Haven't seen this RTC write errors, do they still appear?

Since there is the "self-detected stall" message inbetween all of the RTC errors, it could be that Ethernet issue triggered the RTC communication problem (-110 means timeout, which would make sense if the CPU is stalled for some reason).

The question then is of course why Ethernet makes problem with the 5.x release at your end...

frog32, 2020-12-28:
RTC errors are no longer appearing. It likely caused by the loss of connectivity or the eth0 crash.

5.9 and also 5.8 consistently crashed and since downgrading to 4.2 it runs. Here is the output of journalctl --list-boots so you can have an impression of the frequency of those crashes. Every restart was either changing ha version (upgrade to 5.9, downgrade to .58 and downgrade to 4.2) or me pulling power because the link went down again. So far 4.2 still runs stable. Do you know what version blue was shipped with? I had no crash the first few days.

```-10 23c9654093f94da5a53fb156c599090f Fri 2020-12-25 03:07:11 UTC—Fri 2020-12-25 11:26:49 UTC
 -9 711ad6c27a5f4fa8abe515d29438a648 Fri 2020-12-25 13:24:13 UTC—Fri 2020-12-25 23:35:51 UTC
 -8 c1ec3a1a1167409aae8196e43b5a7061 Sat 2020-12-26 07:11:16 UTC—Sat 2020-12-26 07:58:01 UTC
 -7 1e77fa077446442abd9b2a3bce8cb575 Sun 2020-12-27 02:19:55 UTC—Sat 2020-12-26 23:43:47 UTC
 -6 4fa9fa0c1c294ee68f19aba07e428e93 Sun 2020-12-27 08:35:36 UTC—Sun 2020-12-27 07:54:56 UTC
 -5 05b29f7adc2846bd8c1b6225b3ad6879 Sun 2020-12-27 11:16:14 UTC—Sun 2020-12-27 11:32:01 UTC
 -4 47a6c76c259946f5a2123fa226aff0bf Sun 2020-12-27 11:55:55 UTC—Sun 2020-12-27 16:37:08 UTC
 -3 42078390ab6c4d3d99abbd069e64ea35 Sun 2020-12-27 17:00:42 UTC—Sun 2020-12-27 17:04:12 UTC
 -2 c99bd1d039b143d0b24cc5d801e79432 Sun 2020-12-27 17:09:16 UTC—Sun 2020-12-27 20:05:13 UTC
 -1 30f7fe84a978478bacb3133f62005424 Sun 2020-12-27 20:06:53 UTC—Sun 2020-12-27 22:42:18 UTC
  0 45fd32249cdd447da7eebc2ac29bad62 Mon 2020-12-28 01:02:01 UTC—Mon 2020-12-28 03:27:59 UTC

agners, 2020-12-28:
Home Assistant Blue is getting shipped with 4.20.

For Release 5 we moved from Linux 5.7 to Linux 5.9, so there is quite a difference between the versions from a kernel perspective. Most patches we back-ported to 4.20, the only patch which is only on release 5 is #1038. It would be cool if you can test Home Assistant OS 5.6 (which did not had #1038 merged yet), just to see if it makes a difference.

frog32, 2020-12-28:
Happy to test this and give it one night (I need this working between tomorrow evening and January 5th). If I understood correctly this patch is just a timing difference. I just restarted but I also read out some networking related counters and they showed the follwoing (while running for more than 12h):

```
~ $ cat /sys/devices/platform/soc/ff3f0000.ethernet/net/eth0/carrier_changes
2
~ $ cat /sys/devices/platform/soc/ff3f0000.ethernet/net/eth0/statistics/rx_errors
0
~ $ cat /sys/devices/platform/soc/ff3f0000.ethernet/net/eth0/statistics/tx_errors
0
```

Is there a specific thing that you expect me to look out for? I have a hunch that my switch that is attached to home assistant might be causing issues since my other HA setup was unable to ping the primary one for several hours but both were apparently able to connect to the internet. I cobbled together a container running smokeping on my secondary system to get more data on local connectivity.

agners, 2020-12-28:
Nothing special to look out for, just whether it behaves more like 5.8/5.9 or 4.20.

frog32, 2020-12-29:
Running smooth since my last post on 5.6. Happy to give it an afternoon on 5.7 if you think there is value in that.

agners, 2020-12-29:
Yeah since it starts to look like its related to #1038 I expect it to show the problem, but would be good to confirm.

agners, 2020-12-29:
With 5.6, do you get any logs under Host logs in Supervisor frontend related to `ff3f0000.ethernet`?

frog32, 2020-12-29:
I actually upgraded to 5.7 before you asked me about the Supervisor logs. Are those logs available from the journal?

agners, 2020-12-29:
The "Host" log in supervisor is the kernel log (dmesg). You get the kernel log in journalctl as well, to get the one from the last boot use
```
journalctl -t kernel -b -1
```

frog32, 2020-12-29:
Sorry did not realize that there is now a selector box that lets you peek at kernel and other logs. Nice feature!
Nothing in kernel log after boot:

```
# journalctl -k -b -1 | grep ff3f
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet: IRQ eth_wake_irq not found
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet: IRQ eth_lpi not found
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet: PTP uses main clock
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet: User ID: 0x11, Synopsys ID: 0x37
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet:         DWMAC1000
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet: DMA HW capability register supported
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet: RX Checksum Offload Engine supported
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet: COE Type 2
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet: TX Checksum insertion supported
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet: Wake-Up On Lan supported
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet: Normal descriptors
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet: Ring mode enabled
Dec 28 19:22:39 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet: Enable RX Mitigation via HW Watchdog Timer
Dec 28 19:22:42 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: PHY [0.0:00] driver [RTL8211F Gigabit Ethernet] (irq=31)
Dec 28 19:22:42 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: No Safety Features support found
Dec 28 19:22:42 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: PTP not supported by HW
Dec 28 19:22:42 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: configuring for phy/rgmii link mode
Dec 28 19:22:45 homeassistant kernel: meson8b-dwmac ff3f0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
```

Exactly the same with the current session and running stable on 5.7 for 8h 55min. I will give it some more time before I downgrade to 4.2 until next Monday evening. I will not have access to the box for a few days and could not power cycle it if it crashes.

frog32, 2020-12-30:
Ran successfully on 5.7 for ~20h now back to 4.2 for the rest of the week. Happy to test 5.8 and 5.9 again next week since we could not yet pinpoint what the issue is.

frog32, 2021-01-02:
My downgrade failed due to some partition being busy and I will have another look at it on Monday. However I am running on 5.7 with an uptime of 3.5 days. Either the issue is not happening under 5.7 or it was triggered by something that solved itself.
So far I got those errors only on 5.8 and 5.9
4.2, 5.6 and 5.7 did not show those.

Next I will upgrade to 5.9 again (unless 5.10 makes more sense) to check if it is still happening.

frog32, 2021-01-05:
Upgraded to 5.10 yesterday and got an other crash this night. Before that I was running without any issues on 5.7 for 4 days.

```
Jan 05 02:12:50 homeassistant kernel: NETDEV WATCHDOG: eth0 (meson8b-dwmac): transmit queue 0 timed out
Jan 05 02:12:50 homeassistant kernel: WARNING: CPU: 4 PID: 3778 at net/sched/sch_generic.c:442 dev_watchdog+0x300/0x310
Jan 05 02:12:50 homeassistant kernel: Modules linked in: xfrm_user xfrm_algo cfg80211 dw_hdmi_i2s_audio dw_hdmi_cec btusb btrtl btintel btbcm bluetooth ecdh_generic ecc rfkill realtek dwmac_generic pl2303 meson_dw_hdmi dw_hdmi cec meson_drm drm_kms_helper dwmac_meson8b panfrost stmmac_platform gpu_sched crct10dif_ce stmmac mdio_xpcs rc_odroid meson_ir rc_core meson_canvas rtc_meson_vrtc snd_soc_simple_amplifier cdc_acm fuse drm
Jan 05 02:12:50 homeassistant kernel: CPU: 4 PID: 3778 Comm: exe Not tainted 5.9.16 #1
Jan 05 02:12:50 homeassistant kernel: Hardware name: Hardkernel ODROID-N2Plus (DT)
Jan 05 02:12:50 homeassistant kernel: pstate: 80000005 (Nzcv daif -PAN -UAO BTYPE=--)
Jan 05 02:12:50 homeassistant kernel: pc : dev_watchdog+0x300/0x310
Jan 05 02:12:50 homeassistant kernel: lr : dev_watchdog+0x300/0x310
Jan 05 02:12:50 homeassistant kernel: sp : ffff80001157bd90
Jan 05 02:12:50 homeassistant kernel: x29: ffff80001157bd90 x28: ffff0000db8fc940
Jan 05 02:12:50 homeassistant kernel: x27: 0000000000000004 x26: 0000000000000140
Jan 05 02:12:50 homeassistant kernel: x25: 00000000ffffffff x24: 0000000000000004
Jan 05 02:12:50 homeassistant kernel: x23: ffff0000daf6039c x22: ffff0000daf60000
Jan 05 02:12:50 homeassistant kernel: x21: ffff0000daf60438 x20: ffff8000113d7000
Jan 05 02:12:50 homeassistant kernel: x19: 0000000000000000 x18: 0000000000000000
Jan 05 02:12:50 homeassistant kernel: x17: 0000000000000000 x16: 0000000000000000
Jan 05 02:12:50 homeassistant kernel: x15: ffff8000113f5000 x14: ffff8000114d816a
Jan 05 02:12:50 homeassistant kernel: x13: 0000000000000000 x12: ffff8000114d7000
Jan 05 02:12:50 homeassistant kernel: x11: ffff8000113f5000 x10: ffff8000114d77b0
Jan 05 02:12:50 homeassistant kernel: x9 : 0000000000000000 x8 : 0000000000000000
Jan 05 02:12:50 homeassistant kernel: x7 : 0000000000000004 x6 : 0000000000000217
Jan 05 02:12:50 homeassistant kernel: x5 : ffff0000f462b1f0 x4 : 0000000000000000
Jan 05 02:12:50 homeassistant kernel: x3 : 0000000000000027 x2 : 0000000000000023
Jan 05 02:12:50 homeassistant kernel: x1 : cb6ee8e20cac8100 x0 : 0000000000000000
Jan 05 02:12:50 homeassistant kernel: Call trace:
Jan 05 02:12:50 homeassistant kernel:  dev_watchdog+0x300/0x310
Jan 05 02:12:50 homeassistant kernel:  call_timer_fn.isra.0+0x24/0x80
Jan 05 02:12:50 homeassistant kernel:  run_timer_softirq+0x3f0/0x420
Jan 05 02:12:50 homeassistant kernel:  efi_header_end+0x128/0x264
Jan 05 02:12:50 homeassistant kernel:  irq_exit+0xc0/0xe0
Jan 05 02:12:50 homeassistant kernel:  __handle_domain_irq+0x68/0xc0
Jan 05 02:12:50 homeassistant kernel:  gic_handle_irq+0x54/0xb0
Jan 05 02:12:50 homeassistant kernel:  el1_irq+0xb8/0x180
Jan 05 02:12:50 homeassistant kernel:  _raw_spin_unlock_irq+0x10/0x50
Jan 05 02:12:50 homeassistant kernel:  schedule_tail+0x10/0x110
Jan 05 02:12:50 homeassistant kernel:  ret_from_fork+0x4/0x34
Jan 05 02:12:50 homeassistant kernel: ---[ end trace 8149466a34992085 ]---
```

frog32, 2021-01-06:
OK this gets weirder. I realized that id did often crash shortly after I was gone to sleep. I am not sure what exactly triggers it (might be my laptop or phone) just now it crashed on 5.7. While crashing my other setup that runs smokeping is also loosing full connectivity but when I connected to it from my laptop it worked and smokeping started to be able to reach targets again. I am now downgrading to 5.6 and will wait for tomorrow to see if it is happening again. If yes I will likely have to sniff traffic to figure out what exactly is triggering it.

frog32, 2021-01-06:
Got a crash with 5.6 as well shortly after going back to sleep. Current hypothesis is that it is only triggered when my macbook is docked and going to sleep. So far this would explain every crash I had since roughly December 26th. I will get a better timeline of when my macbook was sleeping to correlate further. After I undocked it last night it did not immediately crash homeassistant when it went to sleep.

When ever it crashed my old RPI4 running hassio with a smokeping addon stopped all its pings (some internet, some local wired, some local WLAN). Firing up a packet capture showed an unusual amount of arp requests and accessing smokeping using a browser somehow recovered it entirely since it never downed the its ethernet port.

Next steps:
- Check if my RPI4 setup shows anything suspicious in its kernel log since it did not require a restart it might have better logs.
- Verify again by deliberately crashing 5.6 by docking my macbook and waiting for ~15 minutes for it to sleep.
- Do the same with 4.2 (it is possible that I never had my macbook sleeping while 4.2 was running since outside holiday period my work laptop is docked.
- Try the same with the Ethernet cable unplugged on my monitor.
- Try to packet capture while the this happens.

frog32, 2021-01-06:
OK, I can now confirm that this is caused by my laptop going to sleep while plugged into the monitor since I was able to provoke that by setting the sleep timer to 5 minutes and waiting while running ping. I am currently trying to evaluate the tcpdump that I took capturing all broadcast traffic. There is a lot of ARP traffic happening shortly after it goes to sleep but it seems that it is not coming from my macbook.

frog32, 2021-01-07:
Quick update: There seem to be more problems at hand here. I can confirm that there is a lot of traffic piling up on my network which seems to trigger spanning tree reconvergence (sonos speakers and my google wifi are able to speak spanning tree).

Is it possible that the newer linux kernel that is used on the n2+ behaves differently and that this is triggering the stacktrace I posted earlier?

frog32, 2021-01-07:
Further investigation shows that shortly before having the problem there is a huge flood of MAC CTRL Pause frames. While this explains why it is loosing all network connection and I have to find the source of these frames it does not explain why the kernel log contains a stacktrace.

This flood is starting 38s before the tcpdump ends (taken from the odroid n2+) and is exactly 60 pps. Here is one of those frames for reference and they all are identical.

Frame 17371: 60 bytes on wire (480 bits), 60 bytes captured (480 bits)
Ethernet II, Src: 00:00:00_00:00:00 (00:00:00:00:00:00), Dst: Spanning-tree-(for-bridges)_01 (01:80:c2:00:00:01)
MAC Control
    Opcode: Pause (0x0001)
    pause_time: 65535

frog32, 2021-01-09:
OK now after some more testing I can confirm that all versions of the home assistant operating system are affected, I managed to crash 4.2 as well.

Next:
- Test if just sending a flood of MAC Control Pause frames is enough or do I need to search further.
- Figure out if those pause frames should be broadcasted or if this should be "absorbed" by the switch.

agners, 2021-01-11:
Interesting investigation. That this is triggered by a somewhat special condition in you network does make sense as it seems not to be a general problem. That said, it still should not crash the kernel, obviously.

> Test if just sending a flood of MAC Control Pause frames is enough or do I need to search further.
If that triggers the problem, I can try reproducing here.

Quite possible that you are tracing an actual kernel bug here. Maybe @xdarklight happens to have an idea?

frog32, 2021-01-11:
Today I reproduced this with scapy. A single frame does get absorbed by the switch but if I send them in a loop HA quickly (unresponsive immediately, turning off its link shortly afterwards) dies while my devsetup running a RPI4 survives it but is almost unresponsive during the attack.
```
>>> from scapy.contrib.mac_control import MACControlPause
>>> packet = Ether(src="00:00:00:00:00:00", dst="01:80:c2:00:00:01") / MACControlPause(pause_time=65535)
>>> hexdump(packet)
0000  01 80 C2 00 00 01 00 00 00 00 00 00 88 08 00 01  ................
0010  FF FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0020  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
0030  00 00 00 00 00 00 00 00 00 00 00 00              ............
>>> sendp(packet, iface="eth0", loop=1)
```

agners, 2021-01-11:
You are missing two imports in your reproducer:
```
from scapy.contrib.mac_control import Ether
from scapy.all import sendp
```

I was able to reproduce it! :+1:  But I had to let it run like 5 minutes or so. Also I connected directly, I think it did not happen when I had my switch in between (not sure about that, maybe I did not let it run long enough).

```
[ 9974.301905] meson8b-dwmac ff3f0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[10120.488799] ------------[ cut here ]------------
[10120.488818] NETDEV WATCHDOG: eth0 (meson8b-dwmac): transmit queue 0 timed out
[10120.488875] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:442 dev_watchdog+0x300/0x310
[10120.488879] Modules linked in: xfrm_user xfrm_algo cfg80211 rfkill dw_hdmi_cec dw_hdmi_i2s_audio realtek dwmac_generic dwmac_meson8b stmmac_platform stmmac mdio_xpcs rc_odroid meson_dw_hdmi meson_ir rc_core meson_drm rtc_meson_vrtc dw_hdmi cec meson_canvas drm_kms_helper panfrost gpu_sched crct10dif_ce ch341 snd_soc_simple_amplifier drm
[10120.488917] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 5.9.15 #1
[10120.488919] Hardware name: Hardkernel ODROID-N2Plus (DT)
[10120.488923] pstate: 20000005 (nzCv daif -PAN -UAO BTYPE=--)
[10120.488926] pc : dev_watchdog+0x300/0x310
[10120.488928] lr : dev_watchdog+0x300/0x310
[10120.488929] sp : ffff800011563d90
[10120.488931] x29: ffff800011563d90 x28: ffff0000f22a8940
[10120.488934] x27: 0000000000000004 x26: 0000000000000140
[10120.488937] x25: 00000000ffffffff x24: 0000000000000001
[10120.488940] x23: ffff0000da91039c x22: ffff0000da910000
[10120.488943] x21: ffff0000da910438 x20: ffff8000113d7000
[10120.488946] x19: 0000000000000000 x18: 0000000000000000
[10120.488949] x17: 0000000000000000 x16: 0000000000000000
[10120.488952] x15: ffff8000113f5000 x14: ffff8000114d816a
[10120.488955] x13: 0000000000000000 x12: ffff8000114d7000
[10120.488958] x11: ffff8000113f5000 x10: ffff8000114d77b0
[10120.488961] x9 : 0000000000000000 x8 : 0000000000000000
[10120.488964] x7 : 0000000000000004 x6 : 0000000000000003
[10120.488966] x5 : 0000000000000000 x4 : 0000000000000000
[10120.488970] x3 : ffff8000113de368 x2 : 0000000000000006
[10120.488972] x1 : 5036ffcbca2a8900 x0 : 0000000000000000
[10120.488976] Call trace:
[10120.488980]  dev_watchdog+0x300/0x310
[10120.488986]  call_timer_fn.isra.0+0x24/0x80
[10120.488989]  run_timer_softirq+0x3f0/0x420
[10120.488993]  efi_header_end+0x128/0x264
[10120.488997]  irq_exit+0xc0/0xe0
[10120.489002]  __handle_domain_irq+0x68/0xc0
[10120.489006]  gic_handle_irq+0x54/0xb0
[10120.489009]  el1_irq+0xb8/0x180
[10120.489013]  arch_cpu_idle+0x14/0x20
[10120.489018]  do_idle+0x208/0x260
[10120.489022]  cpu_startup_entry+0x24/0x70
[10120.489026]  secondary_start_kernel+0x154/0x190
[10120.489029] ---[ end trace aed12587aff7fed9 ]---
[10120.489070] meson8b-dwmac ff3f0000.ethernet eth0: Reset adapter.
[10120.598467] meson8b-dwmac ff3f0000.ethernet eth0: PHY [0.0:00] driver [RTL8211F Gigabit Ethernet] (irq=31)
[10120.660696] meson8b-dwmac ff3f0000.ethernet eth0: No Safety Features support found
[10120.660708] meson8b-dwmac ff3f0000.ethernet eth0: PTP not supported by HW
[10120.660718] meson8b-dwmac ff3f0000.ethernet eth0: configuring for phy/rgmii link mode
[10123.915618] meson8b-dwmac ff3f0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[10123.915643] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[10129.448440] meson8b-dwmac ff3f0000.ethernet eth0: Reset adapter.
[10129.558080] meson8b-dwmac ff3f0000.ethernet eth0: PHY [0.0:00] driver [RTL8211F Gigabit Ethernet] (irq=31)
[10129.624409] meson8b-dwmac ff3f0000.ethernet eth0: No Safety Features support found
[10129.624418] meson8b-dwmac ff3f0000.ethernet eth0: PTP not supported by HW
[10129.624424] meson8b-dwmac ff3f0000.ethernet eth0: configuring for phy/rgmii link mode
[10132.914497] meson8b-dwmac ff3f0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[10132.914526] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

Is your real setup also flooding packages at that rate?

I guess even its a bit artificial test case right now, it should not lead to an adapter reset. Probably the best way forward is to post the issue/stack trace along with the reproducer to the Linux amlogic mailing list, see http://linux-meson.com/doku.php.

frog32, 2021-01-12:
I upgraded to a "smart managed" switch which prevented this from happening since the packets are not forwarded and absorbed by the switch (at least for a few hours, had an other crash early in the morning so it is possible that there is some issue left there).

The imports are not needed if you start with scapy instead of a python shell.

frog32, 2021-01-12:
According to earlier packet captures my screen seem to send packets at a rate of 60/s, individual packets were absorbed by my old switch as well but there seem to be something that makes switches just give up and forward it after a given rate or total number of packets is exceeded.

natedzl, 2021-01-14:
I just received my HA Blue yesterday and shortly after booting it up and restoring my snapshot, the network stopped responding on it.  A quick disconnect/reconnect of the cable resolved the issue, but I noticed it happened again about 2 hours later and then again in the middle of the night. After the first issue, I saw the HA OS was running 4.2 and Supervisor was prompting to update to 5.10 which I did before the second and third crashes.  Unlike @frog32, I've not been able to find the exact sequence of events happening in my environment to replicate the issue just yet but will see if I can figure that out.  

If there are logs that would be pertinent to help locate the issue, please let me know which those are and I can provide.

xdarklight, 2021-01-16:
Hi all,

> [10120.488818] NETDEV WATCHDOG: eth0 (meson8b-dwmac): transmit queue 0 timed out
> [10120.488875] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:442 dev_watchdog+0x300/0x310

for this problem I suggest reporting this problem (and asking for advice on how to help debug it) to the netdev mailing list and Cc'ing the maintainers:
```
$ ./scripts/get_maintainer.pl -f drivers/net/ethernet/stmicro/stmmac/stmmac_main.c | grep -v STM32 | grep -v RESET | grep -v SFP
Giuseppe Cavallaro <peppe.cavallaro@st.com> (supporter:STMMAC ETHERNET DRIVER)
Alexandre Torgue <alexandre.torgue@st.com> (supporter:STMMAC ETHERNET DRIVER)
Jose Abreu <joabreu@synopsys.com> (supporter:STMMAC ETHERNET DRIVER)
"David S. Miller" <davem@davemloft.net> (maintainer:NETWORKING DRIVERS)
Jakub Kicinski <kuba@kernel.org> (maintainer:NETWORKING DRIVERS)
netdev@vger.kernel.org (open list:STMMAC ETHERNET DRIVER)
linux-kernel@vger.kernel.org (open list)
```

The `meson8b-dwmac` driver is just a small "glue" driver which configures some Amlogic specific registers for the PHY connection and then loads the stmmac driver (which is the actual Ethernet controller hardware).
It's listed in the kernel log as being responsible for the Ethernet connection, but in reality it's only responsibly for a small part of that - the majority is handled by the stmmac driver.

Best regards,
Martin

agners, 2021-01-19:
> If there are logs that would be pertinent to help locate the issue, please let me know which those are and I can provide.

If it is still happening, check the Host log in Supervisor. If you see logs like in https://github.com/home-assistant/operating-system/issues/1130#issuecomment-751460290, then your problem might have the same cause.

agners, 2021-01-19:
> According to earlier packet captures my screen seem to send packets at a rate of 60/s, individual packets were absorbed by my old switch as well but there seem to be something that makes switches just give up and forward it after a given rate or total number of packets is exceeded.

Reading https://en.wikipedia.org/wiki/Ethernet_flow_control a bit, it seems like you have a node which is overwhelmed quite quickly. Maybe its just a problem of that node. I would try to identify that and see if its just that one node (you should be able to identify by the senders address).

Is the pause quota/time according you packet capture also `65535`? If the MAC continuously receives pause frames, then the behavior that the transmit queue gets overfilled make sense (`transmit queue 0 timed out`). On the other hand, at 60 pause frames/s, with a 500bit pause, that would add up to 33MBit, so there should still be bandwidth, in theory.

frog32, 2021-01-20:
So far I have worked around the problem by changing to managed switches. The source address that the monitor used was 00:00:00:00:00:00 which is weird but was reproducible. A basic SOHO switch is broadcasting traffic to this address to all its ports (tested with one dlink and one netgear switch) while my new netgear 350 series switches are receiving those pause frames and acting on them internally which lets the rest of my network continue as usual. While this workaround works I also reached out to DELL to report this issue so they can fix it but their L1 support blocked hard and told me that they only would help me if I pay 60$ upfront even though I was just reporting an already diagnosed issue. Since they are essentially just using off the shelf NICs in their monitors I would not be surprised if the same behavior is also seen in some USB-C docks or USB-C to Ethernet adapters if they are still powered when the attached laptop goes to sleep.

agners, 2021-01-20:
I've run the script again in a loop with about 120 frames/sec. With this amount of pause frames the Ethernet link should not starve. And indeed, I did not had a stack trace for 24h.

So I think in your case somehow so many of those pause frame add up that the Ethernet link cannot send anymore. This seems to lead to a link reset with that driver. If that is proper is probably up to debate, but having Ethernet pause frames at a too high rate is probably a network problem anyway...

schieb, 2021-04-15:
> I would not be surprised if the same behavior is also seen in some USB-C docks or USB-C to Ethernet adapters if they are still powered when the attached laptop goes to sleep

Turns out it doesn't need to be connected to the laptop at all...

I had this same issue: same symptoms, same trace from https://github.com/home-assistant/operating-system/issues/1130#issuecomment-751509696. The same underlying issue was present on my network - exactly 60 60-byte frames per second from 00:00:00:00:00:00 to 01:80:c2:00:00:01 with MAC Control (0x8808) payload: `Opcode: Pause (1)`, `pause_time: 0xffff`.

In my case, the frames were coming from a malfunctioning USB-C multiport adapter. The adapter was not plugged in to a computer but was plugged into a power source and connected by Ethernet to the same unmanaged switch as my Odroid N2+ (not Blue). I have two of these USB-C adapters. One reliably sends these frames as described above and the other does not. This issue was causing other devices connected to the same unmanged switch, including a Windows 10 box, to lose Ethernet connection as well.

I can provide more details if it is helpful, though I do agree that this is a local network issue and simply isolating the device and removing it from the network should be sufficient.

------------------

My napkin math might be off but on a GB line you have a bandwidth of approximately 2^30 bps. With 60 (~2^6) frames per second, each specifying a pause of  2^16 quanta (where each quanta measured as 2^9 units of "bit time"), you get a total bit time of `2^-30 * 2^6 * 2^9 * 2^16 = 2`. Which means the link is effectively paused indefinitely.

closed 2021-10-13 by github-actions[bot]