home-assistant/operating-system #1131: hassos_ova-5.9.vhdx corrupted

freakbn777, 2020-12-27:
**Hardware Environment**
- [X ] Hyper-V (on Windows 10 pro)

**Home Assistant OS release:**
- [X ] Fresh installation of release 5.9
- Additional information (if accessible): Windews 10 OS is in French

**Supervisor logs:**
N/A

**Journal logs:**
N/A

**Kernel logs:**
N/A

**Description of problem:**
Hi, I try to use VHDX image from this page: https://www.home-assistant.io/hassio/installation

[image]
The error say cannot obtain disk information. The file or folder is damaged and unreadable.

I downloaded it 3 times but I always get this error.

smoke007, 2020-12-27:
My Hyper-V VM is no longer booting after updating to OS 5.9.  Try using an older version like https://github.com/home-assistant/operating-system/releases/download/4.20/hassos_ova-4.20.vhdx.gz.  

Here are my personal directions that I've shared with friends to get started with HA on a Hyper-V VM.

1. Download the VHDX from https://www.home-assistant.io/hassio/installation/ and extract it to wherever you want to keep the VM’s disk. Rename as required. You might want to pre-create the folder tree for the VM.
2. Create a new virtual machine. Choose to put it in the path you’ve already set up.
3. Name it, select Generation 2, and give it some memory.  RP3 has 1024 which is adequate, but I went with 2048 for all my addons.
4. Choose to use an existing disk - the one you downloaded in step 1.
5. Before booting the VM, turn off Secure Boot in its settings and expand the disk to 32GB. *Warning, expanding the disk too large will cause it to fail.
6. Power on the VM.
7. Find your Home Assistant VM's IP on your network via DHCP leases and login with http://IP:8123

freakbn777, 2020-12-27:
Thanks for your reply but unfortunately it's also not working
[image]
(I switched my OS to english, better than french screenshot)

freakbn777, 2020-12-27:
Oh wait.... I understand ! I thinked extension .xz was just added to be sure vhdx download is not blocked. I didn't know it is an extension for compressed file. It work better like this :)

closed 2020-12-27 by freakbn777

smoke007, 2020-12-27:
Yes, glad you figured out file was compressed!  :)  I was going to ask that next.  Thanks for the update that it's working now.