home-assistant/operating-system #1134: Raspberry pi 400 + SSD - HA 5.9 won't boot

ondrackajan, 2020-12-28:
**Hardware Environment**

- [x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [x] Fresh installation of release x.y
- [x] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

alexbakrasp, 2020-12-28:
facing the same problem here!

agners, 2020-12-28:
Hm, that is interesting. With 5.9 (see #1094) now actually deploy the correct device tree for this device, which if anything should make the Pi 400 properly work :-)

Do you see anything on the screen when the system is stuck? If so, can you make a screenshot?

Does SD card boot gets stuck too?

alexbakrasp, 2020-12-28:
Its stuck in a boot loop (the power led Pi400 remains flashes in a sec). I dit not tried to boot it with an sd card.

alexbakrasp, 2020-12-28:
Everithing works fine on HA 5.8...   see you all next year! ;-)

ondrackajan, 2020-12-29:
> 
> 
> Hm, that is interesting. With 5.9 (see #1094) now actually deploy the correct device tree for this device, which if anything should make the Pi 400 properly work :-)
> 
> Do you see anything on the screen when the system is stuck? If so, can you make a screenshot?
> 
> Does SD card boot gets stuck too?

I'm unable to see anything on the screen as it keeps reconnecting the video source.

SD card boot is stuck too.

BarBaar44, 2020-12-29:
Looks a lot like the issue I have/had with an Odroid C2.

The device is not booting with a fresh install. Also first installing HA 4.2 and then upgrading to HA 5.9 resulted in a unbootable situation..

Screen remains black, and not heartbeat (blue led) on the Odroid either...

agners, 2020-12-29:
@BarBaar44 this is a different issue. Let's track it in #1142

agners, 2020-12-31:
Fix will be in 5.10 onwards.

closed 2020-12-31 by agners