home-assistant/operating-system #1141: hassio-data partition does not exist after datactl move

ssmorris, 2020-12-29:
**Hardware Environment**

- [X] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [X] Fresh installation of release 5.9
- [ ] Updated from version x.y
- Additional information (if accessible):
## System Health

version | 2020.12.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.6
os_name | Linux
os_version | 5.4.79-v8
arch | aarch64
timezone | America/New_York

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.9
-- | --
update_channel | stable
supervisor_version | 2020.12.7
docker_version | 19.03.13
disk_total | 57.8 GB
disk_used | 6.3 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.3.0), AdGuard Home (2.6.0), Duck DNS (1.12.4), File editor (5.2.0), Network UPS Tools (0.3.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
mode | storage
views | 3
resources | 7

</details>

**Description of problem:**
Attempting to migrate data partition from SD card to USB SSD.  SSD is recognized as /dev/sda.  Using the following steps leaves my installation in an unbootable state.  After reimaging the SD card with a fresh install, repeating the steps arrives at the same outcome.  I've attached screenshots of the messages I'm receiving, and am happy to provide any additional details that might be useful.

Followed the steps @ https://github.com/home-assistant/operating-system/blob/dev/Documentation/partition.md
1. With HDMI + keyboard attached to raspberry pi, log into hassio console via root user
2. Type "login" into HA CLI
3. Used "fdisk -l" to confirm USB SSD is at /dev/sda (screenshot 1 below)
4. Ran "datactl move /dev/sda"
5. Typed "yes", everything appears successful
6. Rebooted
7. On next boot, received "[ OK ] Finished Home Assistant OS data partition migration." after approximately 2 minutes
8. Immediately after, received "[FAILED] Failed to start HassOS data resizing" and "[FAILED] Failed to mount HassOS data partition." (screenshot 2 below)
9. Output of fdisk -l is identical after reboot, with USB SSD still listed as /dev/sda.  The first time I ran through these steps it did wipe the existing windows partitions off of the USB SSD and replace them with the linux partition.

Screenshot 1: Output of "fdisk -l"
[image]

Screenshot 2: Output after reboot
[image]

Screenshot 3: Output of systemctl status hassos-expand.service and mnt-data.mount
[image]

Any thoughts or help is greatly appreciated.

agners, 2020-12-29:
Thanks for the detailed report! Very much appreciated!

The `hassos-expand` service should actually not trigger in this case as the external disk already has an expanded partition layout. If the free space at the end of the last partition [is less than 16384](https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/rootfs-overlay/usr/libexec/hassos-expand#L35), hassos-expand should really just exit. From what I can tell from the first screenshot, the partition end of the first partition is 31238138, which is smaller than disk size - 16384 (31238144 - 16384).
 
Is the first screenshot before resizing or after?

Can you post the output of the following command after running using `datactl move`?
```
sfdisk -lqJ /dev/sda
```

ssmorris, 2020-12-29:
Thanks for the quick reply @agners!  The first screenshot was taken after the "failed" resize.

Here's the output of `sfdisk -lqJ /dev/sda`:
[image]

agners, 2020-12-29:
Ok, yeah that is what we end up with. I would be interested in how it looks before rebooting so:

```
datactl move /dev/sda
...
# before reboot
sfdisk -lqJ /dev/sda

```

ssmorris, 2020-12-29:
Got it, here's the output of `sfdisk -lqJ /dev/sda` before reboot, with bonus output of `datactl move /dev/sda` as well since it was still on screen.  Note that I reformatted the USB SSD to NTFS before going through the steps again, just to make sure we had a clean starting point.

[image]

ssmorris, 2020-12-29:
Might be going off on a tangent here, but [mnt-data.mount line 12](https://github.com/home-assistant/operating-system/blob/7959113c977fc881bcce65a6550d8d02a4137830/buildroot-external/rootfs-overlay/usr/lib/systemd/system/mnt-data.mount#L12) is looking for `/dev/disk/by-label/hassos-data`, how does the label on the USB SSD being `hassos-data-external` match up to that?

agners, 2020-12-29:
(File system)label and partition label are two different things. We use the file system label `hassos-data` to mark the active data partition they system should use. The partition label is `hassos-data` for the internal flash drive and `hassos-data-external` for the external disk.

agners, 2020-12-29:
I wonder if the data move really succeeded. Can you get it to do the move again and then check logs of `hassos-data.service`?

ssmorris, 2020-12-29:
Here's the output of `systemctl status hassos-expand.service`:

[image]

For what it's worth, `e2label /dev/sda1` does return `hassos-data`.

agners, 2020-12-29:
And `findfs LABEL="hassos-data"` should return `/dev/sda1`, does it?

I cannot explain what is going on here. The expand service should really not get triggered...

I guess it is some kind of race, such that hassos-expand.service picks the wrong device. But even if so, it should either pick the old (/dev/mmcblk0p8) or the new (/dev/sda1), and then derive all variables from whatever it picks (see https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/rootfs-overlay/usr/libexec/hassos-expand#L7).

What is a bit interesting is that it claims the secondary partition table overlaps the partition which is weird.

agners, 2020-12-29:
Hm, what is the output of `cat /sys/class/block/sda/size`?

ssmorris, 2020-12-30:
Here's the output of each of those commands.  Interestingly, each time I boot I'm getting OK on mnt-data.mount now, but the data drive is instantly unmounting right after it mounts. Still getting the same failure message as before on the hassos-expand step.  Latest output of `systemctl status mnt-data.mount` is included in this screenshot. 

[image]

Either the number returned by `cat /sys/class/block/sda/size` or the calculation [on line 34](https://github.com/home-assistant/operating-system/blob/142e7df6cfe3d83e9b19f2b8d100378e9d28ce84/buildroot-external/rootfs-overlay/usr/libexec/hassos-expand#L34) seems suspect unless I'm missing something here (which is very possible).

`LAST_USABLE_LBA = 249,905,118`
`DATA_PARTITION_END = 31,238,139`

So... `UNUSED_BLOCKS = 218,666,979` when in reality the partition already takes up 118.9gb of a 120gb SSD

That said, I think the most interesting part is that it successfully mounted but then immediately unmounted again, any idea how we can find out what triggered that?  I mounted the partition on another machine and confirmed that the files match between `hassos-data-old` on the SD and `hassos-data` on the SSD, so the copy seems to have worked perfect.

agners, 2020-12-30:
@ssmorris thanks for the output. After looking at the numbers yesterday I started to understand the problem: Your drive uses 4k block size, which is still rather uncommon today (also called [Advanced Format](https://en.wikipedia.org/wiki/Advanced_Format)). And as you noted, the calculations are off in this case. `/sys/class/block/sda/size` returns the size in 512 bytes. So it seems the hassos-expand script really isn't ready for 4k drives.

I am working on fixing the script. I should have a test build sometime today, would be cool if you can give this a try.

ssmorris, 2020-12-30:
Would be happy to give it a test once it's ready, thanks again for the help @agners!

agners, 2020-12-30:
@ssmorris development builds with the fix above are ready to download from https://os-builds.home-assistant.io/6.0.dev20201230/

ssmorris, 2020-12-31:
Success! Flashed with the dev build, ran `datactl move /dev/sda`, rebooted, and it just worked.  Couple command outputs below as confirmation that it's now using the USB SSD.  Also successfully copied and restored a snapshot, working perfect so far.

[image]

agners, 2020-12-31:
@ssmorris thanks for testing! Fix will be in 5.10 onwards.

closed 2020-12-31 by agners