home-assistant/operating-system #1142: ODROID C2 not booting with HA OS 5.9

agners, 2020-12-29:
The device is not booting with a fresh install. Also first installing HA 4.2 and then upgrading to HA 5.9 resulted in a unbootable situation..

Screen remains black, and not heartbeat (blue led) on the Odroid either...

_Originally posted by @BarBaar44 in https://github.com/home-assistant/operating-system/issues/1134#issuecomment-752137692_

agners, 2020-12-29:
@BarBaar44 are you using an eMMC? I do have a device updated to 5.9 with SD card running here...

XDjackieXD, 2020-12-29:
I couldn't get my Odroid C2 to boot with either 5.9 or 4.20.
The C2 works fine with archlinuxarm but when flashing HA OS to the microSD, it won't boot (I can see the bootloader output via HDMI but no HDMI output after that, no heartbeat LED and it doesn't show up via ethernet either).
Is there anything I have to do after flashing the SD card (in my case using dd) for it to work?

BarBaar44, 2020-12-29:
> @BarBaar44 are you using an eMMC? I do have a device updated to 5.9 with SD card running here...

Yes, running a brand new 32gb eMMC. Tested with Hardkernel's Ubuntu, and then it works flawlessly. So it's not the eMMC ;)

agners, 2020-12-29:
@XDjackieXD 
> Is there anything I have to do after flashing the SD card (in my case using dd) for it to work?
Not really. Do you have a USB to serial adapter by chance? The boot loader/kernel might have some debugging output on there.

So you don't have a eMMC just SD card and this does not boot for you? Interesting, that definitely works here.

agners, 2020-12-29:
@BarBaar44 
> Yes, running a brand new 32gb eMMC. Tested with Hardkernel's Ubuntu, and then it works flawlessly. So it's not the eMMC ;)

Ok, yeah I haven't tested with eMMC, will try that.

XDjackieXD, 2020-12-29:
> @XDjackieXD
>
>
> So you don't have a eMMC just SD card and this does not boot for you? Interesting, that definitely works here.

Exactly. I do see U-Boot briefly but then HDMI output turns off and nothing happens anymore for a long time. after about an hour or so it seems to reboot and uboot tries to do pxe-boot which fails. after that it just sits at the uboot shell.

BarBaar44, 2020-12-29:
Just remembered one more this:

My old(er) HA instance is running 5.9. I upgraded that over the years.

But I was moving to a new eMMC, then this issue began. I'm now running 4.2 and am hesitating to press that 'upgrade' button. But it may work if I go in smaller steps (4.20 -> 5.6 -> 5.7 etc)

oOoHoCo, 2020-12-30:
My ODROID C2 stopped working after upgrading to hassos 5.9, too.
When I connect a monitor it shows:

```
Could not get PHY for ethernet@9410000: addr -1
No ethernet found.
Retrieving file: pxelinux.cfg/default
```

this is shown probably 10 or more times and then I see a
`HassOS>`
prompt.

Is there any way to restore my installation?

Regards,
Ben

agners, 2020-12-30:
@oOoHoCo this seems to be a boot loader issue. Best is probably to reflash the working version and restore the snapshot. Probably fixable by dd'ing the boot loader from 5.8 image (`dd if=hassos_odroid-c2-5.x.img of=$DEV bs=512 skip=1 seek=1 count=2048`).

XDjackieXD, 2020-12-30:
> @oOoHoCo this seems to be a boot loader issue. Best is probably to reflash the working version and restore the snapshot. Probably fixable by dd'ing the boot loader from 5.8 image (`dd if=hassos_odroid-c2-5.x.img of=$DEV bs=512 skip=1 seek=1 count=2048`).

well but my fresh flash does exactly the same after sitting around for an hour without any sign of life (first uboot, then no sign of life for a long time and then the pxe attempts and uboot shell as described by @oOoHoCo)

agners, 2020-12-30:
So we have different issues: For @BarBaar44 and @oOoHoCo 4.20 did work with eMMC, but stopped with 5.9. I assume this has to do with the U-Boot update, a test with 5.2 would be interesting to confirm.

I can confirm that behavior on my end: Just tested 5.9 with eMMC and eMMC doesn't boot here either. It seems as if the eMMC is not detected at all by U-Boot. I'll have to do further investigation.

For @XDjackieXD it seems not to work with 4.20 or SD card, so this seems to be a different issue entirely. In this case the output of the serial UART would be helpful.

mikusp, 2020-12-30:
I can also confirm the issue, my Odroid C2 with eMMC stopped working today. I see some output from the bootloader trying to boot via PXE to no avail. Connecting my wireless keyboard with a USB dongle proved unsuccessful, as it doesn't register any key presses, even after displaying `HassOS>` prompt.

If there's anything I can provide to help you pinpoint the issue, please ask.

freddieleeman, 2020-12-31:
I had the same issie with 5.8 on my C2 with eMMC.

[image]
[image]
[image]

agners, 2020-12-31:
This will be fixed in 5.10. To fix a current installation you can `dd` a fixed boot loader. Download [u-boot.gxbb.gz](https://github.com/home-assistant/operating-system/files/5757100/u-boot.gxbb.gz) and run the following commands (replace `$DEVICE` with whatever the eMMC card reader's device name is, e.g. `/dev/sdg`:

```
gunzip u-boot.gxbb.gz
dd if="u-boot.gxbb" of="$DEVICE" conv=notrunc bs=512 seek=97
```

closed 2020-12-31 by agners

alex7614, 2021-01-02:
Hi
Exactly same problem on Odroid N2 on eMMC
HassOS 5.10, 5.9 or 5.8 on eMMC boot -> no boot
If I put eMMC on MicroSD eMMC reader and put the MicroSD on N2 -> works fine

Back to HassOS 5.5 on eMMC boot -> works

oOoHoCo, 2021-01-03:
So for now I have put the eMMC into the SD adapter and into the sd card slot on the odroid.
This seems to work.
The fixed bootloader did not work for me. I still got the same error message.

Hopefulle it will be fixed with the final release of 5.10

alex7614, 2021-01-03:
> So for now I have put the eMMC into the SD adapter and into the sd card slot on the odroid.
> This seems to work.
> The fixed bootloader did not work for me. I still got the same error message.
> 
> Hopefulle it will be fixed with the final release of 5.10

What do you mean by final 5.10 ? Because for me AND for N2 (no C2), i tried 5.10 and same problem.