home-assistant/operating-system #1151: RTC on Odroid N2+ 4Gb

Golab, 2021-01-01:
Hello I'm using Odroid N2+ 4Gb
This model has RTC,

My question is can I in someway check RTC to see if it's working?

HA: 2020.12.2 upgraded from old HA

Supervisor Version: 2020.12.7

Operating system version: 	5.9.15

agners, 2021-01-01:
The RTC is supposed to keep the real-time when there is no power. So to test it shutdown you device. Unplug power and Ethernet, wait for ~5 minutes or so (to make sure all capacitors etc. have been properly discharged), then power the board back on. If the time is set correctly (e.g. by login in using `root`, then `login` to get to the OS shell and typing `date`).

agners, 2021-01-01:
Please consider using Discord or our Community for questions.

closed 2021-01-01 by agners