home-assistant/operating-system #1154: Raspberry Pi4, USB SSD boot, HA Host does not see other USB devices.

sblantipodi, 2021-01-03:
**Hardware Environment**

- [X] Raspberry Pi 4 (4GB)

**Home Assistant OS release:**
- [X] Fresh installation of Home Assistant OS 5.9, supervisor 2020.12.7

```
~ $ ha info
arch: aarch64
channel: stable
docker: 19.03.13
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.9"
homeassistant: 2020.12.2
hostname: homeassistant
logging: info
machine: raspberrypi4-64
operating_system: Home Assistant OS 5.9
supervisor: 2020.12.7
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: Europe/Rome
~ $ 
```

**Supervisor logs:**

```
21-01-03 17:48:20 INFO (SyncWorker_7) [supervisor.docker.interface] Cleaning addon_a0d7b954_nut application
21-01-03 17:48:21 INFO (SyncWorker_7) [supervisor.docker.addon] Starting Docker add-on hassioaddons/nut-aarch64 with version 0.3.1
21-01-03 17:48:23 INFO (MainThread) [supervisor.api.security] /host/info access from a0d7b954_nut
21-01-03 17:48:51 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with a0d7b954_nut!
```

**Kernel logs:**

```
~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.79-v8 (builder@32116dd22ffe) (gcc version 9.3.0 (Buildroot -g45d6d28)) #1 SMP PREEMPT Fri Dec 18 21:50:34 UTC 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1012480
[    0.000000]   DMA zone: 3788 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242432 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 31 pages/cpu s89432 r8192 d29352 u126976
[    0.000000] pcpu-alloc: s89432 r8192 d29352 u126976 alloc=31*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996660
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=b58433c4fe834c2890f43902459357d9 cgroup_enable=memory fsck.repair=yes coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:0F:0D:A1 vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[... 781 lines cut ...]
[  302.953558] audit: type=1325 audit(1609696205.209:255): table=nat family=2 entries=8
[  302.957324] audit: type=1325 audit(1609696205.213:256): table=nat family=2 entries=10
[  302.962369] audit: type=1325 audit(1609696205.217:257): table=nat family=2 entries=11
[  302.967542] audit: type=1325 audit(1609696205.225:258): table=nat family=2 entries=12
[  302.972041] audit: type=1325 audit(1609696205.229:259): table=nat family=2 entries=13
[  303.013864] eth0: renamed from veth001b7c7
[  303.028111] IPv6: ADDRCONF(NETDEV_CHANGE): veth47d38d2: link becomes ready
[  303.028250] hassio: port 7(veth47d38d2) entered blocking state
[  303.028258] hassio: port 7(veth47d38d2) entered forwarding state
[  367.567100] udevd[192]: starting version 3.2.9
[  367.572016] udevd[193]: starting eudev-3.2.9
~ $ 
```

**Description of problem:**

I was using my Raspberry Pi4 4GB along with the Network UPS Tools to monitor my Cyberpower PR1500ELCD UPS.
UPS was connected to raspberry via USB and it worked flawlessly until I switched to a fresh install on HA OS 5.9 on an external USB SSD.

HA works very well on the SSD now but when the HA Host reboot, HA OS does not see the UPS.
I need to manually detach the UPS USB cable and reattach it to make it work.

I think that this is a common problem for connected USB devices when using a USB SSD instead of a microSD.

Thank you guys!!!

agners, 2021-01-03:
Interesting. Not sure what is happening here, it sounds like the EEPROM/RPi firmware enumerating the USB SSD makes the UPS not getting detected anymore... Is this reproducible with Raspberry Pi OS? If so, then this is probably something which  needs to be looked at by the Raspberry Pi developers.

sblantipodi, 2021-01-04:
@agners I attached my SSD with HassOS 5.9 to my PC, I opened the boot partition and added
"usbcore.old_scheme_first=Y" to /boot/cmdline.txt 

this fixed the problem.

is this something that you can fix for everyone?

sblantipodi, 2021-01-04:
another question if possible, 
is cmdline.txt overwritten when I update from HassOS 5.x to 5.x+1?

Asking this because I would not like to edit that file on every update 😄

sblantipodi, 2021-01-07:
unfortunantly the "usbcore.old_scheme_first=Y" solved nothing, behaviour is so random, sometimes it works, sometimes not.
we need a better solution 😄

sblantipodi, 2021-01-08:
I definetely solved the problem by adding quirks for just the UPS.
Tought that the quirks was only related to UAS devices like SSD case ecc...

Closing the issue.

thanks @agners I love when people helps other people. you guys are awesome here.

sblantipodi, 2021-01-08:
@agners ouch, it happened again, the problem is so random I tought it was solved, quircks solved nothing.
sorry for the many unuseful comments.

sblantipodi, 2021-01-08:
just tested on the official latest Raspbian OS, same error on Raspbian.

agners, 2021-01-09:
Hm, then its a driver/interaction issue between RPi 4 and that USB device. I would recommend open an issue in the official RPi Kernel repo.

sblantipodi, 2021-01-09:
@agners thank you for the reply, I appreciate it.
opened the issue as per your suggestion here, linking it here for others who may be interested:
https://github.com/raspberrypi/linux/issues/4056

and closing this one. thanks for the answer and for your work here. bye

closed 2021-01-09 by sblantipodi

sblantipodi, 2021-01-18:
for people who may read this issue, I solved it by changing a bootloader config:
```
sudo -E rpi-eeprom-config --edit
USB_MSD_PWR_OFF_TIME=0
```

when raspberry reboots, it power off the USB for the time specified in this option:
USB_MSD_PWR_OFF_TIME, the default value is 1000.

changing it to 0 solved my problem with my UPS.