home-assistant/operating-system #1155: Raspberry Pi Freezes completely on HA OS 5.9 when Hey Ada! integration is active

cbarreholm, 2021-01-03:
**Hardware Environment**

- [x ] Raspberry Pi 4 with 4 GB

**Home Assistant OS release:**
- [x] Updated from version 4.x
- Additional information (if accessible):
```
arch: armv7
audio_input: None
audio_output: None
boot: true
image: homeassistant/raspberrypi4-homeassistant
ip_address: 172.30.32.1
last_version: 2020.12.2
machine: raspberrypi4
port: 8123
ssl: false
update_available: true
version: 2020.12.1
version_latest: 2020.12.2
wait_boot: 600
watchdog: true
```
I have now updated 2020.12.2. Crashed again now almost exactly at midnight (23 UTC) when examining the kernel logs.

**Supervisor logs:**
```
ha > su logs
21-01-03 20:59:09 INFO (MainThread) [__main__] Initializing Supervisor setup
21-01-03 20:59:09 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-01-03 20:59:09 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4
21-01-03 20:59:09 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/armv7-hassio-supervisor with version 2020.12.7
21-01-03 20:59:09 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-01-03 20:59:09 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-01-03 20:59:09 INFO (MainThread) [__main__] Setting up Supervisor
21-01-03 20:59:09 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-01-03 20:59:10 INFO (MainThread) [supervisor.host.info] Updating local host information
21-01-03 20:59:10 INFO (MainThread) [supervisor.host.services] Updating service information
21-01-03 20:59:10 INFO (MainThread) [supervisor.host.network] Updating local network information
21-01-03 20:59:12 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-01-03 20:59:12 INFO (MainThread) [supervisor.host] Host information reload completed
21-01-03 20:59:12 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
21-01-03 20:59:12 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
21-01-03 20:59:12 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-dns with version 2020.11.0
21-01-03 20:59:12 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-01-03 20:59:12 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_dns application
21-01-03 21:39:35 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS homeassistant/armv7-hassio-dns with version 2020.11.0 - 172.30.32.3
21-01-03 21:39:35 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-01-03 21:39:35 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-audio with version 17
21-01-03 21:39:35 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
21-01-03 21:39:35 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_audio application
21-01-03 21:39:39 INFO (SyncWorker_0) [supervisor.docker.audio] Starting Audio homeassistant/armv7-hassio-audio with version 17 - 172.30.32.4
21-01-03 21:39:39 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-cli with version 2020.11.1
21-01-03 21:39:39 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
21-01-03 21:39:39 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_cli application
21-01-03 21:39:43 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI homeassistant/armv7-hassio-cli with version 2020.11.1 - 172.30.32.5
21-01-03 21:39:43 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-observer with version 2020.10.1
21-01-03 21:39:43 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-multicast with version 3
21-01-03 21:39:43 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
21-01-03 21:39:43 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_multicast application
21-01-03 21:39:45 INFO (SyncWorker_0) [supervisor.docker.multicast] Starting Multicast homeassistant/armv7-hassio-multicast with version 3 - Host
21-01-03 21:39:45 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-01-03 21:39:45 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
21-01-03 21:39:45 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/raspberrypi4-homeassistant with version 2020.12.1
21-01-03 21:39:45 INFO (MainThread) [supervisor.hassos] Detect HassOS 5.9 / BootSlot B
21-01-03 21:39:46 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
21-01-03 21:39:46 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
21-01-03 21:39:48 INFO (MainThread) [supervisor.store] Loading add-ons from store: 66 all - 66 new - 0 remove
21-01-03 21:39:48 INFO (MainThread) [supervisor.addons] Found 2 installed add-ons
[... 313 lines cut ...]
Jan 03 20:59:12 homeassistant hassos-supervisor[1023]: 21-01-03 20:59:12 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_dns application
Jan 03 20:59:12 homeassistant systemd-resolved[299]: Using degraded feature set UDP+EDNS0 instead of TLS+EDNS0 for DNS server 10.10.10.1.
Jan 03 20:59:12 homeassistant systemd-resolved[299]: Using degraded feature set UDP+EDNS0 instead of TLS+EDNS0 for DNS server 10.10.10.1.
-- Reboot --
Jan 03 20:59:19 homeassistant dbus-daemon[304]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.16' (uid=0 pid=306 comm="/usr/sbin/NetworkManager --no-daemon ")
Jan 03 20:59:19 homeassistant dbus-daemon[304]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Jan 03 21:01:41 homeassistant dbus-daemon[304]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.16' (uid=0 pid=306 comm="/usr/sbin/NetworkManager --no-daemon ")
Jan 03 21:01:41 homeassistant dbus-daemon[304]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
-- Reboot --
Jan 03 21:39:32 homeassistant systemd-timesyncd[300]: Initial synchronization to time server 162.159.200.1:123 (time.cloudflare.com).
....
an 03 21:45:12 homeassistant systemd-resolved[299]: Grace period over, resuming full feature set (TLS+EDNS0) for DNS server 10.10.10.1.
Jan 03 21:45:12 homeassistant systemd-resolved[299]: Grace period over, resuming full feature set (TLS+EDNS0) for DNS server 10.10.10.1.
Jan 03 21:45:12 homeassistant systemd-resolved[299]: Using degraded feature set UDP+EDNS0 instead of TLS+EDNS0 for DNS server 10.10.10.1.
Jan 03 21:45:12 homeassistant systemd-resolved[299]: Using degraded feature set UDP+EDNS0 instead of TLS+EDNS0 for DNS server 10.10.10.1.
Jan 03 21:45:15 homeassistant dropbear[4603]: [4603] Jan 03 21:45:15 Pubkey auth succeeded for 'root' with key sha1!!  <CLEANED_OUT>
Jan 03 21:45:28 homeassistant f19031de8a61[396]: 2021-01-03 22:45:28 WARNING (MainThread) [homeassistant.components.websocket_api.http.connection] [2875863272] Disconnected: Did not receive auth message within 10 seconds

```

I'm booting from a SD card. Since the problem occurred I moved over to a High Endurance SD card. I have also configured the recorder to use MariaDB in order to minimize writes. I also configured my WiFi to see if I could attach via that instead of ethernet.

Note that I haven't really been using Hey "Ada!" so I will probably just disable it for now. So I don't really know if there is any regression in the "Hey Ada!" functionality which could correlate to these OS issues.

I'm reporting this on the OS level, since the whole OS becomes unresponsive.

cbarreholm, 2021-01-10:
Now my system hang the same way again, but with `Hey Ada!` **disabled**.  I guess that the add-on just makes it quicker to trigger the issue. 

The system had 4 days and 20 hours up time this time:
```
# journalctl --list-boots | tail -2
 -1 829c25914bfe44a790b9a6765e1631e1 Tue 2021-01-05 08:49:16 UTC—Sun 2021-01-10 04:21:55 UTC
  0 24d5ee351f024f378e8ef7931015af9e Sun 2021-01-10 17:21:56 UTC—Sun 2021-01-10 21:50:28 UTC
```

Kernel logs today:
```
# journalctl -k -b all --since=2021-01-10
-- Logs begin at Tue 2020-10-27 23:34:38 UTC, end at Sun 2021-01-10 21:47:58 UTC. --
Jan 10 04:18:39 homeassistant kernel: uart-pl011 fe201000.serial: no DMA platform data
Jan 10 04:18:39 homeassistant kernel: random: crng init done
Jan 10 04:18:39 homeassistant kernel: random: 7 urandom warning(s) missed due to ratelimiting
Jan 10 04:18:39 homeassistant kernel: audit: type=1400 audit(1610252319.519:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="hassio-supervisor" pid=341 comm="apparmor_parser"
Jan 10 04:18:39 homeassistant kernel: audit: type=1400 audit(1610252319.519:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="hassio-supervisor///usr/bin/gdbus" pid=341 comm="apparmor_parser"
Jan 10 04:18:39 homeassistant kernel: audit: type=1400 audit(1610252319.519:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="hassio-supervisor///usr/bin/git" pid=341 comm="apparmor_parser"
Jan 10 04:18:39 homeassistant kernel: audit: type=1400 audit(1610252319.519:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="hassio-supervisor///usr/bin/socat" pid=341 comm="apparmor_parser"
Jan 10 04:18:39 homeassistant kernel: Bluetooth: Core ver 2.22
Jan 10 04:18:39 homeassistant kernel: NET: Registered protocol family 31
Jan 10 04:18:39 homeassistant kernel: Bluetooth: HCI device and connection manager initialized
Jan 10 04:18:39 homeassistant kernel: Bluetooth: HCI socket layer initialized
Jan 10 04:18:39 homeassistant kernel: Bluetooth: L2CAP socket layer initialized
Jan 10 04:18:39 homeassistant kernel: Bluetooth: SCO socket layer initialized
Jan 10 04:18:39 homeassistant kernel: Bluetooth: HCI UART driver ver 2.3
Jan 10 04:18:39 homeassistant kernel: Bluetooth: HCI UART protocol H4 registered
Jan 10 04:18:39 homeassistant kernel: Bluetooth: HCI UART protocol Three-wire (H5) registered
Jan 10 04:18:39 homeassistant kernel: Bluetooth: HCI UART protocol Broadcom registered
Jan 10 04:18:39 homeassistant kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Jan 10 04:18:39 homeassistant kernel: Bluetooth: BNEP filters: protocol multicast
Jan 10 04:18:39 homeassistant kernel: Bluetooth: BNEP socket layer initialized
Jan 10 04:18:40 homeassistant kernel: NET: Registered protocol family 38
Jan 10 04:18:40 homeassistant kernel: cryptd: max_cpu_qlen set to 1000
Jan 10 04:18:40 homeassistant kernel: bcmgenet: Skipping UMAC reset
Jan 10 04:18:40 homeassistant kernel: bcmgenet fd580000.ethernet: configuring instance for external RGMII
Jan 10 04:18:40 homeassistant kernel: bcmgenet fd580000.ethernet eth0: Link is Down
Jan 10 04:18:40 homeassistant kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Jan 10 04:18:45 homeassistant kernel: bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Jan 10 04:18:45 homeassistant kernel: IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Jan 10 04:18:45 homeassistant kernel: IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Jan 10 04:18:48 homeassistant kernel: audit: type=1400 audit(1610252328.069:3): apparmor="STATUS" operation="profile_load" profile="unconfined" name="docker-default" pid=421 comm="apparmor_parser"
Jan 10 04:18:53 homeassistant kernel: audit: type=1325 audit(1610252333.469:4): table=nat family=2 entries=5
Jan 10 04:18:53 homeassistant kernel: audit: type=1325 audit(1610252333.489:5): table=filter family=2 entries=4
Jan 10 04:18:53 homeassistant kernel: audit: type=1325 audit(1610252333.509:6): table=filter family=2 entries=6
Jan 10 04:18:53 homeassistant kernel: audit: type=1325 audit(1610252333.529:7): table=filter family=2 entries=8
Jan 10 04:18:53 homeassistant kernel: audit: type=1325 audit(1610252333.549:8): table=filter family=2 entries=10
Jan 10 04:18:53 homeassistant kernel: audit: type=1325 audit(1610252333.559:9): table=filter family=2 entries=11
Jan 10 04:18:53 homeassistant kernel: audit: type=1325 audit(1610252333.569:10): table=filter family=2 entries=12
Jan 10 04:18:53 homeassistant kernel: Initializing XFRM netlink socket
Jan 10 04:18:53 homeassistant kernel: audit: type=1325 audit(1610252333.659:11): table=nat family=2 entries=7
Jan 10 04:18:53 homeassistant kernel: audit: type=1325 audit(1610252333.669:12): table=nat family=2 entries=8
Jan 10 04:18:53 homeassistant kernel: audit: type=1325 audit(1610252333.699:13): table=filter family=2 entries=12
Jan 10 04:18:57 homeassistant kernel: hassio: port 1(vethe8dda49) entered blocking state
Jan 10 04:18:57 homeassistant kernel: hassio: port 1(vethe8dda49) entered disabled state
Jan 10 04:18:57 homeassistant kernel: device vethe8dda49 entered promiscuous mode
Jan 10 04:18:58 homeassistant kernel: cgroup: cgroup: disabling cgroup2 socket matching due to net_prio or net_cls activation
Jan 10 04:18:59 homeassistant kernel: kauditd_printk_skb: 55 callbacks suppressed
Jan 10 04:18:59 homeassistant kernel: audit: type=1325 audit(1610252339.119:69): table=nat family=2 entries=0
[... cut ...]
Jan 10 04:19:03 homeassistant kernel: IPv6: ADDRCONF(NETDEV_CHANGE): veth0449b7e: link becomes ready
Jan 10 04:19:03 homeassistant kernel: hassio: port 2(veth0449b7e) entered blocking state
Jan 10 04:19:03 homeassistant kernel: hassio: port 2(veth0449b7e) entered forwarding state
Jan 10 04:19:04 homeassistant udevd[205]: starting version 3.2.9
Jan 10 04:19:04 homeassistant udevd[206]: starting eudev-3.2.9
Jan 10 17:21:58 homeassistant kernel: kauditd_printk_skb: 12 callbacks suppressed
....
```

Any suggestions how to try to catch the root cause?

agners, 2021-01-11:
Could be related to #1129, which is solved in 5.10. Can/did you try 5.10?

agners, 2021-01-11:
Hm I think #1129 should at least print something on the console, in your case it seems to be a hard freeze. Memory looks good judging from the top output. Could it be that there is a power supply or thermal issue?

cbarreholm, 2021-01-11:
I have had an eye on #1129 before, but nothing like that in the logs.
Yesterday I updated to to 5.10, but had a freeze today. 
The power supply is an original Raspberry. The problem did start to occur immediately after upgrading from 4.x to 5.9. Before the system had been running stable for months. So if there is some hardware issue, it would then be related to the system behaving differently. Maybe `Hey Ada!` simply draws more CPU after upgrade, thus raising the temperature faster.
I have been checking the temperature of the Pi by hand whenever there has been a freeze as a way of checking if the CPU is spinning. Sometimes it has been hot, other times it has been fairly cool. 

Is something like `sensors` (https://hwmon.wiki.kernel.org/lm_sensors) available in Hassio? Thinking that I could start polling the hardware temperatures at the console before a potential freeze.

I have another identical Raspberry Pi, which I could switch to. Could I just move the SD card to that one? 

Would it work to flash a fresh Hassio to a completely new SD card and then restore HA snapshots? (I guess that it would be a great disaster recovery exercise I probably should do ....)

agners, 2021-01-11:
> Yesterday I updated to to 5.10, but had a freeze today.

:-(

> The power supply is an original Raspberry. The problem did start to occur immediately after upgrading from 4.x to 5.9.

We moved to the new 5.4 kernel, so that of course can cause many things. That said, I (and others) had/have Home Assistant running on the 5.4 kernel for at least a month without issues on my end, so it must be some interaction with Add-Ons and/or hardware on your side.

> Is something like `sensors` (https://hwmon.wiki.kernel.org/lm_sensors) available in Hassio? Thinking that I could start polling the hardware temperatures at the console before a potential freeze.

There is the System Monitor integration which I think reads directly from sysfs: https://www.home-assistant.io/integrations/systemmonitor/

> I have another identical Raspberry Pi, which I could switch to. Could I just move the SD card to that one?
If it is also a Raspberry 4 just switching SD should work fine. Worth a try.

> Would it work to flash a fresh Hassio to a completely new SD card and then restore HA snapshots? (I guess that it would be a great disaster recovery exercise I probably should do ....)

Yeah snapshot work really well. You can also create a snapshot, and then flash the new image on a new SD just to be sure. I usually use snapshots to move to new hardware and leave the old system for some days just in case migration goes wrong/new hardware makes problem. So far had not to go back, always worked out.

cbarreholm, 2021-01-11:
Found the command line sensor
https://www.home-assistant.io/integrations/sensor.command_line/
so have added monitoring of the CPU temperature. Currently 45 - 47 degrees C. Added system monitoring as well. Also moved the Pi to the marble window sill with leaky 1930:s windows keeping it cooler. Currently at 38 degrees.

I have also flashed a brand new WD Purple high endurance SD card with Hass.io 5.10 and restored a snapshot running on my second Raspberry Pi. So now I basically is running with completely new hardware, except for the same power supply.

cbarreholm, 2021-01-13:
I ran the system for a full day, including "Hey Ada!", without an issue. Then I switched to my initial Raspberry Pi with the fresh SD card and that setup also runs fine.

Note that once I started having issues mid-December I did make a `dd` clone of the initial non-high endurance SD card and switched to a high endurance SD card. The clone went fine, so no issues reading the data. But I still had crashes.

So I'm thinking that a difference here is that in my initial scenario I upgraded from HA OS 4.x to 5.9 and then later 5.10. But now I made a fresh 5.10 install. 

*Could there possibly be an issue with the upgrade process?* 

Can I find out from the logs which version 4.x I upgraded from? Thinking that I could flash a new SD card with that version and upgrade to 5.9 to see if I get the same issue. That way ruling out that the initial SD card was corrupt.

cbarreholm, 2021-01-16:
I had a freeze again last night with the fresh installed system (the new SD card), but running on the initial Raspberry Pi. **So it's not the OS upgrade process.**

I can see that the CPU temperature has been stable around 37 degrees C, which shouldn't be an issue:
[image]

And the CPU load was also stable before the freeze:
[image]

I will let it run for some while more on the initial Raspberry Pi to see if it happens again. Then switching to the new Raspberry Pi to rule out the HW.

agners, 2021-01-19:
Thanks for the update. So when this happens you really don't see anything on the HDMI screen right?

What might be worth a try is if using Supervised on Raspberry Pi OS December release shows the same problem. We use the very same kernel, so if it happens in that combination as well, its most likely a RPi kernel issue.

cbarreholm, 2021-01-19:
Another freeze on my initial RPi today. 

This graph shows the CPU temperature. 
[image]
I moved it off from my chilly marble window till, causing the temperature to increase. 
[image]
The CPU load was constant so not the cause of the increase.
And then it froze. 

45 degrees is certainly not extreme, but I guess that it might be some other component that fails.

I have now switched to my newer RPi (same SD) to see if there is some difference.

When it freezes it looks like the pictures in the initial report. Visible but frozen. Pressing Caps Lock on the keyboard won't toggle the indicator led.

Are you referring to this setup?
https://community.home-assistant.io/t/installing-home-assistant-supervised-on-a-raspberry-pi-with-debian-10/247116

agners, 2021-01-19:
> 45 degrees is certainly not extreme, but I guess that it might be some other component that fails.

45 degree is quite cool actually. The RPi should also not really overheat as it starts to throttle at 85 degree.
> Are you referring to this setup?
No, this setup:
https://community.home-assistant.io/t/installing-home-assistant-supervised-on-raspberry-pi-os/201836

cbarreholm, 2021-01-22:
Three days on the newer Pi (identical specs) with same SD card without an issue. Leaning more and more towards HW issue. I would say if it stays up for a week without issues, then close the ticket.

cbarreholm, 2021-02-04:
The newer Pi works fine. Closing as HW issue.

closed 2021-02-04 by cbarreholm