home-assistant/operating-system #1156: udev rules for proxmox hotplug cpu/mem

chatainsim, 2021-01-04:
Hello,
I'm using hass OS in Proxmox 6.3 and VM only see 972Mi of RAM:
[image]

So i've try to add a udev rules in /mnt/boot/CONFIG/udev/80-hotplug-cpu-mem.rules, then reboot.
```
SUBSYSTEM=="cpu", ACTION=="add", TEST=="online", ATTR{online}=="0", ATTR{online}="1"
SUBSYSTEM=="memory", ACTION=="add", TEST=="state", ATTR{state}=="offline", ATTR{state}="online"
```
File has been copied in /etc/udev/rules.d/ but nothing happen for memory.
How can I add rules in /lib/udev/rules.d ?

Thank you.

Regards,

agners, 2021-01-05:
If it is in `/etc/udev/rules.d/ ` the rules should be active. It seems the kernel is lacking memory hotplug support:

```
# CONFIG_MEMORY_HOTPLUG is not set 
```

I'll add it in the next release.

agners, 2021-01-06:
I tried hotplugging using KVM yesterday, memory hotplug seemed to just work even without this change, not sure why it works with KVM even without this option... But CPU did not work (Virtual Machine Manager said I have to shutdown the machine...). I guess this would  need further conflagration.

Anyway, I created a development build with `CONFIG_MEMORY_HOTPLUG` enabled and also `CONFIG_MEMORY_HOTPLUG_DEFAULT_ONLINE` set which should make the memory available even without the udev rule. I also added the udev rule to set the CPU online by default. Can you maybe give this a try? https://os-builds.home-assistant.io/6.0.dev20210105/

chatainsim, 2021-01-06:
I've tried the dev build but it's the same:
[image]

I've added 2Go of RAM and tried to add 1 CPU with 2 core but it's not take into account:
[image]

agners, 2021-01-07:
Did adding CPU work when you added the udev rule manually?

chatainsim, 2021-01-07:
I wasn't able to add rules in /lib/udev/rules.d
And rules in /etc/udev/rules.d wasn't working.
So for now I've remove hotplug for CPU and RAM so my VM can use the 2Go of RAM.

Thanks

agners, 2021-01-08:
> And rules in /etc/udev/rules.d wasn't working.
They are working, as they get loaded *exactly* the same way as they get loaded from `/lib/udev/rules.d`

In the development release there should be also `/lib/udev/rules.d/80-hotplug-cpu.rules`...

chatainsim, 2021-01-08:
Ok, yes, file /lib/udev/rules.d/80-hotplug-cpu.rules exist in dev version.
Adding theses rules work fine with Debian or Ubuntu VM.
Can you also add memory hotplug ?
Thanks

agners, 2021-01-08:
The memory hotplug udev rule is not necessary as I enabled `CONFIG_MEMORY_HOTPLUG_DEFAULT_ONLINE`. This is better because:
```
Currently, all newly added memory blocks remain in 'offline' state unless
someone onlines them, some linux distributions carry special udev rules
like:

SUBSYSTEM=="memory", ACTION=="add", ATTR{state}=="offline", ATTR{state}="online"

to make this happen automatically. This is not a great solution for virtual
machines where memory hotplug is being used to address high memory pressure
situations as such onlining is slow and a userspace process doing this
(udev) has a chance of being killed by the OOM killer as it will probably
require to allocate some memory.
```

See: https://lwn.net/Articles/668944/

chatainsim, 2021-01-14:
> The memory hotplug udev rule is not necessary as I enabled `CONFIG_MEMORY_HOTPLUG_DEFAULT_ONLINE`. This is better because:
>
> ```
> Currently, all newly added memory blocks remain in 'offline' state [...]

Thanks, I didn't know that, I'm pretty new in virtualization. 👍

agners, 2021-01-19:
Closing as this will be fixed in version 6.

closed 2021-01-19 by agners