home-assistant/operating-system #1157: add fw for dvb tuner

luci-11, 2021-01-05:
I'm trying to modify an existing addon for tvheadend in order to make it compatible with an usb tuner I have at home (MyGica T230C v2) .
The tuner is recognized from tvheadend cofiguration panel but cannot get any channles.
Dmesg confirms USB is recognized properly but I have: `Direct firmware load for dvb-demod-si2168-d60-01.fw failed with error -2`
Same for `dvb-tuner-si2141-a10-01.fw`

Tried tvheadend on raspbian adding these two files manually as written here [https://www.linuxtv.org/wiki/index.php/Geniatech_T230C](https://www.linuxtv.org/wiki/index.php/Geniatech_T230C)
and everything works.

Could you please add them to /lib/firmware for hassos build?
files are here: [t230C.v2-fw](https://github.com/luci-11/t230C-v2-fw)
Thanks a lot.

agners, 2021-01-11:
It seems that this firmware is **not** available from the [linux-firmware git tree](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/). This would mean I need to create a separate buildroot package and all that, which is a bit much. Also, I am not sure about the license. Usually there is a reason if something doesn't find its way into linux-firmware.

admin-lm, 2021-01-11:
Could you add a mount point to /usr/lib/firmware which is writable for the users, instead of creating a seperate buildroot package, to allow adding fw files by the user itself?

agners, 2021-01-11:
@admin-lm the problem is we already ship firmware with the OS today. So we would have to do it in such a way that we can still ship firmware with the OS, e.g. an overlay or something. The question then is how do we make it accessible exactly, e.g. via container from supervisor?

luci-11, 2021-01-11:
> It seems that this firmware is **not** available from the [linux-firmware git tree](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/). This would mean I need to create a [...]

MMM ..Strange story. 
For licence, I think they were created using CrazyCat's media_build tool that is mentioned in the Linux page liked above. He is the owner also for the added support in linux kernel [Si2168-D60 support](https://patchwork.kernel.org/project/linux-media/patch/3148066.V8NoLtYRlU@computer/) . Commits are also there [linux-next](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/log/?qt=grep&q=si2168) . 
In fact adding just the two fw does the job in Raspbian OS..   Strange

admin-lm, 2021-01-11:
I think an overlay for /lib/firmware helps, devices could be made accessible via portainer; /dev/dvb:/dev/dvb

Billiadis, 2022-09-27:
hi, how exactly did you solve this problem?
how to create an overlay for /lib/firmware?

weidi, 2022-11-14:
Would it help to have CONFIG_DVB_USB=m added to device-support.config ?
This has been done but reverted previously #609

HahnBenjamin, 2023-01-08:
I just found a viable solution to the custom firmware issue for my USB DVB Tuner I was trying to attach to my HAOS setup.

There are actually multiple [firmware search paths](https://www.kernel.org/doc/html/latest/driver-api/firmware/fw_search_path.html) the kernel uses to find  firmware files to upload to the device.

There is even a option to specify a custom search path, which is what I utilized.
The path can be specified using a kernel boot parameter. I my case I placed the firmware files in the /share folder exposed to the supervisor, specifically `/share/firmware`. 
Within the underlying operating system the path translates to `/mnt/data/supervisor/share/firmware`.  This path is what is need for the kernel parameter. `firmware_class.path=/mnt/data/supervisor/share/firmware/` must be added to *cmdline.txt* on the PI SD card.
On the next boot the HAOS will happily load the firmware from the specified location.

@agners I am not sure if the usecase of user specific device firmware is relevant enough to establish a more streamlined solution. The built-in the firmware search paths  could be utilized for this. If I understand correctly e.g ‘/lib/firmware/updates/’ could be symlinked\bind mounted to some writeable location within HAOS. If this location would then be exposed to the supervisor there would be a working solution. This should also not interfere with the firmware currently shipped with HAOS.

thatsthat, 2023-03-15:
> I just found a viable solution to the custom firmware issue for my USB DVB Tuner I was trying to attach to my HAOS setup.
>
> There are actually multiple [firmware search [...]

I am trying to use your solution but dmesg output still complains about not finding the file. Do you know if I have to load some extra kernel module?

thatsthat, 2023-03-16:
> I just found a viable solution to the custom firmware issue for my USB DVB Tuner I was trying to attach to my HAOS setup.
>
> There are actually multiple [firmware search [...]

Your solution worked, but only after manually reloading the dvb-usb module after each reboot, otherwise the problem persists. Any idea on how to automate this? HAOS doesn't allow automatically running scripts with root permissions AFAIK

thatsthat, 2023-03-27:
> Is there any progress about this issue?
> 
> I also would like to add some firmware files for TVHeadend on my Home Assistant:
> 
> ```
> # How to download AverMedia TD310 (DVB-C/T/T2 USB stick) firmware files (used on Raspberry Pi 4 Model B)
> #
> sudo wget https://github.com/OpenELEC/dvb-firmware/raw/master/firmware/dvb-usb-it9303-01.fw -O /lib/firmware/dvb-usb-it9303-01.fw
> sudo wget https://github.com/OpenELEC/dvb-firmware/raw/master/firmware/dvb-demod-si2168-02.fw -O /lib/firmware/dvb-demod-si2168-02.fw
> sudo wget https://github.com/OpenELEC/dvb-firmware/raw/master/firmware/dvb-demod-si2168-b40-01.fw -O /lib/firmware/dvb-demod-si2168-b40-01.fw
> ```
> 
> I saved the files under `/lib/firmware/` as indicated above, but after restarting the system, the `/lib/firmware/` folder is empty.
> 
> It would be great to make this possible. Thank you!
> 
> PS1: Rebooting the system seems to delete these files, whereas restarting Home Assistant keeps them.
> 
> PS2: Does this help? [#2222 (comment)](https://github.com/home-assistant/operating-system/issues/2222#issuecomment-1308959053) A permanent solution to be able to have drivers stored in /lib/firmware would be great.

There is already a solution:

> I just found a viable solution to the custom firmware issue for my USB DVB Tuner I was trying to attach to my HAOS setup.
>
> There are actually multiple [firmware search [...]

agners, 2023-03-27:
> But the kernel can be instructed to get the firmware from any folder accessible to addons (e.g. under /mnt/data/supervisor/share)

Yes. With the chosen solution this option is still available to you. :man_shrugging:

thatsthat, 2023-03-31:
> Please forgive me, guys, but I am not such an expert and still unsure what
> this now means for me. After I updated to the latest HA version, what do I
> have to do to permanently store drivers/firmware files, i.e. where should I
> store them? Thanks for your patience and advice! :)

You need to follow the instructions given by HahnBenjamin (posted January
8th)

thatsthat, 2023-04-01:
> I.e., this is all that's needed?

Yes, assuming you copied your firmware files into that folder

thatsthat, 2023-04-01:
> I just flashed the HAOS image for the RPi 4
> *fresh* to a nano SD card and opened cmdline.txt in an editor that shows
> just this *one line*:
> dwc_otg.lpm_enable=0 console=tty1
> usb-storage.quirks=174c:55aa:u,2109:0715:u,152d:0578:u,152d:0579:u,152d:1561:u,174c:0829:u,14b0:0206:u
>
> Can I simply append the path, so the cmdline.txt afterwards looks as
> follows, i.e. just a blank between "u" and "firmware_class.path" etc.?
> (still *one line*)
> dwc_otg.lpm_enable=0 console=tty1
> usb-storage.quirks=174c:55aa:u,2109:0715:u,152d:0578:u,152d:0579:u,152d:1561:u,174c:0829:u,14b0:0206:u
> firmware_class.path=/mnt/data/supervisor/share/firmware/
>
> Thank you! 😃

Yes, it must remain one line, with spaces between different commands, it's
also important that you use a pure text editor (like notepad++ on windows)

admin-lm, 2023-05-26:
I switched to [Home Assistant Supervised](https://www.home-assistant.io/installation/linux#install-home-assistant-supervised)

agners, 2023-07-17:
This is an issue tracker and not a support forum. I've responded to your community post.

agners, 2023-11-07:
Sorry for failing to recognize the problem here exactly, glad you found how to progress.

Note however, as I posted in this [community post](https://community.home-assistant.io/t/how-to-use-firmware-search-path-e-g-mnt-data-supervisor-share-firmware/573387/11?u=agners) and noted in this comment https://github.com/home-assistant/operating-system/issues/1157#issuecomment-1471634024:

The alternative firmware path is available a bit later during boot. For many devices this means that the driver will not be able to load the firmware because the driver is loaded earlier at boot time, leading to the exact same issue.

You can work around by hot-plugging the device (if the device is connected through USB or similar). But if not, placing the firmware there likely won't work, unfortunately :cry: 

I tried to find a solution to that, but from what I understand, there is really no way to tell the kernel to delay loading of firmwares until a later point in time. The HAOS architecture of using a read-only file system really doesn't allow to include firmware into the root partition which is available earlier. So what we are trying to do (add firmware to a mounted location) really seems not to work :cry: If someone has an idea how to implement this, I would be grateful.

Until then, for non-hotpluggable device (or people which don't want to work around the limitation), using Supervised is probably the best alternative right now.

agners, 2023-12-20:
> After unplugging and replugging it, it's visible again. Is this due to the problem you described in post [#1157 (comment)](https://github.com/home-assistant/operating-system/issues/1157#issuecomment-1798225576)?

Yeah the replugging is explained by that. I am not sure why the device disappeared above. Once it is there, it should stay there. Unless there is some USB bus problems, but in that case you should see some errors in the host logs.