home-assistant/operating-system #1159: Hassos 5.10 Rpi 4 2GB no USB boot

theslash, 2021-01-07:
- [x] Raspberry Pi 4
**Home Assistant OS release: 5.10 / 5.9
- [x] Fresh installation of release x.y

Just tried a fresh install of HA on my Raspberry pi 4 2GB.
With two different USB Stickst it won't boot, just showing a blank screen.
Device gets an IP Address, I am able to ping but after 15 min it just shuts down.

Trying a SD Card for now.

agners, 2021-01-07:
Did it work with 5.8? What firmware version are you using? What type of USB stick are you using? Are you using the 64-bit or 32-bit image?

theslash, 2021-01-08:
Didn't try 5.8, but my test system on RPi 3 ran on 5.8 with USB and no Problems.
32Bit Version every time, tried a Sandisk 64GB, A Kingston 32GB Stick, both new.

Runs on a SD Card perfectly.

agners, 2021-01-08:
Unfortunately, USB SSD boot on RPi 4 is not that reliable. What I would recommend is using a SD card for the main operating system and use the `datactl` command to move just the data to the USB flash drive, see: https://github.com/home-assistant/operating-system/blob/dev/Documentation/partition.md#data-disk

theslash, 2021-01-09:
Thats sad. is that a hassos or a raspberry problem? I have a few raspberry 3's run on USB Sticks without problems.
Thanks for your suggestion.

agners, 2021-01-11:
Probably a U-Boot problem (part of Home Assistant OS). You would have to connect a USB to serial adapter to tell for sure.

theslash, 2021-01-12:
Ok, I'll close this here, thanks for your information

closed 2021-01-12 by theslash