home-assistant/operating-system #1160: 5 minutes to boot HA from a Pi4 4GB and a Samsung 860EVO SSD

sblantipodi, 2021-01-07:
**Hardware Environment**

- [X] Raspberry Pi 4 (4GB)

**Home Assistant OS release:**
- [X] Fresh installation of Home Assistant OS 5.9, supervisor 2020.12.7

```
~ $ ha info
arch: aarch64
channel: stable
docker: 19.03.13
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.9"
homeassistant: 2020.12.2
hostname: homeassistant
logging: info
machine: raspberrypi4-64
operating_system: Home Assistant OS 5.9
supervisor: 2020.12.7
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: Europe/Rome
~ $ 
```

**Supervisor logs:**

```
21-01-03 17:48:20 INFO (SyncWorker_7) [supervisor.docker.interface] Cleaning addon_a0d7b954_nut application
21-01-03 17:48:21 INFO (SyncWorker_7) [supervisor.docker.addon] Starting Docker add-on hassioaddons/nut-aarch64 with version 0.3.1
21-01-03 17:48:23 INFO (MainThread) [supervisor.api.security] /host/info access from a0d7b954_nut
21-01-03 17:48:51 WARNING (MainThread) [supervisor.misc.tasks] Watchdog found a problem with a0d7b954_nut!
```

**Kernel logs:**

```
~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.79-v8 (builder@32116dd22ffe) (gcc version 9.3.0 (Buildroot -g45d6d28)) #1 SMP PREEMPT Fri Dec 18 21:50:34 UTC 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1012480
[    0.000000]   DMA zone: 3788 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242432 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 31 pages/cpu s89432 r8192 d29352 u126976
[    0.000000] pcpu-alloc: s89432 r8192 d29352 u126976 alloc=31*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996660
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=b58433c4fe834c2890f43902459357d9 cgroup_enable=memory fsck.repair=yes coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:0F:0D:A1 vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[... 778 lines cut ...]
[  302.931446] audit: type=1325 audit(1609696205.189:252): table=filter family=2 entries=0
[  302.942567] audit: type=1325 audit(1609696205.197:253): table=nat family=2 entries=5
[  302.946278] audit: type=1325 audit(1609696205.201:254): table=nat family=2 entries=7
[  302.953558] audit: type=1325 audit(1609696205.209:255): table=nat family=2 entries=8
[  302.957324] audit: type=1325 audit(1609696205.213:256): table=nat family=2 entries=10
[  302.962369] audit: type=1325 audit(1609696205.217:257): table=nat family=2 entries=11
[  302.967542] audit: type=1325 audit(1609696205.225:258): table=nat family=2 entries=12
[  302.972041] audit: type=1325 audit(1609696205.229:259): table=nat family=2 entries=13
[  303.013864] eth0: renamed from veth001b7c7
[  303.028111] IPv6: ADDRCONF(NETDEV_CHANGE): veth47d38d2: link becomes ready
[  303.028250] hassio: port 7(veth47d38d2) entered blocking state
[  303.028258] hassio: port 7(veth47d38d2) entered forwarding state
[  367.567100] udevd[192]: starting version 3.2.9
[  367.572016] udevd[193]: starting eudev-3.2.9
~ $ 
```

**Description of problem:**

As title.
I have a compatible USB cable using a compatible chip that drives my Samsung 860EVO SSD on a Pi4 4GB.

Booting from USB is slower than booting from microSD.

Is there someone working on this? It' seems that everyone are affected, a host reboot requires 5 minutes here.

agners, 2021-01-07:
It seems that Linux has issues communicating with the SSD. -71 means protocol error. Not sure what the issue is exactly, did you try using different ports/cables? Also sometimes it helps to put a external USB 3.0 Hub inbetween (at least for a test) to see if it helps.

Some SSD's also need special Kernel parameters to work with Raspberry Pis, did you try using USB quirks? https://www.raspberrypi.org/forums/viewtopic.php?t=245931

> Is there someone working on this?

The problem is, there are many USB SSD's out there and some fail and some don't. It is almost impossible to get everything supported, as we would need all the devices out there. That particular device I don't have, so I cannot try to reproduce or fix this issue. But a patch is always welcome ;-)

> It' seems that everyone are affected,

Why do you think so?

I have a test setup which works fine here using a Samsung T1 and it seems to work fine.

sblantipodi, 2021-01-07:
@agners what do you mean to work fine?
have you ever tried to use a chronometer to see how long it take for a full host reboot?

If you boot in HA, then make a full host reboot. how long it take?

20 seconds? 5 minutes?

agners, 2021-01-07:
Wait, you don't trust the OS developer to judge if his system works fine? I think I would switch the OS :-P It works fine here, I also don't have any errors in `dmesg` as in your case.

Just measured, system boots in 29s from power cycle, ~10s is spent in Raspberry Pi firmware. Reboot took 47s.

No its your turn to answer questions, after all I wrote them first: https://github.com/home-assistant/operating-system/issues/1160#issuecomment-756394639

sblantipodi, 2021-01-08:
@agners thanks for the reply.
I obviously trust you, I was asking what do you mean for fine, 5 seconds, 5 minutes, you answered 😄  thanks!

I'll try to answer all your questions.

1) -71 is related to a USB UPS I have attached, if I remove the UPS the error vanish but the boot speed does not improve, here an updated dmesg without that error (without the UPS attached to the pi4):

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.79-v8 (builder@32116dd22ffe) (gcc version 9.3.0 (Buildroot -g45d6d28)) #1 SMP PREEMPT Fri Dec 18 21:50:34 UTC 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1012480
[    0.000000]   DMA zone: 3788 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242432 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 31 pages/cpu s89432 r8192 d29352 u126976
[    0.000000] pcpu-alloc: s89432 r8192 d29352 u126976 alloc=31*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996660
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=183172201d2b47c3920e8abae75ae31b cgroup_enable=memory fsck.repair=yes coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:0F:0D:A1 vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37300000-0x3b300000] (64MB)
[    0.000000] Memory: 3819704K/4049920K available (11068K kernel code, 1176K rwdata, 3744K rodata, 1088K init, 1232K bss, 164680K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 36302 entries in 142 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x320/0x4b0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000222] Console: colour dummy device 80x25
[    0.000647] printk: console [tty1] enabled
[    0.000703] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000737] pid_max: default: 32768 minimum: 301
[    0.000879] LSM: Security Framework initializing
[    0.001006] AppArmor: AppArmor initialized
[    0.001213] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001283] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.003568] ASID allocator initialised with 32768 entries
[    0.003782] rcu: Hierarchical SRCU implementation.
[    0.004505] EFI services will not be available.
[    0.004978] smp: Bringing up secondary CPUs ...
[    0.005866] Detected PIPT I-cache on CPU1
[    0.005924] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.006937] Detected PIPT I-cache on CPU2
[    0.006976] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.007976] Detected PIPT I-cache on CPU3
[    0.008016] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.008151] smp: Brought up 1 node, 4 CPUs
[    0.008238] SMP: Total of 4 processors activated.
[    0.008261] CPU features: detected: 32-bit EL0 Support
[    0.008283] CPU features: detected: CRC32 instructions
[    0.034428] CPU: All CPU(s) started at EL2
[... 534 lines cut ...]
[  184.329620] audit: type=1700 audit(1610070310.189:214): dev=vethae28b6f prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[  184.719772] audit: type=1325 audit(1610070310.577:215): table=nat family=2 entries=0
[  184.728193] audit: type=1325 audit(1610070310.585:216): table=filter family=2 entries=0
[... cut ...]
[  184.814369] eth0: renamed from vethcc7faad
[  184.838425] IPv6: ADDRCONF(NETDEV_CHANGE): vethae28b6f: link becomes ready
[  184.838586] hassio: port 7(vethae28b6f) entered blocking state
[  184.838597] hassio: port 7(vethae28b6f) entered forwarding state
```

2) I Have tried all the USB ports on the PI, same long boot

3) Yes I have tried a different cable, if I attach this disk to the PC, I can get 500MB/sec easy, so the case+cable should be fine

4) I don't have an USB hub to try 😞 

5) I have tried USB quircks

nothing helped. :(

agners, 2021-01-08:
> -71 is related to a USB UPS I have attached, if I remove the UPS the error vanish but the boot speed does not improve, here an updated dmesg without that error (without the UPS attached to the pi4):

Hm, I see, so this errors are probably unrelated.

I must say, my measurements are time it takes until the login prompt appears. What is your measurement point? Since we use multiple containers, it takes a while until everything comes up, so complete bootup is probably closer to 3 minutes, but that is pretty normal.

Please also note that 500MB/sec is sequential read. A boot-up reads pieces from random places, that is where random values come in. In my measurements, my Samsung did "only" 83.1MiB/s in random read and 30.8MiB/s in random writes.

sblantipodi, 2021-01-08:
my measurament is from the "click on the reboot button" in the Home Assistant GUI,
to the moment that home assistant "says that it is up and running".

have some addons, not many, but the point is that is strange that microsd was faster. and this "slowness" is causing this issue #1154

agners, 2021-01-10:
Did you measure SD card with the same OS version?

sblantipodi, 2021-01-11:
@agners yes

agners, 2021-01-11:
I tracked down a 60s slowdown due to Home Assistant Core not shutting down when using the reboot button, see: https://github.com/home-assistant/core/pull/45033.

However, on my end, I really don't see a huge difference between SD card and USB SSD using the same version.

closed 2022-01-11 by github-actions[bot]