home-assistant/operating-system #1161: Power off and wake up using GPIO key button and heartbeat LED action

dieneuser, 2021-01-08:
- [ ] ODROID [N2(+)]

I installed a switch with LED in the odroid n2+ and was able to use it under Ubuntu from hardkernel: (https://wiki.odroid.com/odroid-n2/application_note/gpio/gpio_key_wakeup)
Is it possible to use this function under Home Assistant OS too?
A configuration of the gpiopower flag in the boot.scr did nothing - Doesn't the kernel support that? (argument gpiopower=479 for power button on PIN 11+9)
And a look in the dtb file from home assistant os shows that the parameter and value for the blue LED pin are different to the solution from hardkernel. (PIN 8+6)

thx

agners, 2021-01-11:
Hm, Raspberry Pi solves such kind of hardware modification using device tree overlays. I'd rather prefer not to add/support overlays for ODROID-N2(+). But we could just declare those pins as "power button pins" by default.

We use upstream/mainline Linux, whereas ODROID's Ubuntu uses something downstream. The `gpiopower` kernel argument is a downstream/custom kernel argument. But I think its not required to have such an argument if we would just declare it as default function of those pins.

agners, 2021-01-11:
> And a look in the dtb file from home assistant os shows that the parameter and value for the blue LED pin are different to the solution from hardkernel. (PIN 8+6)

Looks pretty much the same to me:

Home Assistant OS/upstream kernel:
```
	leds {
		compatible = "gpio-leds";

		blue {
			label = "n2:blue";
			gpios = <&gpio_ao GPIOAO_11 GPIO_ACTIVE_HIGH>;
			linux,default-trigger = "heartbeat";
		};
	};
```

Hardkernel 4.9 kernel
```
	leds {
		compatible = "gpio-leds";
		blueled {
			label = "blue:heartbeat";
			gpios = <&gpio_ao GPIOAO_11 GPIO_ACTIVE_HIGH>;
			linux,default-trigger = "heartbeat";
		};
	};
```

Any problems with the pulse led?

dieneuser, 2021-01-11:
> Hm, Raspberry Pi solves such kind of hardware modification using device tree overlays. I'd rather prefer not to add/support overlays for ODROID-N2(+). But we could just declare those pins as "power button pins" by default.

This sounds good, i prefer the same PINs like in the Odroid Wiki (9+11).

> Home Assistant OS/upstream kernel:
>
>```
>	leds {
>		compatible = "gpio-leds";
>
>		blue {
>			label = "n2:blue";
>			gpios = <&gpio_ao GPIOAO_11 GPIO_ACTIVE_HIGH>;
>			linux,default-trigger = "heartbeat";
>		};
>	};
>```
In my installation it looks different:
Home Assistant OS 5.10 dtb:
```
	leds {
		compatible = "gpio-leds";

		blue {
			label = "n2:blue";
			gpios = < 0x48 0x0b 0x00 >;
			linux,default-trigger = "heartbeat";
		};
```
Values: 72 11 0

Hardkernel dtb:
```
	leds {
		compatible = "gpio-leds";

		blueled {
			label = "blue:heartbeat";
			gpios = < 0x18 0x0b 0x00 >;
			linux,default-trigger = "heartbeat";
		};
```
Values: 24 11 0

And for the LED on pin 8+9 i need 26 78 0

The big question is, when the first value 72 is the onboard GIPO_AO from Home Assistant OS, what value is for the GPIO header?

Then I can simply issue the rewrite command: `fdtput meson-g12b-odroid-n2-plus.dtb /leds/blue gpios ?? 78 0` for the button LED.

agners, 2021-01-11:
Oh I see you are trying to use the LED on the pin header pin number 8, which is GPIOX_12. The button is GPIOX_3, so in theory it should work by just using what the button is using (`fdtget buildroot/output_odroid_n2/images/meson-g12b-odroid-n2.dtb /gpio-keys-polled/power-button gpios`) as GPIO reference shifted by 9.

Btw, the first number is device tree internal reference (think of it as pointer), and this can be different every time a device tree gets compiled. In other words, a new Home Assistant OS release might have different numbers... The second number (GPIO number within that controller) is stable in upstream/mainline Linux, but it can be different between downstream/vendor specific kernels and upstream Linux.

dieneuser, 2021-01-12:
Thank you for the explanations and implementation.

jasonmp85, 2023-04-20:
Hey, this _seems_ to have had some work done for it (updating the `dtb` files, etc.) but my ODROID N2+ isn't responding to the button press. Are there changes needed in the boot script?

jasonmp85, 2023-04-20:
Oops, ignore me! Turns out I didn't realize I needed to hold the button down for two seconds or so.

jasonmp85, 2023-04-20:
Ok, power off works for me, but wake-up does not. What is the expected behavior here for you all?