home-assistant/operating-system #1164: [NUC] fails to update to 5.10

bruxy70, 2021-01-10:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [x] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

## System Health

version | 2021.1.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.84
arch | x86_64
timezone | Europe/Prague

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4314
Installed Version | 1.9.0
Stage | running
Available Repositories | 783
Installed Repositories | 45

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | February 3, 2021, 1:00 AM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.9
-- | --
update_channel | stable
supervisor_version | 2020.12.7
docker_version | 19.03.13
disk_total | 916.2 GB
disk_used | 19.0 GB
healthy | true
supported | true
board | intel-nuc
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.3.0), ESPHome (1.15.3), Grafana (5.3.6), InfluxDB (3.7.9), JupyterLab Lite (0.3.1), Let's Encrypt (4.11.0), Mosquitto broker (5.1), Terminal & SSH (8.10.0), Visual Studio Code (2.9.1), Zigbee2mqtt (1.17.0.1), AppDaemon 4 (0.3.2)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
mode | storage
views | 8
resources | 30

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

**Supervisor logs:**

Does not even get there.

**Journal logs:**

**Kernel logs:**

Does not even get there

**Description of problem:**

Failing to update to 5.10. I connected the NUC via HDMI cable, and these are the logs:
[image]
Then it times out, scrolls through bunch of logs, and stops here:
[image]

When I reboot the device (button), the OS is back to 5.9

I have once succeeded the update, but after a few minutes the device stopped responding (could not even SSH to it). So I rebooted, quickly downgraded back to 5.9 before it stopped responding and it was running fine since then. But I can't replicate it anymore - I have tried, connected the HDMI cable, tried the update, and it failed as described above.

I suppose this is something to do with the disk. This is the NUC configuration.
[image]

agners, 2021-01-11:
Looks like a I/O hick-up on your NVMe SSD?

The only thing I can think of is the kernel update from 5.4.84 -> 5.4.86. A quick skimming it doesn't look like something has changed from nvme in this stable releases...

Maybe just a fluke? I'll update to the latest stable once 5.11 comes out, I think I would just skip 5.10 and see how 5.11 does once it is out.

bruxy70, 2021-01-15:
@agners, I was wondering, the BIOS config is all correct, right? There is no other way to configure it (so it be perhaps not as fast, but more compatible. Initially, I had own Linux install and installed HA in docker. Then there was this situation with this config not being supported, so I initially went for Proxmox with a HA VM, and then I bought this NUC, as it was an officially supported configuration. To be honest, I am disappointed. I am on a supported platform, but I did not have this kind of issues before.
(and the installation process was not really straight forward either).

agners, 2021-01-19:
BIOS setting look good. From what I can see the disk is a NVMe model, so the SATA configuration should not matter.

> To be honest, I am disappointed. I am on a supported platform, but I did not have this kind of issues before.

Sorry about that. In the end, this is not a "off the shelf product", its still an open source project, and we just define some platform which have wider testing. Since the SSD etc. could still be different, things can go wrong.

Did you retry updating/is updating to 5.10 failing consistently? Maybe testing the whole SSD using some utilities from a Ubuntu Live USB drive might be worth, maybe there is a hardware issue.

closed 2022-01-09 by bruxy70