home-assistant/operating-system #1168: Installation on Intel NUC fails multiple times per day

cjramseyer, 2021-01-12:
**Hardware Environment**

- [ ] Intel NUC (or compatible) (NUC6CAYH

**Home Assistant OS release:**
- [X ] Fresh installation of release 2021.1.1
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**
21-01-12 21:31:46 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-01-12 21:31:46 INFO (MainThread) [supervisor.host.info] Updating local host information
21-01-12 21:31:46 INFO (MainThread) [supervisor.host.services] Updating service information
21-01-12 21:31:47 INFO (MainThread) [supervisor.host.network] Updating local network information
21-01-12 21:31:48 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-01-12 21:31:48 INFO (MainThread) [supervisor.host] Host information reload completed
21-01-12 21:31:48 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
21-01-12 21:31:48 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
21-01-12 21:31:48 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-dns with version 2020.12.0
21-01-12 21:31:48 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-01-12 21:31:49 INFO (SyncWorker_1) [supervisor.docker.dns] Starting DNS homeassistant/amd64-hassio-dns with version 2020.12.0 - 172.30.32.3
21-01-12 21:31:49 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-01-12 21:31:49 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-audio with version 17
21-01-12 21:31:49 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
21-01-12 21:31:50 INFO (SyncWorker_0) [supervisor.docker.audio] Starting Audio homeassistant/amd64-hassio-audio with version 17 - 172.30.32.4
21-01-12 21:31:50 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-cli with version 2020.11.1
21-01-12 21:31:50 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
21-01-12 21:31:51 INFO (SyncWorker_1) [supervisor.docker.cli] Starting CLI homeassistant/amd64-hassio-cli with version 2020.11.1 - 172.30.32.5
21-01-12 21:31:51 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-observer with version 2020.10.1
21-01-12 21:31:51 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-multicast with version 3
21-01-12 21:31:51 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
21-01-12 21:31:52 INFO (SyncWorker_0) [supervisor.docker.multicast] Starting Multicast homeassistant/amd64-hassio-multicast with version 3 - Host
21-01-12 21:31:52 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-01-12 21:31:52 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/intel-nuc-homeassistant with version 2021.1.1
21-01-12 21:31:52 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 29 Home Assistant secrets
21-01-12 21:31:52 INFO (MainThread) [supervisor.hassos] Detect HassOS 5.10 / BootSlot B
21-01-12 21:31:53 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
21-01-12 21:31:53 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
21-01-12 21:31:53 INFO (MainThread) [supervisor.store] Loading add-ons from store: 66 all - 66 new - 0 remove
21-01-12 21:31:53 INFO (MainThread) [supervisor.addons] Found 9 installed add-ons
21-01-12 21:31:53 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to esphome/esphome-hassio-amd64 with version 1.15.3
21-01-12 21:31:53 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-ssh with version 8.10.0
21-01-12 21:31:53 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-samba with version 9.3.0
21-01-12 21:31:53 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to hassioaddons/chrony-amd64 with version 1.1.3
21-01-12 21:31:53 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to hassioaddons/appdaemon-amd64 with version 0.3.2
21-01-12 21:31:53 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-mariadb with version 2.2.1
21-01-12 21:31:53 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to hassioaddons/node-red-amd64 with version 7.2.11
21-01-12 21:31:53 INFO (SyncWorker_7) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-configurator with version 5.2.0
21-01-12 21:31:53 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/amd64-addon-duckdns with version 1.12.4
21-01-12 21:31:53 INFO (MainThread) [supervisor.snapshots] Found 1 snapshot files
21-01-12 21:31:53 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
21-01-12 21:31:53 INFO (MainThread) [supervisor.ingress] Loaded 4 ingress sessions
21-01-12 21:31:53 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
21-01-12 21:31:53 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-01-12 21:31:53 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
21-01-12 21:31:54 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-01-12 21:31:54 INFO (MainThread) [__main__] Running Supervisor
21-01-12 21:31:54 INFO (MainThread) [supervisor.hassos] Rauc: B - marked slot kernel.1 as good
21-01-12 21:31:54 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
21-01-12 21:31:54 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' starting 2 add-ons
21-01-12 21:31:55 INFO (SyncWorker_4) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-mariadb with version 2.2.1
21-01-12 21:31:56 INFO (SyncWorker_6) [supervisor.docker.addon] Starting Docker add-on hassioaddons/chrony-amd64 with version 1.1.3
21-01-12 21:31:58 INFO (MainThread) [supervisor.api.security] /host/info access from a0d7b954_chrony
21-01-12 21:32:00 INFO (MainThread) [supervisor.services.modules.mysql] Set core_mariadb as service provider for MySQL
21-01-12 21:32:01 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' starting 3 add-ons
21-01-12 21:32:02 INFO (SyncWorker_3) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-samba with version 9.3.0
21-01-12 21:32:03 INFO (SyncWorker_4) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-duckdns with version 1.12.4
21-01-12 21:32:04 INFO (SyncWorker_5) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-ssh with version 8.10.0
21-01-12 21:32:09 INFO (MainThread) [supervisor.core] Start Home Assistant Core
21-01-12 21:32:09 INFO (SyncWorker_2) [supervisor.docker.interface] Starting homeassistant
[... 101 lines cut ...]
21-01-12 21:33:27 INFO (MainThread) [supervisor.api.security] /host/info access from a0d7b954_appdaemon
21-01-12 21:33:27 INFO (SyncWorker_6) [supervisor.docker.addon] Starting Docker add-on esphome/esphome-hassio-amd64 with version 1.15.3
21-01-12 21:33:28 INFO (MainThread) [supervisor.api.security] /host/info access from a0d7b954_nodered
21-01-12 21:33:29 INFO (MainThread) [supervisor.api.security] /host/info access from a0d7b954_esphome
21-01-12 21:33:32 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled
21-01-12 21:33:32 INFO (MainThread) [supervisor.misc.hwmon] Started Supervisor hardware monitor
21-01-12 21:33:32 INFO (MainThread) [supervisor.core] Supervisor is up and running
21-01-12 21:33:32 INFO (MainThread) [supervisor.host.info] Updating local host information
21-01-12 21:33:32 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-01-12 21:33:32 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-01-12 21:33:32 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-01-12 21:33:33 INFO (MainThread) [supervisor.host.services] Updating service information
21-01-12 21:33:33 INFO (MainThread) [supervisor.host.network] Updating local network information
21-01-12 21:33:34 INFO (MainThread) [supervisor.api.security] /dns/info access from a0d7b954_nodered
[... cut ...]
21-01-12 21:33:45 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
21-01-12 21:34:09 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
21-01-12 21:34:10 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
21-01-12 21:34:10 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
21-01-12 22:03:47 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token

**Description of problem:**

agners, 2021-01-13:
I don't see an issue in this log. The message

```
21-01-12 21:33:32 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
```

Just means the system is checking if there are new updates available, installing an update would look differently.

cjramseyer, 2021-01-13:
So then this is likely some kind of hardware issue?

agners, 2021-01-15:
No, its just an information to you, that the system did check for updates. Nothing to worry about.

agners, 2021-01-15:
The supervisor is checking the file at `https://www.home-assistant.io/version.json` to see if there is a new version available (of any component, OS, Core etc...). As you noted, we do that in regular intervals (~8h). And this message just informs you that this has been done. Nothing to worry about.

closed 2021-01-15 by agners