home-assistant/operating-system #1169: Update od HassOS fails – OVA on NUC, Proxmox

fi-sch, 2021-01-14:
**Hardware Environment**
- [X] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
Currently installed: 5.9
Updating to: 5.10
- Additional information (if accessible):
I experienced similar issue when I was trying to upgrade from version 4.17 to 5.8. I tried multiple times, for a couple of days. Finally I succeeded, but I'm still not sure what happened.

## System Health

version | 2021.1.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.84
arch | x86_64
timezone | Europe/Warsaw

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4910
Installed Version | 1.9.0
Stage | running
Available Repositories | 710
Installed Repositories | 29

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 31

</details>

<details><summary>Airly</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 21 stycznia 2021, 1:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>GIOŚ</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.9
-- | --
update_channel | stable
supervisor_version | 2020.12.7
docker_version | 19.03.13
disk_total | 36.7 GB
disk_used | 22.8 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Home Assistant Google Drive Backup (0.103.0), Glances (0.9.1), Check Home Assistant configuration (3.6.0), File editor (5.2.0), deCONZ (6.6.2), ESPHome (1.15.3), MariaDB (2.2.1), Network UPS Tools (0.3.1), AdGuard Home (2.6.1), Log Viewer (0.9.1), Samba share (9.3.0), Let's Encrypt (4.11.0), WireGuard (0.4.0), Plex Media Server (2.4.2), Nginx Proxy Manager (0.8.2), Bitwarden RS (0.6.2), DOODS (2), SSH & Web Terminal (7.8.0), Grafana (5.3.6), InfluxDB (3.7.9), phpMyAdmin (0.1.4), Visual Studio Code (2.9.1), ZeroTier One (0.9.0), Portainer (1.3.0), Samba Backup (4.3), motionEye (0.10.2), chrony (1.1.3), Git pull (7.12), UniFi Controller (0.19.3)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
mode | storage
views | 9
resources | 23

</details>

**Supervisor logs:**
[image]

**Description of problem:**
After hitting 'update' it starts fetching the update package. Then nothing happens (except loader spinning on 'update' button) and after some minutes i get "502: Bad Gateway" error (only if I stay on System page; screenshot below). Then after a whole hour an error appears in supervisor log.
[image]

I don't have any DNS problems, internet connection is fine and uninterrupted, I am able to ping remote servers (via terminal add-on), speedtests also runs fine from HA integration.

fi-sch, 2021-01-14:
I have just updated Home Assistant Core to 2021.1.2. After that I tried again to update supervisor and for some reason it succeeded this time.

agners, 2021-01-15:
Hm, seems like a one time hick-up then? Probably download got interrupted, could have happened also on server side or anywhere in between Github and you...

fi-sch, 2021-01-15:
Unfortunately it was not one time hiccup. I tried multiple times at different time a of a day.
Also, when I was trying to update from 4.17 to 5.9 I experienced exactly the same issue. I also tried to update for a few days straight at different hours.

Please note that I did not experience any issue like that with updates of HA Core (I do every update when it's available and don't skip releases) or add-ons.

Saubratze, 2021-01-27:
Exactly the same issue for the OS update... had it before as well. After ~10 tries it suddenly worked for 5.9. No idea why.
Same now for 5.10.

Always perfectly after 1 hour. When I download the *.raucb file seperately in my browser it takes <1 minute.

21-01-26 22:58:13 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/5.10/hassos_rpi4-5.10.raucb
21-01-26 23:32:41 INFO (MainThread) [supervisor.snapshots] Found 7 snapshot files
21-01-26 23:32:47 ERROR (MainThread) [supervisor.utils] Can't execute update while a task is in progress
21-01-26 23:55:28 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
21-01-26 23:55:28 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-01-26 23:58:14 WARNING (MainThread) [supervisor.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/5.10/hassos_rpi4-5.10.raucb: 
21-01-26 23:58:14 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-667' coro=<process_lock.<locals>.wrap_api() done, defined at /usr/src/supervisor/supervisor/utils/__init__.py:24> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/utils/__init__.py", line 33, in wrap_api
    return await method(api, *args, **kwargs)
  File "/usr/src/supervisor/supervisor/hassos.py", line 149, in update
    int_ota = await self._download_raucb(version)
  File "/usr/src/supervisor/supervisor/hassos.py", line 97, in _download_raucb
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError

no clue why so far...

fi-sch, 2021-01-27:
@Saubratze What is your setup? Also NUC with Proxmox and HassOS running as a VM?

Saubratze, 2021-01-30:
I'm on the Raspberry PI4 4gb 32bit with home assistant... no VM!
Today again... magically... it worked like last time. Don't know what changed.

agners, 2021-02-05:
@Saubratze hm, that sounds like RPi hass issues to download that particular file for some reason. Is the download successful when doing it in the SSH & Web Terminal add-on? (e.g. using `wget https://github.com/home-assistant/operating-system/releases/download/5.10/hassos_rpi4-5.10.raucb`?)

fi-sch, 2021-02-05:
Update from my side. I've just updated HassOS, Supervisor and HA Core without any problems.

Also today, I realised that I did have an issue with my NUC for a couple of weeks, perhaps. I found out that there's a bug within Intel's network interface driver and for some reason it could hang for a couple of seconds, until Proxmox resurrected it.
Anyways, I'm not sure if that was the case at those previous attempts. This bug occurs quite rarely and with rather bigger transfers (e.g. uploading ~2GB backup file to cloud or network drive). Just saying.
I will will keep an eye on it and for sure let you guys know if this problems occurs again.

Saubratze, 2021-02-05:
@agners I did that just now -> the download worked in 7.2 seconds. However as mentioned the update now also worked a week ago. I'll retry the download on the webterminal if the issue is back with the next Home Assistant OS update!

Saubratze, 2021-02-10:
Updated to OS 5.11 today. No issues at all, worked like a charm!

closed 2022-01-02 by fi-sch