home-assistant/operating-system #1171: [Pi4] Docker issue 500 Server Error

KillahB33, 2021-01-18:
**Hardware Environment**

- [ ] Raspberry Pi 4
 
**Home Assistant OS release:**
- [ ] Fresh installation of release y
- Additional information (if accessible):
## System Health

version | 2021.1.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.79-v8
arch | aarch64
timezone | America/Toronto

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | February 3, 2021, 7:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.10
-- | --
update_channel | stable
supervisor_version | 2021.01.4
docker_version | 19.03.13
disk_total | 28.6 GB
disk_used | 2.6 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.3.0), Mosquitto broker (5.1), Node-RED (7.2.11), Terminal & SSH (8.10.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
mode | auto-gen
resources | 0

</details>

**Supervisor logs:**
21-01-18 11:54:28 INFO (SyncWorker_4) [supervisor.docker.interface] Updating image homeassistant/raspberrypi4-64-homeassistant:2021.1.3 to homeassistant/raspberrypi4-64-homeassistant:2021.1.4
21-01-18 11:54:28 INFO (SyncWorker_4) [supervisor.docker.interface] Downloading docker image homeassistant/raspberrypi4-64-homeassistant with tag 2021.1.4.
21-01-18 11:54:29 ERROR (SyncWorker_4) [supervisor.docker.interface] Can't install homeassistant/raspberrypi4-64-homeassistant:2021.1.4 -> 500 Server Error for http+docker://localhost/v1.40/images/create?tag=2021.1.4&fromImage=homeassistant%2Fraspberrypi4-64-homeassistant: Internal Server Error ("toomanyrequests: You have reached your pull rate limit. You may increase the limit by authenticating and upgrading: https://www.docker.com/increase-rate-limit").
21-01-18 11:54:29 WARNING (MainThread) [supervisor.homeassistant.core] Updating Home Assistant image failed
[... 76 lines cut ...]
[   45.318897] hassio: port 7(veth5e18814) entered disabled state
[   45.319199] device veth5e18814 entered promiscuous mode
[   45.319334] kauditd_printk_skb: 9 callbacks suppressed
[   45.319339] audit: type=1700 audit(1610970661.646:136): dev=veth5e18814 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[   45.890673] audit: type=1325 audit(1610970662.218:137): table=nat family=2 entries=0
[   45.899636] audit: type=1325 audit(1610970662.230:138): table=filter family=2 entries=0
[   45.909272] audit: type=1325 audit(1610970662.238:139): table=nat family=2 entries=5
[   45.912607] audit: type=1325 audit(1610970662.242:140): table=nat family=2 entries=7
[   45.918403] audit: type=1325 audit(1610970662.246:141): table=nat family=2 entries=8
[   45.921875] audit: type=1325 audit(1610970662.250:142): table=nat family=2 entries=10
[   45.927493] audit: type=1325 audit(1610970662.258:143): table=nat family=2 entries=11
[   45.932112] audit: type=1325 audit(1610970662.262:144): table=nat family=2 entries=12
[   45.937719] audit: type=1325 audit(1610970662.266:145): table=nat family=2 entries=13
[   45.972929] eth0: renamed from vetha8d2c68
[   45.988260] IPv6: ADDRCONF(NETDEV_CHANGE): veth5e18814: link becomes ready
[   45.988406] hassio: port 7(veth5e18814) entered blocking state
[   45.988413] hassio: port 7(veth5e18814) entered forwarding state
[   51.941147] udevd[192]: starting version 3.2.9
[   51.968320] udevd[193]: starting eudev-3.2.9

**Description of problem:**
So I have solved this issue in the past but it seems everytime supervisor gets updated it get's broken. This time I am unable to fix it.
In the past I changed the dns and restart the host and then docker would be able to connect.
This time it doesn't appear to fix it. 
I know it's an issue with HassOS/Supervisor because when I login to the host with ssh I can do a docker pull without issue.

frenck, 2021-01-18:
`toomanyrequests: You have reached your pull rate limit. You may increase the limit by authenticating and upgrading: docker.com/increase-rate-limit`

You have done too many requests against DockerHub (where the used Docker images are hosted).

More information: <http://docker.com/increase-rate-limit>

If you try again in a couple of hours, it should work gain.

closed 2021-01-18 by frenck

KillahB33, 2021-01-18:
As mentioned I can do a docker pull from the host machine so that is not the case.

frenck, 2021-01-18:
The messages in your log are not created by us, but is what DockerHub is returning / telling. There is not much we can do if DockerHub doesn't serve you anything.

KillahB33, 2021-01-18:
So can I exec into one of the containers and run docker login so I don't have to deal with the message anymore. This keeps happening and is getting frustrating.

frenck, 2021-01-18:
You can go into the Supervisor panel -> Add-on Store -> three dot menu -> Registries

[image]

There you can set up Docker registries and logins.

KillahB33, 2021-01-18:
Amazing! thank you!