home-assistant/operating-system #1172: Upgrade error from 5.8 > 5.10 - Not able to boot anymore

arjanvroege, 2021-01-19:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
-  Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- Updated from version 5.8

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

After an upgrade from 5.8 to 5.10 the upgrade resulted in a non booting system. After 45 minutes I've powered down the NUC and put a monitor on the system. Now the system cannot boot from the Disk anymore. I'm only seeing the message:
"Reboot and select a proper Boot Device"

I was able to boot from a USB Live Ubuntu and the disk contents were visible but still the system does not boot. Is there anything I can fix so that the system will boot again?

arjanvroege, 2021-01-19:
as suggested in other issues I tried to mount the boot partition but with a Ubuntu LiveCD this only possible as 'read-only' so I cannot copy the boot partition of the image back to this partition.

agners, 2021-01-19:
Hm, it seems the partition got corrupted then? Does `fsck.vfat /dev/sdxx` helps? (replace xx with the disk and partition number).

arjanvroege, 2021-01-19:
Yes, you're right the boot partition was corrupted. I removed the corrupted one, created a new one and copied the files/folders from the image to this new partition. After this the system booted but was not fully loaded. There was some error that the hassos-boot filesystem could not be loaded. I found out that this was caused by the fact that I forgot to add the hassos-boot label to the file system. When I did that I was able to fully load Hass OS (and everything else). 

I managed to upgrade to 5.10 after the system was loaded without any issues. I'm not sure what happened this morning but this issue is solved. :-)

agners, 2021-01-19:
Cool, thanks for the update!

closed 2021-01-19 by agners