home-assistant/operating-system #1176: After update to 5.10 HA does not start

davidjirovec, 2021-01-22:
**Hardware Environment**

- [x] Raspberry Pi 3
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 5.9
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

I've launched OS update from ui, version 5.9 -> 5.10. After this HA web ui did not start. This never happened before. After some googling connecting external screen to raspberry seemed good idea to find what's wrong.

[image]

The startup log usually contains many random failures like these. I've started looking for possibilities to downgrade without need of reinstalling whole HA. I've found https://github.com/home-assistant/operating-system/blob/master/Documentation/configuration.md so took usb flash drive, renamed it to CONFIG and copied .raucb file for older os version there, connected the usb drive to rasbberry and restarted.

It did not help :-| It is sill randomly failing as before. Should the process of downgrading OS via CONFIG usb drive really help me? How will I find out it is working and downgrading the system? Should it be even able to perform the downgrade when there are out of memory errors in random steps on startup?

agners, 2021-01-23:
Downgrading via CONFIG USB drive only works after a successful boot. Do you reach the login prompt at one point?

If so, you can also try to login and downgrade manually using `os update --version 5.9` in the `ha >` shell.

davidjirovec, 2021-01-23:
I never had successful boot since the update. Is it possible to copy the files manually to sd card with HA installation? If, what should I copy and where?

quietlikeninja, 2021-01-24:
I am having, what looks like, the same issue after upgrading from 5.8 to 5.10.  Using Raspberry Pi 3 B+.

One boot made it to the Home Assistant login prompt and I was able to login but it was very unresponsive and froze before I could really do anything.  Subsequent boots are not making it to the login prompt after leaving it for hours.

Is there a way to do a "safe boot" to bypass the services that are failing and get to the logs?  I'm currently looking into spinning up an instance on a VM to see if I can restore my config there and then rebuild the Rpi.  Is that likely to work? My smart house is currently very dumb :-(

quietlikeninja, 2021-01-28:
I bought a new Pi 3 B+ and move the SD card across but have the same issue, just wanted to rule out hardware fault.  Occasionally I get to a login prompt but have mixed results from there.  One time I was able to login and look at a log but then it froze, most times it doesn't even let me login before dying.  Is there any other ways to get the logs and even so would they provide any value or should I just wipe it and start again?
I managed to get a snapshot off the SD card and restore to a fresh Hyper-V instance to get basic functionality back, stupid me hadn't done a snapshot in weeks though.  Since before I implemented NodeRed and moved all my automations there :-(

quietlikeninja, 2021-01-28:
I bought a new Pi 3 B+ and move the SD card across but have the same issue, just wanted to rule out hardware fault.  Occasionally I get to a login prompt but have mixed results from there.  One time I was able to login and loog at a log but then it froze, most times it doesn't even let me login before dying.
I managed to get a snapshot off the SD card and restore to a fresh Hyper-V instance to get basic functionality back, stupid me hadn't done a snapshot in weeks though.  Since before I implemented NodeRed and moved all my automations there :-(

davidjirovec, 2021-01-28:
What you describe is exactly the same issue as I have. I've already reinstalled HA with clean 5.10 and it seems to run without any issues. I did not have time to do any configuration, but I made backup of old HA installation from the sd card before I reinstalled, so I hope just copying stuff back will be enough. Well, since now I'll always do snapshot before upgrading :) I am not ignoring backups, I put my config yamls to git, but having snapshot and just restoring it would be easier I guess.

robertvorthman, 2021-03-01:
I have the same issue on my Pi 3.

augelr, 2021-03-03:
I also have this issue I believe. Anyone have a solution?

robertvorthman, 2021-03-03:
I tried for hours to login using a keyboard and mouse attached to the Pi.  Both times it froze before I could complete login and get to the hassio CLI.

I had to wipe the SD card and start over from scratch.  Luckily I had a recent snapshot so I was able to move some yaml over.  At least I got to upgrade to zwavejs.

veeg, 2021-03-03:
Same issue here...

laander, 2021-04-22:
Experiencing the same issue after upgrading. Now stuck with an unbootable RPi 3B+ and unsure how to proceed. Restore from snapshot I guess.

stefferber, 2021-07-30:
Same problem here. I started with a fresh new core `https://github.com/home-assistant/operating-system/releases/download/6.1/haos_rpi3-64-6.1.img.xz` and restored my snapshot.

closed 2021-11-04 by github-actions[bot]