home-assistant/operating-system #1177: Home Assistant Loses Network Connection on OS 5.10

bschatzow, 2021-01-24:
I have been experiencing a system freeze trying to upgrade to OS 5.10
**Hardware Environment**

- [X ] Raspberry Pi [1/2/3/4] pi4 on SSD 
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
Raspberry pi4 4G on SSD using StarTech 3.1 controller, and Kingston SSD
**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

arch: aarch64
channel: stable
docker: 19.03.11
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.2" this works, 5.10 freezes
homeassistant: 2021.1.5
hostname: homeassistant
logging: info
machine: raspberrypi4-64
operating_system: HassOS 5.2
supervisor: 2021.01.5
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: America/New_York

**Supervisor logs:**

-- Logs begin at Thu 2021-01-07 10:48:31 UTC. --
Jan 24 09:29:54 homeassistant hassos-supervisor[1326]: 21-01-24 09:29:54 INFO (MainThread) [supervisor.host.info] Updating local host information
Jan 24 09:29:54 homeassistant hassos-supervisor[1326]: 21-01-24 09:29:54 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
Jan 24 09:29:54 homeassistant hassos-supervisor[1326]: 21-01-24 09:29:54 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
Jan 24 09:29:54 homeassistant hassos-supervisor[1326]: 21-01-24 09:29:54 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
Jan 24 09:29:55 homeassistant hassos-supervisor[1326]: 21-01-24 09:29:55 INFO (MainThread) [supervisor.host.services] Updating service information
Jan 24 09:29:55 homeassistant hassos-supervisor[1326]: 21-01-24 09:29:55 INFO (MainThread) [supervisor.host.network] Updating local network information
Jan 24 09:29:56 homeassistant hassos-supervisor[1326]: 21-01-24 09:29:56 INFO (MainThread) [supervisor.api.security] /dns/info access from a0d7b954_logviewer
Jan 24 09:29:58 homeassistant hassos-supervisor[1326]: 21-01-24 09:29:58 INFO (MainThread) [supervisor.api.security] /addons/core_mariadb/info access from cebe7a76_hassio_google_drive_backup
Jan 24 09:30:02 homeassistant hassos-supervisor[1326]: 21-01-24 09:30:02 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
Jan 24 09:30:02 homeassistant hassos-supervisor[1326]: 21-01-24 09:30:02 INFO (MainThread) [supervisor.host] Host information reload completed
**Journal logs:**

Last boot core log
Core logs
Jan 23 19:47:55 homeassistant systemd-timesyncd[300]: Network configuration changed, trying to establish connection.
Jan 23 19:47:55 homeassistant systemd-timesyncd[300]: Network configuration changed, trying to establish connection.
Jan 23 19:47:55 homeassistant systemd-resolved[299]: Positive Trust Anchors:
Jan 23 19:47:55 homeassistant systemd-resolved[299]: . IN DS 20326 8 2 e06d44b80b8f1d39a95c0b0d7c65d08458e880409bbc683457104237c7f8ec8d
Jan 23 19:47:55 homeassistant systemd-resolved[299]: Negative trust anchors: 10.in-addr.arpa 16.172.in-addr.arpa 17.172.in-addr.arpa 18.172.in-addr.arpa 19.172.in-addr.arpa 20.172.in-addr.arpa 21.172.in-addr.arpa 22.172.in-addr.arpa 23.172.in-addr.arpa 24.172.in-addr.arpa 25.172.in-addr.arpa 26.172.in-addr.arpa 27.172.in-addr.arpa 28.172.in-addr.arpa 29.172.in-addr.arpa 30.172.in-addr.arpa 31.172.in-addr.arpa 168.192.in-addr.arpa d.f.ip6.arpa corp home internal intranet lan local private test
Jan 23 19:47:55 homeassistant systemd-resolved[299]: Using system hostname 'homeassistant'.
Jan 23 19:47:55 homeassistant dbus-daemon[304]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.16' (uid=0 pid=307 comm="/usr/sbin/NetworkManager --no-daemon ")
Jan 23 19:47:55 homeassistant dbus-daemon[304]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jan 23 19:47:55 homeassistant dbus-daemon[304]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.16' (uid=0 pid=307 comm="/usr/sbin/NetworkManager --no-daemon ")
Jan 23 19:47:55 homeassistant dbus-daemon[304]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Jan 23 19:48:43 homeassistant systemd-timesyncd[300]: Initial synchronization to time server 162.159.200.123:123 (time.cloudflare.com).
Jan 23 19:51:20 homeassistant dbus-daemon[304]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.103' (uid=0 pid=13810 comm="gdbus call --system --dest org.freedesktop.hostnam")
Jan 23 19:51:20 homeassistant dbus-daemon[304]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jan 23 21:58:00 homeassistant dbus-daemon[304]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.199' (uid=0 pid=44769 comm="gdbus call --system --dest org.freedesktop.hostnam")
Jan 23 21:58:00 homeassistant dbus-daemon[304]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jan 24 00:04:43 homeassistant dbus-daemon[304]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.282' (uid=0 pid=67159 comm="gdbus call --system --dest org.freedesktop.hostnam")
Jan 24 00:04:43 homeassistant dbus-daemon[304]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jan 24 02:11:27 homeassistant dbus-daemon[304]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.365' (uid=0 pid=92753 comm="gdbus call --system --dest org.freedesktop.hostnam")
Jan 24 02:11:27 homeassistant dbus-daemon[304]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jan 24 04:18:10 homeassistant dbus-daemon[304]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.448' (uid=0 pid=126131 comm="gdbus call --system --dest org.freedesktop.hostnam")
Jan 24 04:18:11 homeassistant dbus-daemon[304]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jan 24 06:24:54 homeassistant dbus-daemon[304]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.531' (uid=0 pid=147861 comm="gdbus call --system --dest org.freedesktop.hostnam")
Jan 24 06:24:54 homeassistant dbus-daemon[304]: [system] Successfully activated service 'org.freedesktop.hostname1'
Jan 24 07:48:18 homeassistant dbus-daemon[304]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.16' (uid=0 pid=307 comm="/usr/sbin/NetworkManager --no-daemon ")
Jan 24 07:48:18 homeassistant dbus-daemon[304]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Jan 24 08:31:38 homeassistant dbus-daemon[304]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.615' (uid=0 pid=169235 comm="gdbus call --system --dest org.freedesktop.hostnam")
Jan 24 08:31:38 homeassistant dbus-daemon[304]: [system] Successfully activated service 'org.freedesktop.hostname1'
**Kernel logs:**

**Description of problem:**

I do not have this issue with the identical system using SD card.  Only difference is 64 Bit vs 32 bit.
[image]
is my history.  Still working after system freeze.
I tried a different controller and no difference.
My router still shows activity of 
Jan 23 22:35:00 rc_service: service 19964:notify_rc restart_letsencrypt
Jan 23 22:40:00 rc_service: service 21560:notify_rc restart_letsencrypt
Jan 23 22:45:00 rc_service: service 23128:notify_rc restart_letsencrypt
Jan 23 22:50:00 rc_service: service 24690:notify_rc restart_letsencrypt
Jan 23 22:55:00 rc_service: service 26154:notify_rc restart_letsencrypt
Jan 23 23:00:00 rc_service: service 27529:notify_rc restart_letsencrypt
Jan 23 23:05:00 rc_service: service 28885:notify_rc restart_letsencrypt
Jan 23 23:10:00 rc_service: service 30332:notify_rc restart_letsencrypt
Jan 23 23:15:00 rc_service: service 31822:notify_rc restart_letsencrypt
This is every 5 minutes which is the duckdns setting of 300 seconds.
I can not ping the IP or connect to it with any browser.
This is similar to https://github.com/home-assistant/operating-system/issues/1119, but I believe my issue point to a network issue rather than a system freeze.  My sensors show that the system is still writing to the DB and it is still active.

ricardoletosa, 2021-01-25:
Hello,

I am experiencing very similar symptoms, although I do not have yet as much logging as you. What I have found is that ping has only stopped working on ipv4, but continues to work on ipv6. Is this the same for you?

bschatzow, 2021-01-25:
i was told to turn off ipv6.

bschatzow, 2021-01-26:
> Hello,
> 
> I am experiencing very similar symptoms, although I do not have yet as much logging as you. What I have found is that ping has only stopped working on ipv4, but continues to work on ipv6. Is this the same for you?

Are you saying ipv4 fails but ipv6 is still working? I wonder if you can disable ipv4 and just use 6?

agners, 2021-01-26:
In the state where Raspberry Pi is not responding, can you still use the local terminal (using HDMI screen/keyboard directly connected to the RPi). If so, is the system responding? Any messages?

bschatzow, 2021-01-27:
I have not had a keyboard attached. I need to get a keyboard for it as I do not have any wired usb keyboards.  I can do this if you think it would help?  Someone above mentioned that he could ping his via IPV6 but not via IPV4.  I was told to turn off IPV6.  Any thoughts on this?

ricardoletosa, 2021-01-27:
In my case it's completely headless, so waiting for the micro-HDMI adapter to arrive... 

Apologies for the HA-rookie question: the forums indicate that there should be some log entries for the time before the last boot, however from "ha core logs" and supervisor I can only see the logs since the last time I booted the system.

How can I get to see the logs before the last boot (e.g. covering the time at which it crashed) so I can contribute to the resolution?

This issue is happening daily for me.

bschatzow, 2021-01-27:
You need to setup SSH on port 22222.  If you read above you will see the commands to get the logs.  If you need help, let me know.  There are several forums that give details on setting up ssh.

bschatzow, 2021-02-02:
> In the state where Raspberry Pi is not responding, can you still use the local terminal (using HDMI screen/keyboard directly connected to the RPi). If so, is the system responding? Any messages?

@agners 
The system is frozen. No messages. The only activity I see is the network light blinking very rapidly.

ricardoletosa, 2021-02-02:
On my side... I have a separate issue with the HDMI port never showing
anything on my monitor, but I suspect that requires a change to .boot or
ensuring the right power supply.

What I have been able to identify is that the issue happened once more this
morning at 3:25 am. I can see nothing relevant on the core logs or
supervisor logs at that time, also nothing on dmesg.

However it is interesting that simply unplugging the eth port and
reconnecting makes it recover the connection.

My current setup: core-2021.1.5, supervisor-2021.01.7 (stable), Home
Assistant OS 5.10.

bschatzow, 2021-02-02:
@ricardoletosa I had to mod the config.txt file to get it to show on my monitor.  Took awhile to figure out how.  Need ssh on port 22222, then log in via root, and the login again. From there you need to modify your config.txt file and change your settings for your monitor.  Make sure that you check your connections first.  Many had problems that the case was preventing the plug for inserting far enough.

ricardoletosa, 2021-02-07:
Ok, apologies for the delay, but I´ve finally managed to get some logs, but no additional information.

My setup on Raspberry Pi 4 4Gb, connected on ethernet:
```
~ $ ha info
arch: aarch64
channel: stable
docker: 19.03.13
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.10"
homeassistant: 2021.2.1
hostname: homeassistant
logging: info
machine: raspberrypi4-64
operating_system: Home Assistant OS 5.10
supervisor: 2021.01.7
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: Europe/Madrid
```

**My system lost network connection on 5th Feb at 23:52 UTC.**

I could not ping it from my PC:
```
myuser@myPC:/mnt/c/Users/User1$ ping 192.168.10.11
PING 192.168.10.11 (192.168.10.11) 56(84) bytes of data.
From 192.168.10.68 icmp_seq=3 Destination Host Unreachable
From 192.168.10.68 icmp_seq=6 Destination Host Unreachable
^C
--- 192.168.10.11 ping statistics ---
8 packets transmitted, 0 received, +2 errors, 100% packet loss, time 7296ms
```

The system was active on the physical console. I could not ping my router from it (Network unreachable).

I found through nmcli that the system had simply lost its IP4 DHCP assignment.

**nmcli device show (while broken):**
```
GENERAL.DEVICE:                         eth0
GENERAL.TYPE:                           ethernet
GENERAL.HWADDR:                         DC:A6:32:A7:30:A1
GENERAL.MTU:                            1500
GENERAL.STATE:                          100 (connected)
GENERAL.CONNECTION:                     HassOS default
GENERAL.CON-PATH:                       /org/freedesktop/NetworkManager/ActiveConnection/1
WIRED-PROPERTIES.CARRIER:               on
IP4.GATEWAY:                            --
IP4.DNS[1]:                             1.1.1.1
IP4.DNS[2]:                             1.0.0.1
IP6.ADDRESS[1]:                         fe80::d825:ebaa:396e:3cea/64
IP6.GATEWAY:                            --
IP6.ROUTE[1]:                           dst = fe80::/64, nh = ::, mt = 100
IP6.ROUTE[2]:                           dst = ff00::/8, nh = ::, mt = 256, table=255

GENERAL.DEVICE:                         wlan0
GENERAL.TYPE:                           wifi
GENERAL.HWADDR:                         DC:A6:32:A7:30:A2
GENERAL.MTU:                            1500
GENERAL.STATE:                          30 (disconnected)
[... 461 lines cut ...]
Feb 05 23:53:14 homeassistant hassos-supervisor[883]: 21-02-05 23:53:14 INFO (MainThread) [supervisor.resolution.check] System checks complete
Feb 05 23:53:20 homeassistant f39944e1ad77[412]: 2021-02-06 00:53:20 WARNING (MainThread) [homeassistant.helpers.entity] Update of switch.diskstation_home_mode is taking over 10 seconds
Feb 05 23:53:20 homeassistant f39944e1ad77[412]: 2021-02-06 00:53:20 ERROR (MainThread) [homeassistant.helpers.entity] Update for switch.diskstation_home_mode fails
Feb 05 23:53:20 homeassistant f39944e1ad77[412]: Traceback (most recent call last):
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:   File "/usr/local/lib/python3.8/site-packages/urllib3/connectionpool.py", line 445, in _make_request
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:     six.raise_from(e, None)
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:   File "<string>", line 3, in raise_from
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:   File "/usr/local/lib/python3.8/site-packages/urllib3/connectionpool.py", line 440, in _make_request
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:     httplib_response = conn.getresponse()
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:   File "/usr/local/lib/python3.8/http/client.py", line 1347, in getresponse
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:     response.begin()
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:   File "/usr/local/lib/python3.8/http/client.py", line 307, in begin
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:     version, status, reason = self._read_status()
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:   File "/usr/local/lib/python3.8/http/client.py", line 268, in _read_status
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:     line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:   File "/usr/local/lib/python3.8/socket.py", line 669, in readinto
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:     return self._sock.recv_into(b)
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:   File "/usr/local/lib/python3.8/ssl.py", line 1241, in recv_into
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:     return self.read(nbytes, buffer)
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:   File "/usr/local/lib/python3.8/ssl.py", line 1099, in read
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:     return self._sslobj.read(len, buffer)
Feb 05 23:53:20 homeassistant f39944e1ad77[412]: socket.timeout: The read operation timed out
Feb 05 23:53:20 homeassistant f39944e1ad77[412]:
Feb 05 23:53:20 homeassistant f39944e1ad77[412]: During handling of the above exception, another exception occurred:
```

ricardoletosa, 2021-02-07:
I am now going to revert to a manual IP configuration, instead of the DHCP (with a static lease) that I had before. I am expecting this will become a work around the problem, but there seems to be an issue with the DHCP assigned addresses.

Unfortunately I am working with my ISP´s router so I don´t have the chance to confirm whether this issue happened when the DHCP lease expired.

bschatzow, 2021-03-02:
Added new info on https://github.com/home-assistant/operating-system/issues/1256.

closed 2021-03-02 by bschatzow