home-assistant/operating-system #1178: OS 5.10 wlan issues

mvogt1, 2021-01-24:
Hello,

since 5.10 I get occasional wlan errors in dmesg, and the raspi 4 is not reachable anymore:

>GET STA INFO failed, -110

A short internet search suggest that the power management should be turned off, but I haven't found a way how to
do this in a reliable way in home-assistant.

powermanagement is enabled on the host:

~ $ uname -a
Linux core-ssh 5.4.79-v8 #1 SMP PREEMPT Fri Jan 1 16:40:13 UTC 2021 aarch64 Linux
~ $ dmesg | grep power
[    1.215667] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
[    5.824110] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

Many thanks.

agners, 2021-02-17:
Are you using 32-bit or 64-bit version?

It seems that Raspberry Pi OS also has it enabled by default...

> A short internet search suggest that the power management should be turned off

Link? Is that referring to current (5.4) kernel versions?

mvogt1, 2021-02-18:
Stefan Agner <notifications@github.com> schrieb am Mi., 17. Feb. 2021,
17:18:

> Are you using 32-bit or 64-bit version?
>
> It seems that Raspberry Pi OS also has it enabled by default...
>
> A short internet search suggest that the power management should be turned
> off
>
> Link? Is that referring to current (5.4) kernel versions?
>

Its a 64 bit version, the supposed fix to disable PM was reported against
4.18 but the Error message is still  reported against 5.4.
I havent had the issued now for some time, so the report can be closed..

closed 2021-02-18 by agners

bschatzow, 2021-02-18:
@mvogt1 Its a 64 bit version, the supposed fix to disable PM was reported against

 Did you figure out how to do this on the pi4>