home-assistant/operating-system #1179: Zwave device not working after upgrading OS to 5.10

xtensa, 2021-01-24:
**Description of problem:**
After upgrading to 5.10 Z-Wave device (Razberry) stopped working (issue identical to #716). It appeared that despite /boot/config.txt contains entry "dtoverlay=pi3-disable-bt" bluetooth is being enabled and therefore Razberry cannot be initialised. It seems like OS is ignoring this entry and starting BT anyway.

I tried to add "dtoverlay=pi3-miniuart-bt" in /boot/config.txt, but with no luck. Downgrading OS back to 4.20 solved the issue as workaround, but this is not a proper fix of course.

**Hardware Environment**
- HW -  Raspberry Pi [3b]
- Home Assistant OS release - Updated from 4.20 (currently 5.10)

ha > info
arch: armv7
channel: stable
docker: 19.03.13
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.10"
homeassistant: 2021.1.5
hostname: homeassistant
logging: info
machine: raspberrypi3
operating_system: Home Assistant OS 5.10
supervisor: 2021.01.7
supported: true
supported_arch:
- armv7
- armhf
timezone: Europe/Warsaw

**Supervisor logs:**
ha > su logs
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[11:22:04] INFO: Update udev information
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[11:22:05] INFO: Starting local supervisor watchdog...
21-01-24 11:22:08 INFO (MainThread) [__main__] Initializing Supervisor setup
21-01-24 11:22:08 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-01-24 11:22:08 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi3
21-01-24 11:22:08 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/armv7-hassio-supervisor with version 2021.01.7
21-01-24 11:22:08 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
21-01-24 11:22:09 INFO (SyncWorker_0) [supervisor.docker.interface] Cleanup images: ['homeassistant/armv7-hassio-supervisor:2021.01.5']
21-01-24 11:22:10 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-01-24 11:22:10 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-01-24 11:22:10 INFO (MainThread) [__main__] Setting up Supervisor
21-01-24 11:22:10 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-01-24 11:22:11 INFO (MainThread) [supervisor.host.info] Updating local host information
21-01-24 11:22:11 INFO (MainThread) [supervisor.host.services] Updating service information
21-01-24 11:22:12 INFO (MainThread) [supervisor.host.network] Updating local network information
21-01-24 11:22:14 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-01-24 11:22:14 INFO (MainThread) [supervisor.host] Host information reload completed
21-01-24 11:22:14 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
21-01-24 11:22:14 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
21-01-24 11:22:14 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-dns with version 2020.12.0
21-01-24 11:22:14 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-01-24 11:22:15 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-audio with version 17
21-01-24 11:22:15 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-cli with version 2020.11.1
21-01-24 11:22:15 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-observer with version 2020.10.1
21-01-24 11:22:15 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-multicast with version 3
21-01-24 11:22:15 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-01-24 11:22:15 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
21-01-24 11:22:15 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/raspberrypi3-homeassistant with version 2021.1.5
21-01-24 11:22:15 INFO (MainThread) [supervisor.hassos] Detect HassOS 5.10 / BootSlot A
21-01-24 11:22:15 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
21-01-24 11:22:15 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
21-01-24 11:22:16 INFO (MainThread) [supervisor.store] Loading add-ons from store: 63 all - 63 new - 0 remove
21-01-24 11:22:16 INFO (MainThread) [supervisor.addons] Found 4 installed add-ons
21-01-24 11:22:16 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-ssh with version 8.10.0
21-01-24 11:22:16 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to hassioaddons/ssh-armv7 with version 7.8.0
21-01-24 11:22:16 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-check_config with version 3.6.0
21-01-24 11:22:16 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-configurator with version 5.0.0
21-01-24 11:22:16 INFO (MainThread) [supervisor.snapshots] Found 4 snapshot files
21-01-24 11:22:16 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
21-01-24 11:22:16 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
21-01-24 11:22:17 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
21-01-24 11:22:17 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-01-24 11:22:17 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
21-01-24 11:22:17 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-01-24 11:22:17 INFO (MainThread) [__main__] Running Supervisor
21-01-24 11:22:17 INFO (MainThread) [supervisor.hassos] Rauc: A - marked slot kernel.0 as good
21-01-24 11:22:17 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
21-01-24 11:22:17 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' starting 0 add-ons
21-01-24 11:22:17 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' starting 1 add-ons
21-01-24 11:22:19 INFO (SyncWorker_3) [supervisor.docker.addon] Starting Docker add-on hassioaddons/ssh-armv7 with version 7.8.0
21-01-24 11:22:22 INFO (MainThread) [supervisor.api.security] /host/info access from a0d7b954_ssh
21-01-24 11:22:24 INFO (MainThread) [supervisor.core] Start Home Assistant Core
21-01-24 11:22:24 INFO (SyncWorker_2) [supervisor.docker.interface] Starting homeassistant
21-01-24 11:22:24 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
21-01-24 11:28:04 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-01-24 11:28:18 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
21-01-24 11:28:18 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.APPLICATION' starting 1 add-ons
21-01-24 11:28:21 INFO (SyncWorker_3) [supervisor.docker.addon] Starting Docker add-on homeassistant/armv7-addon-configurator with version 5.0.0
21-01-24 11:28:26 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled
21-01-24 11:28:26 INFO (MainThread) [supervisor.misc.hwmon] Started Supervisor hardware monitor
21-01-24 11:28:26 INFO (MainThread) [supervisor.core] Supervisor is up and running
21-01-24 11:28:26 INFO (MainThread) [supervisor.host.info] Updating local host information
21-01-24 11:28:27 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-01-24 11:28:27 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-01-24 11:28:27 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-01-24 11:28:27 INFO (MainThread) [supervisor.host.services] Updating service information
21-01-24 11:28:28 INFO (MainThread) [supervisor.host.network] Updating local network information
21-01-24 11:28:31 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-01-24 11:28:31 INFO (MainThread) [supervisor.host] Host information reload completed
21-01-24 12:22:17 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
21-01-24 12:22:17 INFO (MainThread) [supervisor.resolution.check] System checks complete

agners, 2021-01-26:
Which Z-Wave integration are you using? Do you have logs showing how Z-Wave communication exactly fails?

xtensa, 2021-01-29:
Z-Wave is 1.4. The error log was exactly the same as in #716 with DriverFailed error. I tried upgrading again to 5.10 to reproduce the issue and now miraculausly it is working. So unfortunately I cannot reproduce the issue at the moment. I guess it can be closed then.

agners, 2021-02-05:
Ok, thanks for the update!

closed 2021-02-05 by agners