home-assistant/operating-system #1180: Updating to 5.10 on VirtualBox results in empty EFI directory

iopenguin, 2021-01-24:
**Hardware Environment**
- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [X] OVA Virtualbox 6.1.12 on Ubuntu linux custom hardware

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [X] Updated from version 5.9
- Additional information (if accessible):
## System Health

version | 2021.1.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.84
arch | x86_64
timezone | America/Los_Angeles

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | February 16, 2021, 4:00 PM
relayer_connected | true
remote_enabled | false
remote_connected | false
alexa_enabled | true
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.9
-- | --
update_channel | stable
supervisor_version | 2021.01.7
docker_version | 19.03.13
disk_total | 30.7 GB
disk_used | 16.4 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.2.0), Check Home Assistant configuration (3.6.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
mode | auto-gen
resources | 0

</details>

**Supervisor logs:**
N/A

**Journal logs:**
N/A

**Kernel logs:**
N/A

**Description of problem:**
Had HA working on 114 and Hassos 4.12, updated everything due to security issues. Update to 2012.12.2 first, worked fine. Then update to OS 5.9, worked fine. Then update to 2021.1.5, worked fine. Finally update to OS 5.10, dropped me into an EFI shell. FS0: followed by ls shows EFI with a size of 0. cd EFI followed by ls gives an error that FS0:EFI doesn't exist. Nothing I've tried will boot.

Reverted back to snapshot created before all of the updates and everything worked again. Reapplied first three updates (to OS 5.9, HA 2021.1.5) and everything is working. Took a snapshot and tried again for the OS 5.10 update, same problem. Reverted to the snapshot and continuing to use 5.9 for now.

I can take a snapshot and attempt the upgrade again if there is additional data I can capture. I also have a snapshot saved after the first failed update.

iopenguin, 2021-01-25:
Note that I was able to get it to boot again by following the instructions here (but with 5.10 instead of 5.9): https://github.com/home-assistant/operating-system/issues/1125#issuecomment-750457231

agners, 2021-01-28:
Hm, I was hoping that pull-request #1101 helps with this issue, but that was included in release 5.9. From your description it sounds like the corruption of the boot partition happened with release 5.9 where this fix was included :-(

agners, 2021-01-28:
But yeah it seems really still the same as in #1125, which is still open. Let's track the problem in there.

closed 2021-01-28 by agners