home-assistant/operating-system #1181: Cannot get UART2-UART5 to work on Raspberry Pi 4

henrikssn, 2021-01-25:
I have added the following to `/mnt/boot/config.txt`:

```
[all]
enable_uart=1                                                              
dtoverlay=miniuart-bt                                                      
dtoverlay=uart2                                                            
dtoverlay=uart3                                                            
dtoverlay=uart4                                                            
dtoverlay=uart5
```
but the only UART I see are `/dev/ttyAMA0` (hardware UART0) and `/dev/ttyS0` (mini UART). If I do the same on another RPi4 device running Rasbian then the new UARTs do show up.

Any ideas why I can't get this to work on HassOS (v 5.10)?

henrikssn, 2021-01-25:
Kernel logs:

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.79-v7l (builder@c988a7e976d3) (gcc version 9.3.0 (Buildroot -ge648285f)) #1 SMP Fri Jan 1 16:44:27 UTC 2021
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 504832
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 308224 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49932 r8192 d23796 u81920
[    0.000000] pcpu-alloc: s49932 r8192 d23796 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 502528
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=77a6248307034173908ea3640edf7242 cgroup_enable=memory fsck.repair=yes coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:15:99:00 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootfstype=squashfs ro rauc.slot=B
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x25400000-0x29400000] (64MB)
[    0.000000] Memory: 1844736K/2019328K available (10240K kernel code, 770K rwdata, 3100K rodata, 2048K init, 872K bss, 109056K reserved, 65536K cma-reserved, 1232896K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 34408 entries in 68 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x338/0x51c with crng_init=0
[    0.000007] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000031] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000111] bcm2835: system timer (irq = 17)
[    0.000776] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000796] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000816] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000832] Switching to timer-based delay loop, resolution 18ns
[    0.001069] Console: colour dummy device 80x30
[    0.001599] printk: console [tty1] enabled
[    0.001662] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001705] pid_max: default: 32768 minimum: 301
[    0.001860] LSM: Security Framework initializing
[    0.001979] AppArmor: AppArmor initialized
[    0.002177] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002219] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003261] CPU: Testing write buffer coherency: ok
[    0.003779] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004710] Setting up static identity map for 0x200000 - 0x20003c
[    0.004930] rcu: Hierarchical SRCU implementation.
[    0.005616] smp: Bringing up secondary CPUs ...
[    0.006761] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008061] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.009279] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009433] smp: Brought up 1 node, 4 CPUs
[    0.009504] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.009531] CPU: All CPU(s) started in HYP mode.
[    0.009555] CPU: Virtualization extensions available.
[    0.010354] devtmpfs: initialized
[    0.024415] VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
[    0.024684] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.024734] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.026716] pinctrl core: initialized pinctrl subsystem
[    0.027682] NET: Registered protocol family 16
[    0.031590] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.032185] audit: initializing netlink subsys (disabled)
[    0.032453] audit: type=2000 audit(0.030:1): state=initialized audit_enabled=0 res=1
[... 365 lines cut ...]
[   46.417385] device vethda1c0eb entered promiscuous mode
[... cut ...]
[   47.331131] audit: type=1325 audit(1611533085.389:75): table=nat family=2 entries=11
[   47.339270] audit: type=1325 audit(1611533085.399:76): table=nat family=2 entries=12
[   47.408745] eth1: renamed from vethe86cc70
[   47.457276] IPv6: ADDRCONF(NETDEV_CHANGE): vethda1c0eb: link becomes ready
[   47.457473] hassio: port 2(vethda1c0eb) entered blocking state
[   47.457489] hassio: port 2(vethda1c0eb) entered forwarding state
[   48.322249] udevd[205]: starting version 3.2.9
[   48.346788] udevd[206]: starting eudev-3.2.9
```

agners, 2021-01-25:
Which version of HAOS are you using, 32-bit or 64-bit?

henrikssn, 2021-01-25:
32-bit (`uname -m` returns `armv7l`)

agners, 2021-01-28:
Hm, I cannot reproduce this here:

```
# tail /mnt/boot/config.txt 
[all]
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2

enable_uart=1                                                              
dtoverlay=miniuart-bt                                                      
dtoverlay=uart2                                                            
dtoverlay=uart3                                                            
dtoverlay=uart4                                                            
dtoverlay=uart5
# ls -la /dev/ttyAMA*
crw--w----    1 root     tty       204,  64 Jan 28 12:33 /dev/ttyAMA0
crw-rw----    1 root     dialout   204,  65 Jan 28 12:31 /dev/ttyAMA1
crw-rw----    1 root     dialout   204,  66 Jan 28 12:31 /dev/ttyAMA2
crw-rw----    1 root     dialout   204,  67 Jan 28 12:31 /dev/ttyAMA3
crw-rw----    1 root     dialout   204,  68 Jan 28 12:31 /dev/ttyAMA4
# cat /etc/os-release 
NAME="Home Assistant OS"
VERSION="5.10 (RaspberryPi 4)"
ID=hassos
VERSION_ID=5.10
PRETTY_NAME="Home Assistant OS 5.10"
CPE_NAME=cpe:2.3:o:home_assistant:hassos:5.10:*:production:*:*:*:rpi4:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS RaspberryPi 4"
VARIANT_ID=rpi4
SUPERVISOR_MACHINE=raspberrypi4
SUPERVISOR_ARCH=armv7
```

agners, 2021-01-28:
Maybe related to the Raspberry Pi Firmware? What version are you using?

henrikssn, 2021-01-28:
Here is what I get if I run same commands:

```
# tail /mnt/boot/config.txt 
[all]
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2
enable_uart=1
dtoverlay=miniuart-bt
dtoverlay=uart2
dtoverlay=uart3
dtoverlay=uart4
dtoverlay=uart5

# ls -la /dev/ttyAMA*
crw-rw----    1 root     dialout   204,  64 Jan 28 10:11 /dev/ttyAMA0
# cat /etc/os-release 
NAME="Home Assistant OS"
VERSION="5.10 (RaspberryPi 4)"
ID=hassos
VERSION_ID=5.10
PRETTY_NAME="Home Assistant OS 5.10"
CPE_NAME=cpe:2.3:o:home_assistant:hassos:5.10:*:production:*:*:*:rpi4:*
HOME_URL=https://hass.io/
VARIANT="Home Assistant OS RaspberryPi 4"
VARIANT_ID=rpi4
SUPERVISOR_MACHINE=raspberrypi4
SUPERVISOR_ARCH=armv7
```

agners, 2021-01-29:
Can you try with an updated firmware (e.g. by using a Raspberry Pi OS SD card and following the "Updating from Raspberry Pi OS" section in https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md).

closed 2022-01-11 by github-actions[bot]