home-assistant/operating-system #1183: Odroid XU4 cannot finish install of HassOS 5.10, lacks space on 64GB eMMC

ClearlyDazed, 2021-01-25:
I have flashed HassOS image 5.10 onto my Odroid XU4 64GB eMMC (orange v.0.4). It boots up and starts the install from the eMMC but can not complete it due to limited space being reported. In the initial stage it tries to run an app (maybe cleanup) but can not due to limited space issue it also reports that GitRepo is blocked from execution because of not enough free space. The install proceeds to download and install the docker image but can not complete the install of homeassistant/odroid-xu-homeassistant:2021.1.5 because of no space; this docker image download and install is repeated continuously.

It seems the other people have been able to install and run HassOS on Odroid XU4 that utilize 32GB eMMC.  How can this be done for the 64GB eMMC?
I have seen installed Armbian debian image and Hass Supervised on this Odroid eMMC just to see if it works. I would like to however run HassOS. 

If you require logs I could attempt reflashing.

agners, 2021-01-25:
Hm, do you have an SD card or USB flash drive attached? HAOS uses partition label to resize the correct partition on first boot, so ideally on first boot all other storage devices should be disconnected to avoid interference (it only is problematic if the SD card or USB flash drive has been flashed with HAOS as well, but just to be sure removing should avoid any interference).

ClearlyDazed, 2021-01-25:
I managed to get home assistant HassOS running on my Odroid XU4 with a 64GB eMMC. First setup the Odroid to boot from the eMMC as per [https://github.com/home-assistant/operating-system/blob/dev/Documentation/boards/hardkernel/odroid-xu4.md](url). Had to flash both eMMC and SD and use echo etc.. Booted odroid on eMMC and waited for HomeAssistant CLI to come up to check that it works. Shutdown the Odroid. Attach a second storage device, hard drive or USB and start up the Odroid. When HA CLI comes up, login to the linux prompt.  Follow the instructions on [https://github.com/home-assistant/operating-system/blob/dev/Documentation/partition.md](url) to move the data partition to the storage device. Reboot and wait for everything to load. Access HA from the web interface.

ClearlyDazed, 2021-01-30:
> Hm, do you have an SD card or USB flash drive attached? HAOS uses partition label to resize the correct partition on first boot, so ideally on first boot all other storage devices should be [...]

ClearlyDazed, 2021-01-30:
Hi agners. My initial install did not have any other storage device attached. I got the same out of space error as reported on issue #1193

syserror2021, 2021-02-02:
Hi ClearlyDazed,

I think the issue here and #1193 might be related. See my comment over there.

In the end I ended up installing hassos_odroid-xu4-4.20.img, no issues with that version.
After that I upgraded to 2021.1.5 and moved my data to the attached hard drive using datactl as described over here:
(https://community.home-assistant.io/t/hass-io-transfer-from-sd-card-to-ssd-or-usb/97452/269)

agners, 2021-02-05:
Yes, this indeed seems to be the same issue as #1193. It has been addressed with pull request #1204, which will also get backported to rel-5. So this will be fixed 5.12 onwards.

closed 2021-02-05 by agners

ClearlyDazed, 2021-02-06:
Thank you @syserror2021. The install method you suggested worked and I was also able to update HassOS to the latest. The Odroid is running HA entirely on the eMMC. Thanks @angers for keeping the HassOS image up-todate.