home-assistant/operating-system #1191: Cannot connect to host updater.home-assistant.io:443

vannetta, 2021-01-28:
**Hardware Environment**

- [ x] Raspberry Pi [3B+/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [x ] Fresh installation of release 5.10
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem: I had my SD burned, so I write the image of the OS for a new installation and then restore the snapshot.
HA starts, but the HassOS cannot update. Tha date is that one of the release (16Dec2020).
If I change the host's date, the error desappear, but if I shutdown the host, it comes back to Dec 2020.
Moreover if I reboot or update, it doesn't.**

This is the log:
`Logger: homeassistant.components.updater
Source: helpers/update_coordinator.py:153
Integration: Updater (documentation, issues)
First occurred: December 16, 2020, 6:26:38 PM (1 occurrences)
Last logged: December 16, 2020, 6:26:38 PM
Error requesting Home Assistant update data: Cannot connect to host updater.home-assistant.io:443 ssl:True [SSLCertVerificationError: (1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1125)')]`

This is the info of the system:
`System Health

version: 2021.1.5
installation_type: Home Assistant OS
dev: false
hassio: true
docker: true
virtualenv: false
python_version: 3.8.7
os_name: Linux
os_version: 5.4.79-v7l
arch: armv7l
timezone: Europe/Rome

logged_in: false
can_reach_cert_server: ok
can_reach_cloud_auth: ok
**can_reach_cloud: failed to load: unreachable**

host_os: Home Assistant OS 5.10
update_channel: stable
supervisor_version: 2021.01.7
docker_version: 19.03.13
disk_total: 57.8 GB
disk_used: 5.7 GB
healthy: true
supported: true
board: rpi4
supervisor_api: ok
version_api: ok
installed_addons: Duck DNS (1.12.4), Samba share (9.3.0), Terminal & SSH (8.10.0), NGINX Home Assistant SSL proxy (3.0.1), File editor (5.2.0), Check Home Assistant configuration (3.6.0), AppDaemon 4 (0.3.1), ESPHome (1.15.3), Log Viewer (0.9.0), Mosquitto broker (5.1), Network UPS Tools (0.3.1), Zigbee2mqtt (1.16.1), zigbee2mqtt-edge (test), RTL_433 to MQTT Bridge (0.2)

dashboards: 1
mode: storage
views: 13
resources: 2`

agners, 2021-01-29:
Could it be that the system time was wrong at that point? Typically the system time is synchronized using NTP. Does the problem still persist after a while/a reboot?

vannetta, 2021-01-29:
The main problem seems to be that the host stays at the time of the release (16 Dec 2020);  I can force it through the port 22222 to the right date and everything works, but if I shutdown the system or reboot it cames back to Dec 16th.
This afternoon without rebooting the host comes back again to Dec 16th.

agners, 2021-02-05:
The Raspberry Pi does not have a RTC, so as soon as power goes out, the time is lost. Systemd sets the time to the date when it has been compiled (which seems to be Dec 16th).

> This afternoon without rebooting the host comes back again to Dec 16th.

That indeed is very strange. Are you sure the system did not reboot? E.g. due to a short power failure or something like that?

vannetta, 2021-02-05:
I can understand what you said, but:
- it is the first time that it happens, and I've had it for one year.
- it happens also if I reboot the system or upgrade the OS, without shutdown.
- the Rpi is under UPS
- I've tried a new installation, with another Sd, another Rpi, without restoring snapshot.  Until I don't force the date through the port 22222, it is not able to synchronize the NTC date, and the error is: 
`Logger: homeassistant.components.updater Source: helpers/update_coordinator.py:153 Integration: Updater (documentation, issues) First occurred: December 16, 2020, 6:26:38 PM (1 occurrences) Last logged: December 16, 2020, 6:26:38 PM Error requesting Home Assistant update data: Cannot connect to host updater.home-assistant.io:443 ssl:True [SSLCertVerificationError: (1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1125)')]`

agners, 2021-02-05:
The error is expected when time is off: Your time is too far behind and the SSL certificate `updater.home-assistant.io` is presenting is from the future.

The Raspberry Pi has no RTC. So after each reboot you have to re-synchronize the clock.

This normally is being done by systemd-timesyncd at boot time using NTP. I guess the question is why it is not doing that in your case (or why it is no longer doing that with newer version?). You can check `journalctl -u systemd-timesyncd` to get the log of systemd-timesyncd.

Do you maybe have a firewall active which might prevent NTP traffic?

vannetta, 2021-02-05:
This is journalctl of the system with tha date upgraded via SSH
```
Feb 05 15:30:18 homeassistant systemd-timesyncd[288]: Timed out waiting for repl                                                                                                                               y from 162.159.200.1:123 (time.cloudflare.com).
Feb 05 15:30:28 homeassistant systemd-timesyncd[288]: Timed out waiting for repl                                                                                                                               y from 162.159.200.123:123 (time.cloudflare.com).
```
This is the log with another Rpi, with a new installation of HassOS 5.10, without changing the date:
**` #date
Wed Dec 16 17:30:29 UTC 2020**

#journalctl -u systemd-timesyncd
-- Logs begin at Wed 2020-12-16 17:25:16 UTC, end at Fri 2021-01-29 09:03:09 UTC. --
Dec 16 17:25:19 homeassistant systemd[1]: Starting Network Time Synchronization...
Dec 16 17:25:19 homeassistant systemd[1]: Started Network Time Synchronization.
Dec 16 17:25:19 homeassistant systemd-timesyncd[271]: Network configuration changed, trying to establish connection.
Dec 16 17:25:19 homeassistant systemd-timesyncd[271]: Network configuration changed, trying to establish connection.
-- Reboot --
Dec 16 17:25:21 homeassistant systemd[1]: Starting Network Time Synchronization...
Dec 16 17:25:21 homeassistant systemd[1]: Started Network Time Synchronization.
Dec 16 17:25:21 homeassistant systemd-timesyncd[291]: Network configuration changed, trying to establish connection.
Dec 16 17:25:21 homeassistant systemd-timesyncd[291]: Network configuration changed, trying to establish connection.
-- Reboot --
Dec 16 17:26:00 homeassistant systemd-timesyncd[271]: Timed out waiting for reply from 162.159.200.1:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:26:01 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:26:10 homeassistant systemd-timesyncd[271]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:26:12 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.1:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:26:53 homeassistant systemd-timesyncd[271]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:26:54 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.1:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:27:03 homeassistant systemd-timesyncd[271]: Timed out waiting for reply from 162.159.200.1:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:27:04 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:28:18 homeassistant systemd-timesyncd[271]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:28:19 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.1:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:28:28 homeassistant systemd-timesyncd[271]: Timed out waiting for reply from 162.159.200.1:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:28:29 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:30:46 homeassistant systemd-timesyncd[271]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
-- Reboot --
Dec 16 17:30:48 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.1:123 (time.cloudflare.com).
Dec 16 17:30:58 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
Dec 16 17:35:24 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.1:123 (time.cloudflare.com).
Dec 16 17:35:35 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
Dec 16 17:44:17 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
Dec 16 17:44:27 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.1:123 (time.cloudflare.com).
Dec 16 18:01:42 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
Dec 16 18:01:52 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.1:123 (time.cloudflare.com).
Jan 29 08:09:12 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.1:123 (time.cloudflare.com).
Jan 29 08:09:23 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
Jan 29 08:43:41 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.123:123 (time.cloudflare.com).
Jan 29 08:43:51 homeassistant systemd-timesyncd[291]: Timed out waiting for reply from 162.159.200.1:123 (time.cloudflare.com).
Jan 29 09:03:09 homeassistant systemd[1]: Stopping Network Time Synchronization...
Jan 29 09:03:09 homeassistant systemd[1]: systemd-timesyncd.service: Succeeded.
Jan 29 09:03:09 homeassistant systemd[1]: Stopped Network Time Synchronization.
`
The firewall should be in the modem? I have a Fritzbox 7530.

agners, 2021-02-05:
Yes, I guess your router Fritzbox 7530 acts as a firewall. Normally those services should be open. It could also be that your provider blocks NTP, but that would be rather unusual.

vannetta, 2021-02-05:
I put the same server that is in the Fritzbox page in Win10,  and it synchronizes; I thought that shouldn't be blocked, isn' it?

vannetta, 2021-02-06:
SOLVED!
I followed this instruction:
https://github.com/home-assistant/operating-system/blob/dev/Documentation/configuration.md
I created the timesyncd.conf file in the USB, named CONFIG:

```
[Time]
NTP=2.europe.pool.ntp.org
FallbackNTP=2.europe.pool.ntp.org
```

I shutdown the host e switched on again (because reboot crashed the installation, and I need to flash the SD again).
When it switches on again the date was ok, no errors in log.
Thanks so much @agners for your inspiration!
I'm trying now to reboot the system (hoping everything ok...)

closed 2021-02-08 by agners