home-assistant/operating-system #1193: Error while fresh install: No such image: homeassistant/odroid-xu-homeassistant:2021.1.5

algope, 2021-01-29:
**Hardware Environment**

- [x] ODROID [C2/C4/N2(+)/XU4]

**Home Assistant OS release:**
- [x] Fresh installation of release x.y

**Supervisor logs:**

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[22:22:51] INFO: Update udev information
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[22:22:53] INFO: Starting local supervisor watchdog...
21-01-29 22:22:56 INFO (MainThread) [__main__] Initializing Supervisor setup
21-01-29 22:22:56 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-01-29 22:22:56 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: odroid-xu
21-01-29 22:22:56 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/armv7-hassio-supervisor with version 2021.01.7
21-01-29 22:22:56 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
21-01-29 22:22:57 INFO (SyncWorker_0) [supervisor.docker.interface] Cleanup images: ['homeassistant/armv7-hassio-supervisor:2020.12.7']
21-01-29 22:23:00 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-01-29 22:23:00 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-01-29 22:23:00 INFO (MainThread) [__main__] Setting up Supervisor
21-01-29 22:23:00 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-01-29 22:23:02 INFO (MainThread) [supervisor.host.info] Updating local host information
21-01-29 22:23:02 INFO (MainThread) [supervisor.host.services] Updating service information
21-01-29 22:23:03 INFO (MainThread) [supervisor.host.network] Updating local network information
21-01-29 22:23:08 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-01-29 22:23:08 INFO (MainThread) [supervisor.host] Host information reload completed
21-01-29 22:23:08 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
21-01-29 22:23:08 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
21-01-29 22:23:08 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-dns with version 2020.12.0
21-01-29 22:23:09 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-01-29 22:23:09 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-audio with version 17
21-01-29 22:23:09 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-cli with version 2020.11.1
21-01-29 22:23:09 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-observer with version 2020.10.1
21-01-29 22:23:09 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-multicast with version 3
21-01-29 22:23:09 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-01-29 22:23:09 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 0 Home Assistant secrets
21-01-29 22:23:09 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/odroid-xu-homeassistant with version landingpage
21-01-29 22:23:09 INFO (MainThread) [supervisor.homeassistant.core] Starting HomeAssistant landingpage
21-01-29 22:23:09 INFO (MainThread) [supervisor.hassos] Detect HassOS 5.10 / BootSlot A
21-01-29 22:23:09 WARNING (MainThread) [supervisor.jobs] 'GitRepo.clone' blocked from execution, not enough free space (0.2GB) left on the device
21-01-29 22:23:09 WARNING (MainThread) [supervisor.store] Skip update to later for a0d7b954
21-01-29 22:23:09 WARNING (MainThread) [supervisor.jobs] 'GitRepo.clone' blocked from execution, not enough free space (0.2GB) left on the device
21-01-29 22:23:09 WARNING (MainThread) [supervisor.store] Skip update to later for core
21-01-29 22:23:09 INFO (MainThread) [supervisor.store] Loading add-ons from store: 0 all - 0 new - 0 remove
21-01-29 22:23:09 INFO (MainThread) [supervisor.addons] Found 0 installed add-ons
21-01-29 22:23:09 INFO (MainThread) [supervisor.snapshots] Found 0 snapshot files
21-01-29 22:23:09 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
21-01-29 22:23:09 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
21-01-29 22:23:09 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
21-01-29 22:23:09 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-01-29 22:23:09 ERROR (MainThread) [supervisor.homeassistant.api] Can't update Home Assistant access token!
21-01-29 22:23:09 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
21-01-29 22:23:09 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-01-29 22:23:09 INFO (MainThread) [__main__] Running Supervisor
21-01-29 22:23:09 INFO (MainThread) [supervisor.hassos] Rauc: A - marked slot kernel.0 as good
21-01-29 22:23:09 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
21-01-29 22:23:09 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' starting 0 add-ons
21-01-29 22:23:09 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' starting 0 add-ons
21-01-29 22:23:09 INFO (MainThread) [supervisor.core] Skiping start of Home Assistant
21-01-29 22:23:09 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.APPLICATION' starting 0 add-ons
21-01-29 22:23:09 INFO (MainThread) [supervisor.misc.tasks] All core tasks are scheduled
21-01-29 22:23:09 INFO (MainThread) [supervisor.misc.hwmon] Started Supervisor hardware monitor
21-01-29 22:23:09 INFO (MainThread) [supervisor.core] Supervisor is up and running
21-01-29 22:23:09 INFO (MainThread) [supervisor.homeassistant.core] Home Assistant setup
21-01-29 22:23:09 INFO (MainThread) [supervisor.host.info] Updating local host information
21-01-29 22:23:09 INFO (SyncWorker_0) [supervisor.docker.interface] Updating image homeassistant/odroid-xu-homeassistant:landingpage to homeassistant/odroid-xu-homeassistant:2021.1.5
21-01-29 22:23:09 INFO (SyncWorker_0) [supervisor.docker.interface] Downloading docker image homeassistant/odroid-xu-homeassistant with tag 2021.1.5.
21-01-29 22:23:10 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-01-29 22:23:10 WARNING (MainThread) [supervisor.jobs] 'FixupStoreExecuteReload.process_fixup' blocked from execution, not enough free space (0.2GB) left on the device
21-01-29 22:23:10 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-01-29 22:23:10 INFO (MainThread) [supervisor.host.services] Updating service information
21-01-29 22:23:10 INFO (MainThread) [supervisor.host.network] Updating local network information
21-01-29 22:23:14 INFO (MainThread) [supervisor.host] Host information reload completed
21-01-29 22:23:50 ERROR (SyncWorker_0) [supervisor.docker.interface] Can't install homeassistant/odroid-xu-homeassistant:2021.1.5 -> 404 Client Error for http+docker://localhost/v1.40/images/homeassistant/odroid-xu-homeassistant:2021.1.5/json: Not Found ("no such image: homeassistant/odroid-xu-homeassistant:2021.1.5: No such image: homeassistant/odroid-xu-homeassistant:2021.1.5").
21-01-29 22:23:50 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec

```

**Description of problem:**
Trying a fresh install of Hassio in an Odroid XU4. After some minutes, the above error occurs. I've tried with different SD cards and aparently it seems an inssue with the latest release 2021.1.5.

agners, 2021-01-30:
It seems that there was no space on the data partition:

```
21-01-29 22:23:10 WARNING (MainThread) [supervisor.jobs] 'FixupStoreExecuteReload.process_fixup' blocked from execution, not enough free space (0.2GB) left on the device
```

Which version of the OS did you install? The OS does a resize on first boot. If that gets interrupted, then the resize did not happen anymore. A new install and an uninterrupted boot should fix the problem.

algope, 2021-01-31:
@agners I tried a couple of times with different sd cards. I gave up and tried with a Raspberry Pi and it worked.
So, it could be something related to my Odroid or the Odroid OS version of Hassio... I used the latest available at the HA site.

syserror2021, 2021-02-02:
I have experienced similar issues trying to install hassos_odroid-xu4-5.10.img on an Odroid HC1.

Here are my observations:
1.) After flashing the SD card, the HC1 boots up
2.) http://homeassistant:8123 shows a splash screen, asking you to be patient
3.) Even if I leave the device at this stage on its own over night, this splash screen never disappears
4.) During the whole time I see significant download activity on my internet link, as if Home Assistant tries to download some packages in an endless loop
5.) After login into the device using the serial console I saw error messages indicated that there is insufficient space on the SD Card, even though I used a 64 GB card. (Unfortunately I did not copy the logs)
6.) df -h showed clearly that the partitions have not been resized properly

My work around:
1.) Install hassos_odroid-xu4-4.20.img on the same device, otherwise unchanged
2.) System comes up after installation, no error messages, SD card has been resized properly
3.) Upgrade to latest version of assistant

agners, 2021-02-03:
Hm, that sounds like we have a problem with resizing on a new installation on this platform. Will take a look at this.

agners, 2021-02-05:
This has been addressed with pull request #1204, which will also get backported to rel-5. So this will be fixed 5.12 onwards.

closed 2021-02-05 by agners

lancelon, 2021-02-08:
[I had this on a Pi](https://github.com/home-assistant/supervisor/issues/2523), caused me to lost an entire afternoon and evening. Think it's the same issue despite the difference in platform here.