home-assistant/operating-system #1194: When hassio is online other devices on network lose wifi connection

Mobrockers, 2021-01-30:
**Hardware Environment**

Hassio raspberrypi image
raspberrypi 4
Connected with wifi

**Home Assistant OS release:**
Updated from version 2020.12.1

- Additional information (if accessible):

```
~ $ ha info
arch: armv7
channel: stable
docker: 19.03.13
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.10"
homeassistant: 2021.1.5
hostname: hassio
logging: info
machine: raspberrypi4
operating_system: Home Assistant OS 5.10
supervisor: 2021.01.7
supported: true
supported_arch:
- armv7
- armhf
timezone: Europe/Amsterdam
```

**Supervisor logs:**

```
~ $ ha su logs
[11:35:27] INFO: Starting local supervisor watchdog...
21-01-30 11:35:31 INFO (MainThread) [__main__] Initializing Supervisor setup
21-01-30 11:35:31 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-01-30 11:35:31 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4
21-01-30 11:35:31 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/armv7-hassio-supervisor with version 2021.01.7
21-01-30 11:35:31 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-01-30 11:35:31 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-01-30 11:35:31 INFO (MainThread) [__main__] Setting up Supervisor
21-01-30 11:35:32 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-01-30 11:35:32 INFO (MainThread) [supervisor.host.info] Updating local host information
21-01-30 11:35:32 INFO (MainThread) [supervisor.host.services] Updating service information
21-01-30 11:35:32 INFO (MainThread) [supervisor.host.network] Updating local network information
21-01-30 11:43:14 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-01-30 11:43:14 INFO (MainThread) [supervisor.host] Host information reload completed
21-01-30 11:43:14 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor', 'core_dnsmasq'}
21-01-30 11:43:14 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
21-01-30 11:43:14 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-dns with version 2020.12.0
21-01-30 11:43:14 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-01-30 11:43:14 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_dns application
21-01-30 11:43:17 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS homeassistant/armv7-hassio-dns with version 2020.12.0 - 172.30.32.3
21-01-30 11:43:17 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-01-30 11:43:17 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-audio with version 17
21-01-30 11:43:17 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
21-01-30 11:43:17 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_audio application
21-01-30 11:43:19 INFO (SyncWorker_0) [supervisor.docker.audio] Starting Audio homeassistant/armv7-hassio-audio with version 17 - 172.30.32.4
21-01-30 11:43:19 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-cli with version 2020.11.1
21-01-30 11:43:19 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
21-01-30 11:43:19 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_cli application
21-01-30 11:43:21 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI homeassistant/armv7-hassio-cli with version 2020.11.1 - 172.30.32.5
[... 57 lines cut ...]
21-01-30 11:44:30 INFO (MainThread) [supervisor.host.services] Updating service information
21-01-30 11:44:30 INFO (MainThread) [supervisor.host.network] Updating local network information
21-01-30 11:44:33 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-01-30 11:44:33 INFO (MainThread) [supervisor.host] Host information reload completed
21-01-30 11:44:58 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
21-01-30 11:44:58 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
21-01-30 11:44:58 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
21-01-30 11:48:45 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning addon_core_ssh application
21-01-30 11:48:47 INFO (SyncWorker_3) [supervisor.docker.addon] Starting Docker add-on homeassistant/armv7-addon-ssh with version 8.10.0
21-01-30 11:51:33 INFO (MainThread) [supervisor.snapshots] Found 3 snapshot files
21-01-30 11:55:13 INFO (MainThread) [supervisor.api.security] /supervisor/logs access from core_ssh
```
**Journal logs:**

I enabled SSH using the SSH addon but could not execute this command

**Kernel logs:**

I enabled SSH using the SSH addon but could not execute this command

**Description of problem:**

Since the update, when hassio is online my devices (phones, laptops, google homes, thermostats, vaccuums etc) on wifi start losing connection every couple of minutes until I turn off hassio. I have stopped all addons that I think might cause this issue (dnsmasq, nginx proxy, adguard) but the problem persists suggesting it is a problem with hassio itself.

After some time hassio itself will crash, then devices come back on wifi as well. SSH is still available, suggesting some part of hassio is still online. WebUI will be unavailable.

agners, 2021-01-30:
Hm, interesting. Do you seen any specific network traffic on the network (e.g. using Wireshark)? Is the host log reporting issues?

Mobrockers, 2021-01-30:
The only errors I see in the logs are because hass cannot reach wifi devices like nest mini (because they were kicked off wifi).

I have seen this kind of behavior before some months ago. My ISP said that router logs showed IP-lease errors on a number of devices. They advised that I restart all affected devices a couple times and/or disconnect and reconnect their wifi connection. I did so multiple times. After some days the disconnect issue went away. I might have done a hass update before the issue went away, but I am not sure since it was long ago. I asked my ISP again yesterday if they saw IP-lease issues, they said they did not. Unfortunately I cannot check the router logs myself as my ISP-issued router does not allow me access to the logs.

I have not tried to use wireshark to inspect the network, but also the only device from which I can do this is connected by ethernet so I'm not sure if I could see anything on the wifi devices anyway. You think it would show up?

Mobrockers, 2021-01-30:
Here's all logs from the webui. Please let me know if there's any other logs I should collect:

**Supervisor**
```
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[22:51:28] INFO: Starting local supervisor watchdog...
21-01-30 22:51:31 INFO (MainThread) [__main__] Initializing Supervisor setup
21-01-30 22:51:31 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-01-30 22:51:31 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4
21-01-30 22:51:31 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/armv7-hassio-supervisor with version 2021.01.7
21-01-30 22:51:31 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-01-30 22:51:32 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-01-30 22:51:32 INFO (MainThread) [__main__] Setting up Supervisor
21-01-30 22:51:32 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-01-30 23:22:46 INFO (MainThread) [supervisor.host.info] Updating local host information
21-01-30 23:22:46 INFO (MainThread) [supervisor.host.services] Updating service information
21-01-30 23:22:46 INFO (MainThread) [supervisor.host.network] Updating local network information
21-01-30 23:22:48 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-01-30 23:22:48 INFO (MainThread) [supervisor.host] Host information reload completed
21-01-30 23:22:48 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor', 'core_dnsmasq'}
21-01-30 23:22:48 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
21-01-30 23:22:48 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-dns with version 2020.12.0
21-01-30 23:22:48 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-01-30 23:22:48 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_dns application
21-01-30 23:22:50 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS homeassistant/armv7-hassio-dns with version 2020.12.0 - 172.30.32.3
21-01-30 23:22:50 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-01-30 23:22:50 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-audio with version 17
21-01-30 23:22:50 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
21-01-30 23:22:50 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_audio application
21-01-30 23:22:53 INFO (SyncWorker_0) [supervisor.docker.audio] Starting Audio homeassistant/armv7-hassio-audio with version 17 - 172.30.32.4
21-01-30 23:22:53 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-cli with version 2020.11.1
21-01-30 23:22:53 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
21-01-30 23:22:53 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_cli application
21-01-30 23:22:55 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI homeassistant/armv7-hassio-cli with version 2020.11.1 - 172.30.32.5
21-01-30 23:22:55 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-observer with version 2020.10.1
21-01-30 23:22:55 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-multicast with version 3
21-01-30 23:22:55 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
21-01-30 23:22:55 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_multicast application
21-01-30 23:22:57 INFO (SyncWorker_0) [supervisor.docker.multicast] Starting Multicast homeassistant/armv7-hassio-multicast with version 3 - Host
21-01-30 23:22:57 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-01-30 23:22:57 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
21-01-30 23:22:57 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/raspberrypi4-homeassistant with version 2021.1.5
21-01-30 23:22:58 INFO (MainThread) [supervisor.hassos] Detect HassOS 5.10 / BootSlot A
21-01-30 23:22:58 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
21-01-30 23:22:58 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
21-01-30 23:23:01 INFO (MainThread) [supervisor.store] Loading add-ons from store: 65 all - 65 new - 0 remove
21-01-30 23:23:01 INFO (MainThread) [supervisor.addons] Found 12 installed add-ons
21-01-30 23:23:01 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-ssh with version 8.10.0
21-01-30 23:23:01 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-check_config with version 3.6.0
21-01-30 23:23:01 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/adguard/armv7 with version 3.0.0
21-01-30 23:23:01 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/node-red/armv7 with version 8.0.1
21-01-30 23:23:01 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-dnsmasq with version 1.3
[... cut ...]
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] interface.sh: executing... 
[cont-init.d] interface.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
mdns-repeater (6): dev wlan0 addr 192.168.1.104 mask 255.255.255.0 net 192.168.1.0
mdns-repeater (6): dev hassio addr 172.30.32.1 mask 255.255.254.0 net 172.30.32.0
```

Mobrockers, 2021-01-31:
@agners I just did a completely fresh install and _during first boot_ my devices started losing wifi connection... Even before hass was fully done with the initial install.. The only thing I did was set up wifi with DHCP using the my-network file on the SD card.

closed 2022-01-11 by github-actions[bot]