home-assistant/operating-system #1198: Cant add the vhdx at hyperv

Luekuu, 2021-01-31:
Hello dear HS team

I can not add the CHDX to the VM. 

Error:
[https://i.imgur.com/ke2Uw3P.png](url)

Does anyone have a tip for me? 

Thanks and  regards
Lukas

agners, 2021-02-02:
Can you add more information? What kind of Hypervisor are you using? Version? Operating System and Operating System version?

TheAndyMac, 2021-02-03:
Same issue - using Microsoft Hyper-V on Windows 10 Pro x64 (20H2). Hyper-V working fine. VHDX image seems to be corrupted - can't open in Windows, can't Edit in Hyper-V manager and setting up new VM using Hyper-V complains that the image is corrupt.

agners, 2021-02-03:
Are there more information in the Event Log or similar what the problem is exactly?

TheAndyMac, 2021-02-03:
Can't find anything in the event logs - no mention of the disk, probably as it is failing when you try to setup the new virtual machine (i.e. not even started at this point). Have tried setting up the VM without a disk, and then adding this - and again, the add of the virtual disk fails, so that you can't get any further.

agners, 2021-02-04:
I cannot reproduce this. Both images, 5.10 and 5.11 work fine on my end. I tried using Windows 10 Pro 10.0.19041 with Hyper-V. Make sure the download and unpack operations completes successfully (I used 7zip).

TheAndyMac, 2021-02-04:
Ok, so you need to unzip/unpack it... had not worked that out and it is not obvious from the instructions or from the extension (I'd just assumed it was named with an extra extension to cope with the vagueries of the upload/download process). Have now unzipped or unpacked (using PowerArchiver which recognised it properly) and it now works :-)

Can I suggest that this step is clearly added to the instructions on https://www.home-assistant.io/hassio/installation/ as part of step 2 which does not mention this at all for the virtual machines.

agners, 2021-02-05:
The instructions are now updated: https://github.com/home-assistant/home-assistant.io/pull/16412

closed 2021-02-05 by agners